robotology/community #332: Setting the order of module/application in yarpmanager

Yeshasvitvs, 2019-01-24:
I am trying to write a new yarpmanager application for `HumanDynamicsEstimation` https://github.com/dic-iit/component_andy/issues/192

In the xml file the application is divided into modules and embedded applications with the following order
- one module
- an embedded application
- another embedded application
- another module
- another module

Ideally I want to preserve and see the same order of the modules and application as written above. But when I am launching yarpmanger, the embedded applications go onto and the modules are loaded after them as shown below

[image]

The graph on the side shows the order I want to have

Is there any way to assign them the ordering based on using fixed process id in the config file or another way?

@Nicogene

Yeshasvitvs, 2019-01-24:
Changing all the modules to embedded applications preserves the order as defined in the main application xml file 

[image]

Nicogene, 2019-01-25:
Hi @Yeshasvitvs ,

you want to obtain this order in term of execution/stop of the modules?

Yeshasvitvs, 2019-01-25:
Ciao Nicolo,

Primarily I want to preserve the order of loading modules and embedded applications as I define them in the application xml file

Nicogene, 2019-01-25:
Digging int `libYARP_manager` I found out that here : 

https://github.com/robotology/yarp/blob/master/src/libYARP_manager/src/kbase.cpp#L215-L237

a `sort` is performed over the modules.

The sorting seems alphabetical : 

https://github.com/robotology/yarp/blob/master/src/libYARP_manager/include/yarp/manager/kbase.h#L210-L222

Maybe @apaikan knows if there is a sort of priority for applications.

Yeshasvitvs, 2019-01-25:
From what I understand during my testing the order with in the modules and with in the embedded applications is preserved i.e if my main application is written as 

- Module1
- EmbeddedApplication1
- EmbeddedApplication2
- Module2
- Module3

Then when they are loaded in `yarpmanager` the order is

- EmbeddedApplication1
- EmbeddedApplication2
- Module1
- Module2
- Module3

Alternatively, if my main application is written as 

- Module2
- EmbeddedApplication2
- EmbeddedApplication1
- Module1
- Module3

Then when they are loaded in `yarpmanager` the order is

- EmbeddedApplication2
- EmbeddedApplication1
- Module2
- Module1
- Module3

Nicogene, 2019-01-25:
Ah ok so that sort is not alphabetical or the order is changed again later, it would be nice solve this mystery 😅

Yeshasvitvs, 2019-01-25:
A simple way to handle this problem is to put even a single modules into an embedded application and the resulting order in yarpmanager is exactly as defined in the application xml file. However I wonder if using a `ID` tag to enforce the order is a better way to tackle the issue. what do you think ?

pattacini, 2019-03-04:
Any update on this?

Nicogene, 2019-03-13:
> Primarily I want to preserve the order of loading modules and embedded applications as I define them in the application xml file

Since it is not clear the order of the loaded xml files I suggest to open a PR that make possible the preservation of the order that is in the xml file.

Yeshasvitvs, 2019-03-13:
> I suggest to open a PR that make possible the preservation of the order that is in the xml file

@Nicogene before the PR we need to see what is the best way to enforce the order of the modules/applications

pattacini, 2019-04-11:
This is a `yarp` related discussion that didn't come to a conclusion yet.
Moving to `yarp` repo then.