robotology/community #333: iCub velocity control

guiolpei, 2019-01-25:
Hello,

I am a master student working on the iCub at Munich ICS (the robot is iCubMunich01 with upgrades) and we are trying to control the robot using the velocity commands, through the 'IVelocityControl' class.

The joints of the robot can be set to velocity control using the 'IControlMode' class or manually in the 'yarpmotorgui' control panel, but it doesn't seem to respond to the 'velocityMove' commands nor we can modify the velocity value using the slider in the 'yarpmotorgui'.

We have noticed that the 'Velocity PID' controller values are not defined, and we can't modify them using the PID window from the 'yarpmotorgui':

[image]

When the desired values are changed and sent, current values won't update. However, the values of the 'Position PID' controller can be changed without any problems.

We have found the file where the parameters are defined, 'icub_right_arm.xml'. There are PID controllers defined for position and torque but not for velocity.

Is defining values in this 'xml' file the solution to be able to change the values of the PID controller, or the problem is that the joints can't be velocity-controlled?

Regards,
Guillermo

[icub_right_arm.txt](https://github.com/robotology/QA/files/2796244/icub_right_arm.txt)

pattacini, 2019-01-25:
Hi @guiolpei welcome to our community ✋ 

Let's leave the use of the GUI aside for a moment.

Could you post on [Gist][1] the relevant portion of the code you're using to control the robot in velocity? This way we can inspect, run and comment.

What I suspect is that you're not aware of the safety timeout of `100 ms` (if I'm not mistaken) that is used underneath the velocity control to avoid integrating indefinitely the command in case e.g. there's a network drop or similar critical events.

With this in mind, you have to keep sending velocity commands (even the same command as a sort of _heartbeat_) at a proper rate to sustain the control mechanism and prevent timeout expiration.

At the low level, the velocity commands are integrated to yield position setpoints; thereby, the relevant PID gains are still those of the position domain. In this respect, I don't know actually what are pure velocity PID gains used for; perhaps they're operational for those devices where a native low-level velocity control can be performed (@randaz81 might be of help here).

[1]: https://gist.github.com

guiolpei, 2019-01-25:
Thank you for your quick response, Ugo. 

The code we are using is here:

[https://gist.github.com/guiolpei/383cc7f24a5c8f233b5d90ed9719458a](url)

The control is done through another class that acts as a wrapper to the 'IControlMode' class.

This code works correctly on the iCub simulator, but fails to work on the real robot. In the simulator the joint moves at 5 deg/s until the 0 velocity command is sent. The GUI also shows the correct value for the velocity.

We weren't aware that there was a safety timeout of `100 ms`, so probably this is the problem...

I understand then that PID velocity is not needed as you said, and as it is depicted in the following figure:

[image]

pattacini, 2019-01-25:
Yep, the diagram is correct 👍 

In the following piece of your main
```c++
iCubRightArmCtrl->velocityMove(3, 5.0);
iCubRightArmCtrl->velocityMove(3, 0);
```
you're basically commading a movement in velocity and soon afterward you stop it.

A typical velocity control loop resembles this snippet instead:
```c++
while (abs(error)>threshold) {
    error=desired-feedback;
    double velocity=Kp*error;
    ivelocity->velocityMove(joint,velocity);
}
```

pattacini, 2019-01-26:
Added another possible loop for completeness.

closed 2020-12-08 by pattacini