robotology/community #335: Hand tactile skin

tasbolat1, 2019-01-31:
Hello everyone,
iCub's finger tactile skin seems less sensitive that other parts. Recently I was informed that there are some configuration files that can handle the issue. First of all, I would like to know about those files, can you please direct me where those files are located?

tasbolat1, 2019-01-31:
Hi @xEnVrE  thank you for your response,
I understand that the raw tactile data are processed through `skinManager`. However, I'm interested in raw data. When I visualize the raw data through `iCubSkinGui`, I can see that the fingers are much more sensitive. Previously I also asked question related to that [in this link](https://github.com/robotology/QA/issues/296). Recently I was informed that there are some configuration files that can help me to  tune sensitivity. I would like to  know the location of those files?

lornat75, 2019-01-31:
Hi @tasbolat1, there is a gain level that can be tuned but I am not sure it is accessible from configuration files. Plus, increasing that gain also amplifies the noise, fingertips should be already shipped with the 'optimal' gain.

maggia80, 2019-02-06:
please have a look at the PR: https://github.com/robotology/robots-configuration/pull/89/commits/84b94b3222b6f1d4fcf38c39726c9179c30633c6

if you want to test it, you could do the change in your local configuration files. This change applies to the left hand.

julijenv, 2019-02-26:
Hi @tasbolat1 , did you try the suggestion done by @maggia80 ?
let us know,

pattacini, 2019-03-04:
Closing, feel free to open it again if needed.

tasbolat1, 2019-03-13:
Hi @maggia80 ,
I have made suggested changes locally and conducted simple experiment. Basically I am repeating the following experiment given in the [issue](https://github.com/robotology/QA/issues/296).  So, I put weight meter on the table and I let icub's index finger to touch it slowly. I keep monitoring the finger movement by `yarpmotorgui`. After some time, fingers physically cannot move further and weight meter shows ~200 g. The results are given in the figure below
[image]
Now about results: since I am working with raw tactile data, only three taxels are activated (green, orange, brown).  Usually less taxels are activated, however in this experiment I wanted to show the best scenario where more taxels are activated. Hence, in the best scenario we have three taxels activated and for 2N force, the maximum activated taxel level is 55 (including initial DC bias). However, for many manipulation in hand tasks or any task related to tactile skin, this level of sensitivity is not enough. Can anyone help me to understand whether I can improve sensitivity or this is how icub's skin operates?

julijenv, 2019-03-13:
Hi @tasbolat1 ,
could you please confirm that the values you took are coming from the left fingertips on which you appplied the fix provided by @maggia80 ?!?!
Please catch me on skype ; icub.support

pattacini, 2019-04-11:
Discussions went on offline between @xEnVrE @giuliavezzani and @tasbolat1.
Closing for inactivity.

closed 2020-12-15 by pattacini