robotology/community #337: FW update of eLoader and eUpdater: iCubLausanne02

MichaelBombile, 2019-03-22:
Hi,
I have updated the Firmware of iCubLausanne02 by following all the steps an everything seemed to have gone well.
But when I launch yarprobotinterface I am getting the described in the log
I tried to update the robots-configuration files but it didn't work
Can anyone help us to resolve this issue??

Thanks

[yarprobotinterface_log_iCubLausanne02_22_03_19.txt](https://github.com/robotology/QA/files/2997808/yarprobotinterface_log_iCubLausanne02_22_03_19.txt)

pattacini, 2019-03-24:
Multiple occurrences of the following error are reported:
```sh
------ ATTENTION!!!! Wrong value of <MotioncontrolVersion> parameter !!!! --------
------ This means that the configuration files of this device are not compatible with my parser, so I cannot start.  
------ I need version  5 , but in configuration files have version  4 . 
------ Please update configuration files in robots-configuration repository. (see http://wiki.icub.org/wiki/Robot_configuration for more information).  
```

@MichaelBombile could you please double check whether the robot configuration files you updated are those actually loaded at start up? The so-called shadowing system may subtly interfere with that. See #42 for more info.

From the log, it comes out that you're currently starting up the robot with conf files located under `/home/icub/.local/share/yarp/robots/iCubLausanne02`. Updating and installing robot files in `robot-configuration` repo does not automatically take the home path into account. To this end, instead, you ought to proceed by issuing:

```sh
$ yarp-config robot --import iCubLausanne02
```

Since you're relying on the home path, I wonder if you're using your own customization for the robot that is not present in `robot-configuration/iCubLausanne02`.

If this is the case, my warm recommandation is, in order:
1. Backup the content of the `/home/icub/.local/share/yarp/robots/iCubLausanne02` elsewhere.
1. Purge the content of `/home/icub/.local/share/yarp/robots/iCubLausanne02`, no leftover.
1. Import the new files into the home path (see above).
1. Reapply your customization by doing a folder-wise diff between `/home/icub/.local/share/yarp/robots/iCubLausanne02` and the backed up archive (you can use `meld` for that).

Alternatively, if you don't care about any possible customization, just purge the current content of `/home/icub/.local/share/yarp/robots/iCubLausanne02`. As a result, pristine conf files will be loaded from the location where you installed `robot-configuration`. To check for this do:

```sh
$ yarp resource --context yarprobotinterface --from yarprobotinterface.ini
```

---
@marcoaccame in the meantime could you please verify that `iCubLausanne02` conf files are up-to-date to be used along with protocol 5 of the FW?

MichaelBombile, 2019-03-30:
Hi @pattacini,

Thanks for your response and sorry for my late feedback.
I am on leave for a couple of days.
Once, back to the lab I will try what you suggested. 
But as far as I can remember, I purged the  `$  /home/icub/.local/share/yarp/robots/iCubLausanne02 ` and imported the new files that I pulled from the master branch of `$ robot-configuration `.
Could it be a problem with the new files ?

pattacini, 2019-04-01:
Hi @MichaelBombile 
Never mind 👍 

I'm not responsible for the content of those files, hence I won't be of help.
Let me point this to @marcoaccame.

pattacini, 2019-04-01:
I had a quick visual inspection of the files together with @marcoaccame and it came out that they actually need to be upgraded to version 5.

cc @randaz81

Nicogene, 2019-04-01:
For the migration to the version five you can follow what it has been done in these 2 PRs

https://github.com/robotology/robots-configuration/pull/88
https://github.com/robotology/robots-configuration/pull/90

Pay attention to the commit messages, the `sed` commands have been reported there.

pattacini, 2019-04-01:
Thank you @Nicogene for sharing this info!

@MichaelBombile you might be willing to give it a try by opening a PR, although I think @randaz81 would like to oversee this migration.

valegagge, 2019-04-02:
If I'm not wrong, exist a document that describes the procedure to migrate from version 4 to version 5.
@Nicogene @randaz81 Do you have it?

MichaelBombile, 2019-04-18:
Hi all,
Thank you very much for your feedbacks. I am managed to migrate from version 4 to version 5 as indicated by @Nicogene and it resolved the first problem.
However, I am now getting some errors with one board as you can see in the attached log file
[log_error_iCubLausanne02_18042019.txt](https://github.com/robotology/QA/files/3093171/log_error_iCubLausanne02_18042019.txt)

Based on those errors, I tried to update the firmware of that board, but the problem still persists.

Can anyone suggest what I should do now?

Thanks in advance.

Nicogene, 2019-04-18:
> I am managed to migrate from version 4 to version 5 as indicated by @Nicogene and it resolved the first problem.

Glad to hear that 😃 

> Based on those errors, I tried to update the firmware of that boards, but the problem still persists.

Another necessary step is to update the firmware of **all** the board, getting the latest firmware( [`icub-firmware-build`](https://github.com/robotology/icub-firmware-build)) and loading it into the boards using the `FWUpdater` with `--admin` parameter.

Is it your robot `CAN` or `ETH` ? With a rapid glance to the log your posted it seems that is `ETH` but you loaded the firmware for `CAN`.

The firmware for `ETH` is located [here](https://github.com/robotology/icub-firmware-build/tree/master/ETH).

Please @marcoaccame and @valegagge correct me if I'm telling something wrong.

MichaelBombile, 2019-04-18:
Hi @Nicogene ,

Our robot is an ETH. However, the CAN boards that caused problem are attached to an EMS board.
I tried to update both the EMS and the CAN FW, with EMS/bin/application/ems.hex file and the /icub-firmware-build/CAN/2foc/2foc.hex respectively. It seemed to have worked for the other boards except that one.
@marcoaccame, is there something wrong I did with the FW? and how can I resolve this FW problem.

julijenv, 2019-04-18:
Hi @MichaelBombile ,
please send us he printsceen of the FirmwareUpdater after having dopne the discover on ETH, entering for all the EMS in manutention mode, and do a discover again below each EMS.
thx in advance

MichaelBombile, 2019-04-18:
Hi @julijenv 

Here are the printscreen
[image]
[image]
[image]

Thx.

julijenv, 2019-04-19:
Hi @MichaelBombile ,
Could you run another yarprobotinterface in order to get the new logfile after having done the FW update on the boards underneath the EMSs.
thx in advance

MichaelBombile, 2019-04-19:
Hi @julijenv,
Do you mean that I do another FW update and run yarprobotinterface? 
Cause, before sending the printscreen, I had tried to update the boards and run yarprobotinterface to get the same error with the board with IP 10.0.1.3 .

julijenv, 2019-04-19:
Hi @MichaelBombile ,
Well if you already did the update I dont reallly know why you got this message but I talked to one of our FW consultant and he told me he should have an eye on your case, so I wanted to get a brand new logfile after the update,just in case that it would be a little bit different BUT If you tell me that the log file above quoted is sill the same even after having updated the FW on the board 10.0.1.3 and also on the 2FOC boards I think that it still a piece of information that could be requested by my colleague later and it is good to mention it now, So lets wait for him to see the thread.
@marcoaccame WCS if you dont get any reply before tuesday (on monday unfortunately it is national holiday her ein Italy) just try to catch me on skype, my business ID is icub.support.
cheers

MichaelBombile, 2019-04-23:
Hi @julijenv,
Thanks for the skype call suggestion. 
Yesterday was also a public holiday here.
I have just tried to call you, but you are not online.
could please let me know when you will be available.

Thanks

marcoaccame, 2019-04-23:
Hi @MichaelBombile, 
there is a typo in the file `right_arm-eb3-j0_3-mc_service.xml`. 

It must have required can protocol = 1.6 for the foc board as all other `mc_service.xml` file have.

Here is how to modify it.

```
            <group name="CANBOARDS"> 
                <param name="type">                 foc         </param> 
                <group name="PROTOCOL">
                    <param name="major">            1           </param>    
                    <param name="minor">            6           </param>     
                </group>                    
                <group name="FIRMWARE">
                    <param name="major">            0           </param>    
                    <param name="minor">            0           </param> 
                    <param name="build">            0           </param>
                </group>
            </group>
```

[image]

MichaelBombile, 2019-04-23:
Hi @marcoaccame,

Thanks for this fix. The error on the board has been resolved.
However, when I run yarprobotinterface, there are not errors but some warnings about some Autocalibration parameters and it stops with segmentation fault. Here is the log file
[log_error_iCubLausanne02_23042019.txt](https://github.com/robotology/QA/files/3106948/log_error_iCubLausanne02_23042019.txt)

Thanks

marcoaccame, 2019-04-23:
I saw the warnings. 
The missing parameters in the xml files are optional and should do no harm (as @ale-git confirms).

The crash must be something else. What about the Linux version and gcc version you have in your pc104? I remember a nasty crash case in iCubNancy01 which has an old linux distro. I will look for the issue in icub-support.

By the way, maybe it is better to move this issue as well? what do you think @julijenv ?

MichaelBombile, 2019-04-23:
Thanks @marcoaccame , so I am waiting for @julijenv 's suggestions.

marcoaccame, 2019-04-23:
In issue https://github.com/robotology/icub-support/issues/658, around May 25, there are some tests we did to find out the reason of a similarly strange crash.

julijenv, 2019-04-23:
Hi @MichaelBombile , from what I could compare to another robot like yours, it appeared that the upgrade to version 5 has not been done properly. I saw a lot of differences that could create ths behavour but of course I would need @valegagge  or @randaz81  to confirm that.

MichaelBombile, 2019-04-23:
Hi @julijenv ,

Thanks for your response. So I am waiting for their confirmation.
Cheers

MichaelBombile, 2019-04-25:
Hi @julijenv,
Did you get any feedback form  @valegagge or @randaz81 about the issue?

Nicogene, 2019-04-26:
Hi @MichaelBombile , 

if you used the `sed` commands I posted, maybe you need to these 2 commit to fix the conf files:

https://github.com/robotology/robots-configuration/commit/41017338ea6ef2192dda74eb172c2ac0d4424642#diff-5b505bc49e91ee919e0eaf4035de89d7

https://github.com/robotology/robots-configuration/commit/f01877782b8a63a71a11a86f9a77a77dec998b6d#diff-5b505bc49e91ee919e0eaf4035de89d7

MichaelBombile, 2019-04-26:
Hi @Nicogene ,

Thanks for your suggestion. I implemented it, but the problem still persists.

julijenv, 2019-04-29:
Hi @MichaelBombile,
can you try to contact me on skype, my ID is icub.support and we will try to understand better the problem under the hood.
thx inadvance

MichaelBombile, 2019-04-30:
Hi @julijenv,
I am trying to contact you on skype but it seems that you are not online.
Can we schedule an meeting so that we can talk when we are all available?

Thanks

julijenv, 2019-04-30:
Hi @MichaelBombile ,
sorry I was out of the office today because I was repairing some robots in another place.
TOmorrow is a national holiday here in genova so I guess we can get in touch on thursday if it is ok for you

MichaelBombile, 2019-05-02:
Hi @julijenv 

Here is the last log file

[log_error_iCubLausanne02_26042019.txt](https://github.com/robotology/QA/files/3137406/log_error_iCubLausanne02_26042019.txt)

MichaelBombile, 2019-05-02:
Here is the link to the config files

https://drive.google.com/file/d/1vkZ1seeFCsYclwJjhSH-fDd1Z6Y5ZJz9/view?usp=sharing

MichaelBombile, 2019-05-03:
Hi, Thanks very much to all for your help.
And special thanks to @julijenv, who spent an entire afternoon in fixing the problem.
The robot can now start despite some errors on two joints: MC: hard limit reached,
and some problem on the finger with broken tendons.

So, I can close this issue.

closed 2020-12-15 by pattacini