robotology/community #338: [Announcement] YCM 0.10.0 Released

drdanz, 2019-04-01:
YCM 0.10.0 was released today and is now available for download at: https://github.com/robotology/ycm/releases/.

Changes made since YCM 0.9 include the following.

## Build System

* YCM can now be included safely in another project using `FetchContent`.

## Variables

* The following varibales are no longer cached, since they should be set from the project maintainer, and not by the user:
  * `YCM_USE_CMAKE_<VERSION>`
  * `YCM_USE_CMAKE_NEXT`
  * `YCM_USE_CMAKE_PROPOSED`
  * `YCM_USE_3RDPARTY`
  * `YCM_USE_DEPRECATED`
* The `YCM_USE_CMAKE` can now be used to disable simultaneously all the `YCM_USE_CMAKE_<VERSION>`, `YCM_USE_CMAKE_NEXT`, and `YCM_USE_CMAKE_PROPOSED` variables.

## Modules

### Generic Modules

* The `InstallBasicPackageFiles` module learned to forward the `ARCH_INDEPENDENT` option to `write_basic_package_version_file`.
* The `IncludeUrl` module no longer uses locks.
* The `InstallBasicPackageFiles` module no longer looks for the `Config.cmake.in` in the root directory, only current source directory is supported.
* The `YCMDefaultDirs` module default dirs now use `CMAKE_CURRENT_BINARY_DIR` instead of `CMAKE_BINARY_DIR`.
* The `YCMDefaultDirs` module is now deprecated.
* The `InstallBasicPackageFiles` module no longer needs `FIRST_TARGET` when used with `EXPORT`. Using `FIRST_TARGET` is deprecated.
* `InstallBasicPackageFiles`: the `TARGETS`, `TARGETS_PROPERTY`, and `TARGETS_PROPERTIES` arguments are deprecated.
* The `AddUninstallTarget` module must now be included in your main `CMakeLists.txt`. If included in a subdirectory it does nothing.
  This allows you to use it safely in your main `CMakeLists.txt` and include your project using `add_subdirectory` (for example when using it with `FetchContent`).
* The `AddUninstallTarget` module no longer tries to add the `uninstall` target if the target it already exists, instead it does nothing.
* `InstallBasicPackageFiles`: The list of targets is no longer used to decide whether `PRIVATE_DEPENDENCIES` should be enabled or not (only the `BUILD_SHARED_LIBS` variable is used).
  Dependencies for `STATIC`, `OBJECT`, and `INTERFACE` libraries should be passed as `DEPENDENCIES` instead.
* `InstallBasicPackageFiles`: The `NO_COMPATIBILITY_VARS` option is deprecated. The compatibility variables are no longer generated by default, and their generation is broken when using `EXPORT`. If required, it is possible to re-enable it using the `ENABLE_COMPATIBILITY_VARS` option (deprecated) or define them using either `INCLUDE_FILE` or `INCLUDE_CONTENT` (recommended).
* `InstallBasicPackageFiles`: Fixed malformed `find_package()` when a package dependency had a required version, for CMake older than 3.9.
* The `InstallBasicPackageFiles` module now considers a relative `EXPORT_DESTINATION` as relative to `CMAKE_CURRENT_BINARY_DIR` instead of `CURRENT_BINARY_DIR`.
* `InstallBasicPackageFiles`: Fixed export destination when `EXPORT_DESTINATION` is an absolute path.

### Superbuild Helper Modules

* The `YCMEPHelper` module now supports the `CMAKE_CACHE_DEFAULT_ARGS` argument.
* The `YCMEPHelper` module supports the `SOURCE_SUBDIR` argument. The `CONFIGURE_SOURCE_DIR` argument is now deprecated in favour of `SOURCE_SUBDIR`.
* The `YCMEPHelper` module now passes the variable `CMAKE_TOOLCHAIN_FILE` to the children projects.
* `YCMEPHelper`: The `CMAKE_PREFIX_PATH` list passed to the subproject now has the first element the installation directory of the superbuild. This is to ensure that if a package is available in the system but it is disabled using `USE_SYSTEM_<package>` set to `OFF`, the version installed by the superbuild is the one found by the other packages in the superbuild.
* The `YCMEPHelper` module now adds subproject to the main install target, if they provide a `cmake_install.cmake` file in their build directory. This file is automatically provided by CMake for projects installing at least one file. This feature should be considered experimental and might be modified or removed in the future. Please note that non-relocatable file will contain the path for the project installation (i.e. the path to the `install` folder inside the superbuild binary directory), not to `CMAKE_INSTALL_PREFIX`.

### Find Modules

* The `FindLibOVR` module now works properly with VS2015.
* The `FindPortAudio` module should now detect the library installed by vcpkg.
* The `FindPortAudio` module now handles debug and release libraries.
* Import `FindI2C` module from [YARP Git Repository](https://github.com/robotology/yarp/).
* The `FindACE` module now creates the `ACE::ACE` and `ACE::ACE_INLINE` imported targets. The `ACE_ADDR_HAS_LOOPBACK_METHOD` and `ACE_COMPILES_WITHOUT_INLINE_<CONFIG>` are now deprecated.
* The `FindQt3` module is deprecated in favour of one from CMake.
* Added `Findassimp` module to locate the Open Asset Importer Library (assimp).
* Added `FindGLM` module to locate the OpenGL Mathematics (GLM) library.

### CMake Next

* Updated to CMake v3.14.0 + commit `e67e4dbaadf3583d665fb5409b425e1a238d9748`.
* Imported `FindDoxygen` module from CMake 3.9.
* Imported `WriteBasicConfigVersionFile` and `CMakePackageConfigHelpers` modules from CMake master, to support the `ARCH_INDEPENDENT` option in `write_basic_package_version_file`.
* Imported `FetchContent` module from CMake 3.11.
* Imported `FindOpenGL` module from CMake 3.10.
* The `ExternalProject` module was updated from CMake next. All the customizations were removed, with the exception of the `git` deleting clones fix.
* The `FindGlew` module is now dowloaded from CMake master.

### 3rd Party

* Imported `FindGObject` module from [qt-gstreamer Git Repository](https://github.com/GStreamer/qt-gstreamer/) + some unmerged changes from robotology-dependencies fork.
* Imported `CMakeRC` module from [CMakeRC Git Repository](https://github.com/vector-of-bool/cmrc/).
* Update [Catch2 Git Repository](https://github.com/catchorg/Catch2/) to commit `4902cd721586822ded795afe0c418c553137306a` + some unmerged changes from robotology-dependencies fork.
* Imported `UseLATEX` module from [UseLATEX Git Repository](https://gitlab.kitware.com/kmorel/UseLATEX/).

PeterBowman, 2019-04-01:
@drdanz the hash id for `ExternalProject.cmake` needs to be updated, `file(DOWNLOAD)` [reports errors](https://travis-ci.org/roboticslab-uc3m/asibot-main/jobs/514105296) after [this commit](https://github.com/robotology/ycm/commit/9249071386282ab20ca80016e03cdd477902408c).

drdanz, 2019-04-01:
Fixed, thanks.
I will release 0.10.1 in a couple of days

closed 2020-12-12 by pattacini