robotology/community #342: How to calibrate camera

guiolpei, 2019-04-10:
The left eye camera from the iCub robot at Munich ICS had to be replaced some time ago and we are trying to calibrate it again (obtain the extrinsic and intrinsic parameters) with no success.

As far as we know, the calibration must be done using the _stereoCalib_ module, but we can not make it work. 

This is what we have tried:

1. Run the stereoCalib module, which gives the following output:

```
icub@icubsrv:~$ stereoCalib
||| clearing context
||| adding context [cameraCalibration]
||| configuring
||| default config file specified as icubEyes.ini
||| checking [/home/icub/icubEyes.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubMunich01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubMunich01] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-gnome/yarp/robots/iCubMunich01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/iCubMunich01] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/iCubMunich01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot//yarp/build/share/yarp/robots/iCubMunich01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-main/build/share/iCub/robots/iCubMunich01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot//icub-main/build/share/iCub/robots/iCubMunich01
||| checking [/usr/local/src/robot//icub-contrib-install/share/ICUBcontrib/robots/iCubMunich01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//yarp/build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-main/build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-contrib-install/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot//icub-main/build/share/iCub/robots/iCubMunich01/icubEyes.ini] (robot)
||| found /usr/local/src/robot//icub-main/build/share/iCub/robots/iCubMunich01/icubEyes.ini
yarp: Port /stereoCalib/cmd active at tcp://10.0.0.1:10034
yarp: Port /stereoCalib/cam/left:i active at tcp://10.0.0.1:10035
yarp: Port /stereoCalib/cam/right:i active at tcp://10.0.0.1:10036
yarp: Port /stereoCalib/cam/left:o active at tcp://10.0.0.1:10037
yarp: Port /stereoCalib/cam/right:o active at tcp://10.0.0.1:10038
yarp: Port /stereoCalib/client/head/rpc:o active at tcp://10.0.0.1:10039
yarp: Port /stereoCalib/client/head/command:o active at tcp://10.0.0.1:10040
yarp: Port /stereoCalib/client/head/state:i active at tcp://10.0.0.1:10041
yarp: Port /stereoCalib/client/head/stateExt:i active at tcp://10.0.0.1:10042
yarp: Sending output from /stereoCalib/client/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /stereoCalib/client/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/state:o to /stereoCalib/client/head/state:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /stereoCalib/client/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
yarp: Port /stereoCalib/client/torso/rpc:o active at tcp://10.0.0.1:10303
yarp: Port /stereoCalib/client/torso/command:o active at tcp://10.0.0.1:10304
yarp: Port /stereoCalib/client/torso/state:i active at tcp://10.0.0.1:10305
yarp: Port /stereoCalib/client/torso/stateExt:i active at tcp://10.0.0.1:10306
yarp: Sending output from /stereoCalib/client/torso/rpc:o to /icub/torso/rpc:i using tcp
yarp: Sending output from /stereoCalib/client/torso/command:o to /icub/torso/command:i using udp
yarp: Receiving input from /icub/torso/state:o to /stereoCalib/client/torso/state:i using udp
yarp: Receiving input from /icub/torso/stateExt:o to /stereoCalib/client/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]Running Stereo Calibration Mode... 
```

2. We then connect using rpc and start the calibration process:

```
icub@icubsrv:~$ yarp rpc /stereoCalib/cmd
start
Response: "Starting Calibration..."
```

In the other terminal we have the following response:

`command received: start`

We show the calibration chessboard pattern to the robot and nothing seems to happen...

Any idea why this does not work?

Regards,
Guillermo

guiolpei, 2019-04-10:
The content of the icubEyes.ini in `/usr/local/src/robot//icub-main/build/share/iCub/robots/iCubMunich01/icubEyes.ini` is the following:

```
[CAMERA_CALIBRATION_RIGHT]

projection         pinhole
drawCenterCross    0

w 320
h 240
fx 217.597
fy 217.061
cx 163.336
cy 130.794
k1 -0.37136
k2 0.151529
p1 -0.00131403
p2 -0.00130408

[CAMERA_CALIBRATION_LEFT]

projection         pinhole
drawCenterCross    0

w 320
h 240
fx 220.263
fy 220.072
cx 186.25
cy 118.905
k1 -0.386021
k2 0.168109
p1 0.000383998
p2 -0.000356036

[CAMERA_CALIBRATION_CONFIGURATION_LEFT]

numPatternImagesRequired    10
numPatternInnerCornersX     8
numPatternInnerCornersY     6
patternSquareSideLength     25
outputFilename              /tmp/results.ini
outputGroupname             CAMERA_CALIBRATION_LEFT

[CAMERA_CALIBRATION_CONFIGURATION_RIGHT]

numPatternImagesRequired    10
numPatternInnerCornersX     8
numPatternInnerCornersY     6
patternSquareSideLength     25
outputFilename              /tmp/results.ini
outputGroupname             CAMERA_CALIBRATION_RIGHT

[STEREO_CALIBRATION_CONFIGURATION]
boardWidth 8
boardHeight 6
boardSize 0.051
numberOfPairs 30
MonoCalib 0 
```

fbottarel, 2019-04-11:
Hi Guillermo, 
I'm taking a look at the `stereoCalib` run log you posted. Everything seems nominal, however it looks like these ports (opened by stereoCalib)
```
yarp: Port /stereoCalib/cam/left:i active at tcp://10.0.0.1:10035
yarp: Port /stereoCalib/cam/right:i active at tcp://10.0.0.1:10036
```
do not receive any connection. They should receive input streams from the robot cameras, otherwise the system has no images to perform calibration with. Please make sure you are connecting ports 
```
/icub/cam/right
/icub/cam/left
```
to, respectively, 
```
/stereoCalib/cam/left:i
/stereoCalib/cam/right:i
```
before sending the rpc command. If everything is running smooth, you should see the chessboard pattern in images being streamed to `stereoCalib` image output ports. In order to check for this, you can just open a couple of viewers and connect the ports
```
yarpview --name /view/left &
yarpview --name /view/right &
yarp connect /stereoCalib/cam/left:o /view/left
yarp connect /stereoCalib/cam/left:o /view/right
```

If this is not helpful, we can look further into it!

pattacini, 2019-04-11:
Nice catch @fbottarel 👍 

@guiolpei you may find this [app][1] useful to run the module since it also contains the connections to make after launch.

Another useful resource on camera calibration is available in our [wiki][2].

[1]: https://github.com/robotology/icub-main/blob/master/app/cameraCalibration/scripts/stereoCalib.xml.template
[2]: http://wiki.icub.org/wiki/Stereo_calibration

guiolpei, 2019-04-11:
Connecting the ports fixed the problem. Thank you very much!

Regards,
Guillermo

closed 2020-12-15 by pattacini