robotology/community #346: various Wiki pages & tutorials offline

frank-foerster, 2019-04-24:
Hi, 

I'm just updating some of my Python code that uses cartesian control, position control, ARE commands, and some other stuff, and noticed that most of the control-related Wiki pages are offline / not accessible. 

Could someone put them online again, or point me to the equivalent documentation in case it has been replaced? 
Here is a (non-exhaustive) list of webpages, that is currently offline and linked from various other sites such as the VVV tutorials. 

wholeBodyDynamics:
http://eris.liralab.it/iCub/main/dox/html/group__wholeBodyDynamics.html
(linked, for example, from Force Control Wiki entry: http://wiki.icub.org/wiki/Force_Control)

gravityCompensator:
http://eris.liralab.it/iCub/main/dox/html/group__gravityCompensator.html
(dito)

iCub tutorials:
ttp://www.icub.org/doc/icub-main/dox/html/icub_tutorials.html
(linked from main Wiki site: http://wiki.icub.org/wiki)

Cartesian Interface:
ttp://www.icub.org/doc/icub-main/dox/html/icub_cartesian_interface.html

Gaze interface:
ttp://www.icub.org/doc/icub-main/dox/html/icub_gaze_interface.html

Many thanks in advance!

pattacini, 2019-04-24:
Hi @frank-foerster 

We no longer use the server `eris` hosted at **Liralab** for publishing the documentation since long by now.

You can reach out to our pages at http://www.icub.org/doc/icub-main/.

frank-foerster, 2019-04-24:
Thanks! Yes, I already thought that the liralab links were outdated. But some pages that were offline, were hosted on the (new) server hosting icub.org. 

They are online now, however, so thanks for fixing that!

pattacini, 2019-04-24:
Actually, if you carefully compare the URL you provided with the one I posted you'll notice differences:
- you posted http://www.icub.org/doc/icub-main/dox/html/icub_cartesian_interface.html
- I posted http://www.icub.org/doc/icub-main/icub_cartesian_interface.html

This is to say that we haven't fix any problem, rather the way doxygen generates links must have changed at some point in the past. To avoid getting trapped in these hitches, the best way is to always reach those pages through our main entry point http://www.icub.org/doc/icub-main/.

frank-foerster, 2019-04-24:
Hi Ugo,
I actually did obtain the posted URLs via going through  http://www.icub.org/doc/icub-main/ - I did not save them in my bookmarks or the like.  So either my browser held a very old buffer from those pages, or something did change on the server side.