robotology/community #347: Changing  force thresholds in hand_sequences.ini for ARE

frank-foerster, 2019-04-25:
Hi,

when trying to initiating some grasping and other hand-related actions using the ActionsRenderingEngine (ARE), the force thresholds as specified for each action and joint are surpassed pretty much all the time such that no grasping takes place (see log-file excerpt below). _This happens without any object being present_. 

The (Python-) script I am using used to work, but after a hard drive failure and a complete re-install of the OS, yarp, icub-main etc, I am now experiencing the above problem.

**I am planning to iteratively increase the allowed thresholds in the `hand_sequences.ini` file such that the grasping (and other) actions will finally be executed and wonder if anything speaks against that.** 

Also: if anyone has any idea what could have caused this change/spike in the sensor-readings I'd be curious to know.

The script I am using is the same that was used in the context of [icub-main/issue #417](https://github.com/robotology/icub-main/issues/417), which means that I send a `0` to `/wholeBodyDynamics/rpc:i` for re-initialization prior to sending the ARE command. 
Back then, I had a similar issue with some arm joint, but that was solved by a) increasing the allowed threshold and, b) having wholeBodyDyamics re-initialize prior to each ARE action.

Here are some snippets from the ARE log-file that show  how the force thresholds are exceeded once some command is received:

[log_icubsrv_actionsRenderingEngine_7138_excerpt.txt](https://github.com/robotology/QA/files/3117170/log_icubsrv_actionsRenderingEngine_7138_excerpt.txt)

As can be seen from the log-file pretty much every involved joint exceeds the thresholds specified in `hand_sequences.ini`.

Another thing that changed in the script is the use of the new `IControlMode2` interface as the old one has become deprecated. I am using a mix of cartesian control, position control, and ARE commands, and prior to issuing position control commands I issue a
`iCtrl.setControlMode(j, yarp.encode('pos')`, where `iCtrl` is an instance of `iControlMode2`
(before that, in the old interface, I used `posCtrl.setPositionMode()`).
I don't think this has any bearing on the issue, but I just thought I mention it in case anything else has changed in the meantime. 
(I don't, for example, switch the control mode back to some other mode after using position control and prior to issuing ARE commands, assuming ARE will take care of that by itself).

Thanks in advance for any help.

pattacini, 2019-04-26:
Hi @frank-foerster 

### Thresholds in `hand_sequences.ini`
Did you try to recalibrate the fingers? This may help out:
```sh
$ yarp rpc /actionsRenderingEngine/cmd:io
>> calib fingers
```

### ARE w/ WBD
`ARE` already talks to `WBD` to make sure that it is calibrated force-wise prior to commencing an action, thus, in principle, you wouldn't need to do that yourself explicitly.
The connection in question is: 
```xml
<connection>
  <from>/actionsRenderingEngine/wbd:rpc</from>
  <to>/wholeBodyDynamics/rpc:i</to>
  <protocol>fast_tcp</protocol>
</connection
```

frank-foerster, 2019-04-29:
Hi @pattacini 

no, I hadn't recalibrated the fingers. I did actually not know that you can do that. 
I will try this tomorrow and will report back. 

Thanks

frank-foerster, 2019-05-02:
@pattacini 
This did the trick! After calibration, no more "threshold violations"!
Many thanks!

pattacini, 2019-05-02:
Happy to know it helped!

closed 2020-12-08 by pattacini