robotology/community #353: reading joint angles from datasetplayer

pfdominey, 2019-06-05:
GReetings,
Please excuse this probably very simplistic issue.  I am running the datasetplayer with some recorded data.  I would like to use a simple interface (e.g. browser or telent) to read the joint angles for the torso, arms, head at different times durring replay.  
If there was an RPC port, I could just type "get pos all" or something like that.
but there is no rpc port.
I imagine there is a simple solution, just not simple for me!
thanks
cheers
Peter

pfdominey, 2019-06-05:
> Dear Peter,
> How are you? Good to hear/read from you :).
>
> I think the solution is easy indeed. Let us consider the example in
> https://www.yarp.it/group__yarpdataplayerExample.html
>
> Once you replay the data, you can use the same tools as usually to read
> the data. For example, in the terminal, you can yarp read ...
> /icub/head/state:o, which results in:
>
> yarp: Port /tmp/port/1 active at tcp://10.0.0.90:10025/
> yarp: Receiving input from /icub/head/state:o to /tmp/port/1 using tcp
> -6.021978 0.043956 0.835165 -4.692308 -2.986781 7.254816
> -6.197802 0.043956 0.835165 -4.428571 -3.357685 7.286458
> -6.285714 0.043956 0.835165 -4.164835 -3.751441 7.265364
>
> If you want to use your browser, simply open the following url (you might
> need to replace localhost with the IP of the machine where yarpserver is
> running): http://localhost:10024
>
> Then, click on the port you want to read, lets say /icub/torso/state:o,
> and finally click on "read".
>
> I hope this helps.
>
> Best, Tobi

Hi Tobias,
Nice to hear from you too.  Thanks, this almost works, but when I type

C:\Users\PeterDell>yarp read ... /icub/head/state:o

I get the beginnig of the expected response,

yarp: Port /tmp/port/1 active at tcp://172.30.133.171:10027/
yarp: Receiving input from /icub/head/state:o to /tmp/port/1 using tcp

but then it hangs, and there is no further response.
Similarly in the browser, I click on read and get the beginning of the
expected response:

Reading data from port...

but then nothing else.

could this be a firewall problem?

thanks

cheers

Peter

Tobias-Fischer, 2019-06-05:
Hi Peter,
I see two potential problems:
1) Firewall / network related problems as you mentioned. Try to run `yarpserver` and `yarpdataplayer` on the same machine to avoid this problem.
2) There is no data being streamed from `yarpdataplayer`. To make sure it is not a problem with your recording, could you use the [testdata](http://www.icub.org/download/software/datasetplayer-demo/testData_20120803_095402.zip) provided on https://www.yarp.it/group__yarpdataplayerExample.html instead of your recording and replay it? Make sure to be "fast enough", i.e. to read from the port before the recording has stopped.

pfdominey, 2019-06-05:
Thanks for sticking with me Tobias!
It is amazing how "dumb" an old researcher can be!
The problem was that I was trying to read these ports while the
datasetplayer was paused!
This is because I have the old habit of reading RPC ports when things are
paused.
So once I ran the data, it came spewing out just as desired!
Thanks - if only all problems were so easily solved!
Cheers
Peter

Tobias-Fischer, 2019-06-05:
Hi Peter, great to hear the problem is solved :+1:!

pfdominey, 2019-06-05:
one more question, to set all the joints at once in the simulator with RPC
is the command somehting like

set pos all -22.945055 89.145055 -14.978464 20.451648 -42.508243 -5.918975
-1.359729 -0.240035 9.842271 6.17 23.579767 1.515021 15.675892 5.390324
18.926066 14.851057

pfdominey, 2019-06-05:
got it
set poss (-22.945055 89.145055 -14.978464 20.451648 -42.508243 -5.918975
-1.359729 -0.240035 9.842271 6.17 23.579767 1.515021 15.675892 5.390324
18.926066 14.851057)
:)

closed 2020-12-15 by pattacini