robotology/community #357: Did anybody get married?

DanielePucci, 2019-06-29:
I would like to know if anybody in iCub married. Please help me.

francesco-romano, 2019-06-29:
[image]

IMHO @traversaro

DanielePucci, 2019-06-29:
Thank you very much @francesco-romano 

I think I got the  point.

Closing. 

If @traversaro doesn't agree, he can clearly open it again.

lorejam, 2019-07-02:
@Silvio Traversaro <Silvio.Traversaro@iit.it> , what a commit(ment)!

Cheers :)
Lorenzo

-- 
===================================================================
Lorenzo Jamone
Lecturer in Robotics
Turing Fellow - https://www.turing.ac.uk/people/researchers/lorenzo-jamone
ARQ - Advanced Robotics at Queen Mary - https://www.qmul.ac.uk/robotics/
School of Electronic Engineering and Computer Science -
https://www.eecs.qmul.ac.uk/
Queen Mary University of London - https://www.qmul.ac.uk/
https://lorejam.blogspot.com
l.jamone@qmul.ac.uk

Visiting Associate Researcher at VisLab (ISR/IST-Lisbon)
ljamone@isr.tecnico.ulisboa.pt
VisLab - http://vislab.isr.tecnico.ulisboa.pt
Institute for System and Robotics - https://welcome.isr.tecnico.ulisboa.pt/
Instituto Superior Técnico - https://tecnico.ulisboa.pt/

====================================================================

closed 2020-12-12 by pattacini