robotology/community #359: Could not find a package configuration file provided by   "YARP_robottestingframework"

smwasif, 2019-08-10:
Hi dear,
            I am trying to install using "robotology_superbuild" in Ubuntu 18.04, but during **make** I am encountered with the following error. Please help to sort it out.

[ 81%] Performing configure step for 'icub-tests'
Not searching for unused variables given on the command line.
loading initial cache file /home/wasif/robotology-superbuild/build/robotology/icub-tests/CMakeFiles/YCMTmp/icub-tests-cache-Release.cmake
CMake Error at /home/wasif/opt/yarp/lib/cmake/YARP/YARPConfig.cmake:183 (find_package):
  Could not find a package configuration file provided by
  "YARP_robottestingframework" with any of the following names:

    YARP_robottestingframeworkConfig.cmake
    yarp_robottestingframework-config.cmake

  Add the installation prefix of "YARP_robottestingframework" to
  CMAKE_PREFIX_PATH or set "YARP_robottestingframework_DIR" to a directory
  containing one of the above files.  If "YARP_robottestingframework"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:25 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/wasif/robotology-superbuild/build/robotology/icub-tests/CMakeFiles/CMakeOutput.log".
CMakeFiles/icub-tests.dir/build.make:101: recipe for target 'robotology/icub-tests/CMakeFiles/YCMStamp/icub-tests-configure' failed
make[2]: *** [robotology/icub-tests/CMakeFiles/YCMStamp/icub-tests-configure] Error 1
CMakeFiles/Makefile2:1173: recipe for target 'CMakeFiles/icub-tests.dir/all' failed
make[1]: *** [CMakeFiles/icub-tests.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

traversaro, 2019-08-10:
Hi @smwasif ! 
From your error, I can see that you your robotology-superbuild is finding an already existing YARP installation in your system at `/home/wasif/opt/yarp/` . However, that YARP installation is not compiled with the `YARP_COMPILE_RobotTestingFramework_ADDONS` that is instead required by `icub-tests` and other projects compiled by the superbuild. 

For this reason, I suggest either two possible solutions: 
* If you do not care of using your non-superbuild YARP installation, you can either uninstall it, or force the superbuild to ignore it by calling the CMake in the superbuild build directory with the `USE_SYSTEM_YARP` option set to OFF (see http://robotology.github.io/ycm/gh-pages/git-master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-developers), i.e. as : `cmake .. -DUSE_SYSTEM_YARP:BOOL=FALSE`
* If instead for any reason you need to have a `YARP` installation external to the superbuild, enable in your yarp installation the `YARP_COMPILE_RobotTestingFramework_ADDONS` option, and recompile/reinstall the project. Note that in this case you should also take care of installing the robot-testing-framework ( https://github.com/robotology/robot-testing-framework ) externally to the superbuild.

Let us know if this info solves your problem, thanks!

smwasif, 2019-08-10:
> Hi @smwasif !
> From your error, I can see that you your robotology-superbuild is finding an already existing YARP installation in your system at `/home/wasif/opt/yarp/` . However, that YARP [...]

smwasif, 2019-08-10:
Hi @traversaro,
                         Thank you very much for your prompt reply. You are right I had previous installation which I uninstalled using make uninstall but it was conflicting with new installation. Now after your devised changes my issue has resolved.

closed 2020-12-15 by pattacini