robotology/community #363: 📣 Upgrade ecosystem software by sticking to release tags

pattacini, 2019-09-29:
Hi @robotology/everyone

:mega: We'd like to let you know that from now onwards **we recommend developers to stick to specific releases of our software when building the source code**, instead of relying on `master` branches.

This way, we can make sure that upgrading your local copies of ecosystem's software (e.g. `yarp`, `icub-main`...) can be achieved as smoothly as possible.

:warning: Clearly, this new procedure applies only when you **build the main code yourself** and not when you opt for installing binaries.

Typically, when you decided to upgrade your system, you were used to doing the following:
```sh
$ cd $yarp
$ git fetch origin
$ git checkout master
$ git pull --rebase
...
$ cd $icub-main
$ git fetch origin
$ git checkout master
$ git pull --rebase
...
```
Now, you are required to switch to the proper release tags that are reported in our [**Software Versioning Table**][1] :rocket:

#### Here's an example:
```sh
$ cd $yarp
$ git fetch origin
$ git checkout v3.2.1
$ git pull --rebase
$
$ cd $icub-main
$ git fetch origin
$ git checkout v1.13.0
$ git pull --rebase
$
$ cd $robots-configuration
$ git fetch origin
$ git checkout v1.13.0
$ git pull --rebase
```

The [sections of our current manual][2] regarding these steps have been updated accordingly.

---
@mbrunettini @julijenv @davidetome take this message into consideration when preparing/deploying/testing code on **community robots**. 

Moreover, code running on **IIT/CRIS robots** `iCubGenova01`, `iCubGenova02`, `iCubGenova04` will stay unchanged (on `master` or `devel` branches) as per their present policies.

cc @drdanz @traversaro

[1]: [https://wiki.icub.org/wiki/Software_Versioning_Table](https://wiki.icub.org/wiki/Software_Versioning_Table)
[2]: [https://wiki.icub.org/wiki/ICub_Software_Installation](https://wiki.icub.org/wiki/ICub_Software_Installation)

davidetome, 2019-10-04:
Ciao @pattacini .

 trying to do the procedure described above, but i get :

[image]

Can u help please?

traversaro, 2019-10-04:
Hi @davidetome ! @pattacini can comment more in detail, but I guess that `git pull --rebase` is not necessary as a command, and it is sufficient to just run 
~~~
git fetch origin
git checkout v3.2.1
~~~
to use the v3.2.1 tag.

pattacini, 2019-10-04:
Yep, thanks @traversaro for pointing this out 👍 
It was a leftover from the previous procedure.

I've made the correction above.

Tobias-Fischer, 2019-10-08:
Hi,
Can I briefly ask what the policy regarding Windows builds is? There have not been any released packages for a long time (Yarp 3.1.0 / iCub 1.12 - i.e. over one year ago). Is it now (unofficial) policy to build from source also on Windows? How are the dependencies treated etc.?

Best, Tobi

pattacini, 2019-10-08:
Hi Tobi, good to hear you 😃 

Regarding Windows, we are currently working out a new solution under the hood that will allow us to provide binaries more frequently as it relies on an automatized toolchain. However, we expect it to be fully operatinal not earlier than the first semester of 2020.

That being said, there's good news anyway since building the source code on Windows is now much easier if you use [`vcpkg`][1].

Just to give you an example, here's attached a [list of the dependencies][2] I can build on my Windows machine using vcpkg in order to bootstrap the ecosystem. You can see how complete it is indeed. There's only one peculiarity related to `ipopt binaries`, which instead are managed through a specific port we crafted that you can find at https://github.com/robotology-dependencies/robotology-vcpkg-binary-ports (instructions are included).

Further, you can also have a glance at our [`Robotology Superbuild`][3] project that aims to help developers build the ecosystem altogether, so to say, in a single step.

cc @traversaro 

[1]: https://github.com/microsoft/vcpkg
[2]: https://github.com/robotology/QA/files/3702208/vcpkg-dependencies.txt
[3]: https://github.com/robotology/robotology-superbuild

traversaro, 2019-10-08:
> Further, you can also have a glance at our Robotology Superbuild project that aims to help developers build the ecosystem altogether, so to say, in a single step.

In particular, we maintain up-to-date instructions on how to install the libraries necessary to compile YARP and ICUB and the rest of the robotology repos on Windows at https://github.com/robotology/robotology-superbuild#system-dependencies-2 . Unfortunately those instructions are quite complex due to the issues in the existing installers, but they should work fine. We used to have instructions on how to use vcpkg, but we removed them earlier this year due to problems with vcpkg's Qt, see https://github.com/robotology/robotology-superbuild/issues/134. As those problems have been solved, we will eventually add back instructions on how to install those dependencies using vcpkg.

Tobias-Fischer, 2019-10-08:
Hi @pattacini @traversaro,
Thanks a lot for the quick and informative responses! The `vcpkg` solution looks great - and we already employ robotology-superbuild on our Linux machines so we will adapt it to Windows, too.

/cc @mazrk7 @umgnunes
In particular, @umgnunes it might be worth setting up a new Windows machine to resolve problems with the speech recognizer.

davidetome, 2019-10-10:
Ciao @pattacini , in the version table I can't find **icub-firmware-build**; should it used in "**master**" branch?

cc @julijenv

pattacini, 2019-10-10:
@davidetome I'm not aware of the status of the firmware binaries. I can see that the last release is `v1.12.0` and not `v1.13.0`. @mbrunettini and @marcoaccame can be of help on this.

pattacini, 2019-10-10:
@davidetome we are going to tag the firmware binaries as well and update the table accordingly. I'll let you know when everything's done. Stay tuned.

pattacini, 2019-10-10:
@marcoaccame has just tagged the firmware binaries and I've updated the table.
@davidetome go with the flow 👍

closed 2020-12-08 by pattacini