robotology/community #366: Simulating the iCub hand and arm

teresalves, 2019-10-23:
Hello,

For my Master Thesis I will be using the iCub hand and arm to grasp objects. I was wondering how I should go about that. The iCub simulator does not simulate the hands, correct? Is there anything in Gazebo I can use? Or what simulator is best?

Thank you,

Teresa Alves

pattacini, 2019-10-26:
Hello @teresalves 

Let me give you a brief overview of the software components you may consider for your thesis work.

For controlling the arm to implement reaching, we have a quite solid and well-established layer that is responsible for doing Inverse Kinematics and trajectory generation. You can browse all the relevant resources at http://www.icub.org/doc/icub-main/icub_cartesian_interface.html. [Gazing][1] is also available. This library is abstract enough to be used in conjunction with whatever simulator we have in our portfolio.

Instead, to judge how well the iCub hand can grasp objects is a matter of which specific simulation environment is used, as per the following considerations. 

### iCub_SIM
Our traditional simulator is still [iCub_SIM][2], which can deal with the hand although it is not properly up to the task in simulating the physics of manipulation. This is essentially due to two reasons:
- the hand itself is not reproduced in all the required details;
- the underlying physical engine does not perform sufficiently well in this respect.

However, you could still employ iCub_SIM depending on the level of accuracy you aim to reach, considering that you can exploit tricks to circumvent the lack of performance in dealing with collisions by for example using the [magnet function][3]. This would be of course valid only if the trick won't affect the core part of your study.

I can provide you with a couple of nice results that built on iCub_SIM to explore the interaction of the hand with the environment:
1. There's a great blog maintained by **Lukáš Zdechovan** on the modeling object grasping with neural networks available at http://masterthesis.zdechovan.com. Click on the following image to watch a video:
  [![modeling-object-grasping][4]][5]
1. The paper ["Catching Objects in Flight"][6] from **Aude Billard**'s group at LASA, whose video is also available on the web:
  [![grabbing-flying-objects][7]][8]

### iCub Simulator in Gazebo
Some time ago we decided to phase out iCub_SIM and switch to Gazebo for a number of reasons including the fact that Gazebo has become somehow the reference in the robotics community.

You can find instructions on how to enable the use of Gazebo within our ecosystem at https://github.com/robotology/gazebo-yarp-plugins.

Although we've obtained great achievements in recent years with Gazebo, especially in the field of dynamic whole-body control - look at the results presented at https://dic.iit.it - we cannot consider the transition completed for what concerns manipulation as well as other devices that have not been ported yet (e.g. the skin).

In particular, all the set of routines that we use for doing grasping (e.g. gazing, reaching, control of the fingers) still need to be properly tuned up. Anyway, @xEnVrE has made recently a great attempt in this direction, providing the following improvements I can point you at if you'd like to give them a shot:
- https://github.com/robotology-playground/visual-tactile-localization-simulation/tree/devel makes available the `.world` with the table and object to play with plus some insights on how to add further objects and a good `yarpmanager` application to start up the components.
- https://github.com/xEnVrE/gazebo-yarp-plugins/tree/visual_tactile_loc_plugins_yarp3 lets gazing work within Gazebo.
- https://github.com/xEnVrE/icub-gazebo/commits/visual_tactile_loc tunes up fingers dynamical parameters to prevent them from moving crazily plus an RSDF file, which is a scriptable SDF where you can set up friction coefficients that are essential to perform grasping correctly.

Quite a big chunk of work that allowed us eventually to get this interesting outcome:

[image]

It will be certainly our job in the upcoming quarters to port @xEnVrE patches into the mainstream.

### Alternatives
Lately, the community in Reinforcement Learning has been focusing on the use of different simulators as [V-REP][9] and [PyBullet][10] that seem very promising too. We're making progress also on this:
- https://github.com/robotology-playground/pybullet-robot-envs is a Python package to develop and test RL algorithms for grasping (@giuliavezzani @eleramp). It's still in Alpha and doesn't have a complete hand yet.
- https://github.com/robotology/gym-ignition creates reproducible robotics environments for research in RL (@diegoferigo) combining OpenAI Gym capabilities within Gazebo.

[1]: http://www.icub.org/doc/icub-main/icub_gaze_interface.html
[2]: https://wiki.icub.org/wiki/Simulator_README 
[3]: https://wiki.icub.org/wiki/Simulator_README#GRAB_FUNCTION_.28MAGNET.29:
[4]: https://user-images.githubusercontent.com/3738070/67617209-1abff200-f7e1-11e9-8415-e19e68aa7ab3.jpg
[5]: https://www.youtube.com/watch?time_continue=6&v=BdDP43v4j1w
[6]: http://lasa.epfl.ch/publications/uploadedFiles/TRO2316022.pdf
[7]: https://user-images.githubusercontent.com/3738070/67617262-adf92780-f7e1-11e9-8cbc-c121a1dfd55f.jpg
[8]: http://lasa.epfl.ch/videos/downloads/robotcatchingsim.mp4
[9]: http://www.coppeliarobotics.com
[10]: https://pypi.org/project/pybullet

teresalves, 2019-10-27:
Thank you so much for all the information! 
This will certainly be very helpful for my research.

Kind Regards,
Teresa

plinioMoreno, 2020-02-18:
Hi @pattacini and @xEnVrE,

Many thanks for the detailed answer. @teresalves wants to use the tactile sensors on the iCub hand as main observation to infer which kind of object is (we will start with simple objects like cylinders, parallelograms, spheres), based on the hand pose and the tactile information. In addition, the objects at our initial tests can be attached to the ground.

Our main question is if considering the developments by @xEnVrE , we can move the hand to a set of predefined points, close the hand until the tactile sensors reach a predefined threshold and store the data samples.

Looking forward to your answer.

pattacini, 2020-02-18:
Hi @plinioMoreno 

I would say definitely yes given the work @xEnVrE did in this respect, but @xEnVrE can say the last word on it.

Let me only add to the discussion that you may find this article interesting: https://ieeexplore.ieee.org/document/7803305.

cc @lornat75

xEnVrE, 2020-02-18:
Hi @plinioMoreno 

I would also say that the experiment is doable in a simulated setting like the one I was working on. 

However, let me add some considerations:

> infer which kind of object is (we will start with simple objects like cylinders, parallelograms, spheres), based on the hand pose and the tactile information

while the hand pose is available within the simulated environment, the tactile information (implemented in one of the plugins that I added) only consists of a list of the 3D points on the fingertips that are in contact with the environment. Hence, this information can be used to detect contacts and find their locations. No information on the exchanged forces is provided.

>  the objects at our initial tests can be attached to the ground

this hypothesis can be easily enforced within Gazebo by using a dummy joint between the object and the ground. This will help you a lot since some objects tend to move in the simulation environment, due to the exchanged contacts with the environment, even if no other external agents move them. Of course, in case you need unconstrained objects in future steps of the project, you can alleviate the phenomenon by playing with the weight of the object and/or the friction between the object and the ground.

> we can move the hand to a set of predefined points

this is already working but some effort might be needed to ensure that the desired hand pose is actually reached - especially if you need enough precision to ensure that the finger will touch the object. If not, this would require to tune a bit the low-level PID gains of the simulated robot.

>  close the hand until the tactile sensors reach a predefined threshold 

as I said before, right now the plugins implementing the tactile information only provides contact locations. However, you can either stop the motion as soon as a contact is detected or implement also some kind of force sensing within the plugin. In the second case, I can provide further explanations on the code of the plugin.

Please let me know in case you need further information.

plinioMoreno, 2020-02-19:
Many thanks @pattacini and @xEnVrE ! @teresalves will set up this and we will let you know.

closed 2020-12-08 by pattacini