robotology/community #374: Cannot Setup ICUBcontrib

pattacini, 2019-12-05:
@YANGGANG1971 aked the following in https://github.com/vvv-school/vvv18/issues/62:

## Brief description
When trying to set up the[ ICUBcontrib](http://wiki.icub.org/wiki/Linux:Installation_from_sources), the reminders said I need to change CMAKE_INSTALL_PREFIX entry here to the path of your liking, preferably in your home and set up the environment.
But I cannot change the CMAKE_INSTALL_PREFIX entry and it does not work as follows.
[image]
Also, while setting up the new environment, ICUBcontrib_DIR, CMAKE_PREFIX_PATH, YARP_DATA_DIRS cannot be found in the setting panel as follows.
[image]

## Detailed context

## Accompanying material

pattacini, 2019-12-05:
@YANGGANG1971 honestly it's very unclear what your problem is about. Try to better describe all the context.

YANGGANG1971, 2019-12-06:
The main problem is that I cannot set up ICUBcontrib correctly followed by the links.
http://wiki.icub.org/wiki/Linux:Installation_from_sources
1. The tutorial said that I must change CMAKE_INSTALL_PREFIX entry. I tried and but failed (cannot save these settings). Or is any special requirement for this CMAKE_INSTALL_PREFIX entry? 
 
[image]

2. I am not sure whether I need to configure the environment. 

### Configure your environment
If you installed YARP and/or iCub or ICUBcontrib to non-default directories, you need to set your environment.

> Set the ICUBcontrib_DIR env var to point to the installation directory.
> Append $ICUBcontrib_DIR/bin to your PATH.
> Set the CMAKE_PREFIX_PATH env var to point to the installation directory $ICUBcontrib_DIR.
> Modify the variable YARP_DATA_DIRS accordingly, i.e.:

` YARP_DATA_DIRS=<YARP_INSTALL_PREFIX>/share/yarp:<ICUB_INSTALL_PREFIX>/share/iCub:<ICUBCONTRIB_INSTALL_PREFIX>/share/ICUBcontrib`

While configuring the environment, I cannot find the following variables (ICUBcontrib_DIR, CMAKE_PREFIX_PATH, YARP_DATA_DIRS). 
[image]

I cannot set up the configuration correctly. I must type sudo command line in the terminal as the prefix each time.
e.g.
`sudo yarpview ....`

pattacini, 2019-12-08:
1. Since `YARP` gets found correctly :+1: make sure that:
    1. the path `/home/icub/robot-code` does exist;
    1. press the key `c` to let CMake configure the project first;
    1. the option `g` should bring up soon afterward; press `g` then.
    1. run `make install`.
1. You might have installed `YARP` with root privilege (e.g. `sudo make install`), hence the need for `sudo`. You could try to revert the installation and repeat it without `sudo`.

YANGGANG1971, 2019-12-09:
I made a recording video to show the installation error as follows.
https://drive.google.com/open?id=10gs83Wm0qU2zNzul-0u0bP7bMXgfK7w9

pattacini, 2019-12-10:
Hi @YANGGANG1971 

I might have missed the point of the video and I'm not sure about what kind of message you meant to convey. Maybe you're not familiar with the use of `ccmake`, so why don't you try the following instead?

```sh
$ export ICUBcontrib_DIR=/home/icub/robot-code/iCubContrib
$ mkdir -p ${ICUBcontrib_DIR}
$ cd /home/icub/robot-code/icub-contrib-common
$ mkdir -p build && cd build
$ cmake -DCMAKE_INSTALL_PREFIX=${ICUBcontrib_DIR} ../
$ make install
```

YANGGANG1971, 2019-12-11:
Yes, I am not familiar with the use of ccmake. Following your suggestion, the last command line still ran into problems. 
$ make install
[image]

YANGGANG1971, 2019-12-11:
Great! It works!

closed 2020-12-15 by pattacini