robotology/community #375: BufferedPortImageRgb image read put system on hang problem using Matlab

smwasif, 2019-12-09:
Hi. 
    I have opened `BufferedPortImageRgb` in Matlab using `yarpImage=port.read;`, the system hangs to this `port.read` call and return nothing.  I opened open only once but before read I closed this port to avoid system hang but still I am getting same problem.
Please to resolve this issue.

traversaro, 2019-12-09:
Hi @smwasif, by default the [`yarp::os::BufferedPort<T>::read`](https://www.yarp.it/classyarp_1_1os_1_1BufferedPort.html#ad88d8d901b29e3776281b943a4f47596) is blocking, i.e. it "hangs" until a message is received. If you want to change its behaviour, pass a boolean parameter set to `false` to it, see the related YARP documentation for more info: https://www.yarp.it/classyarp_1_1os_1_1BufferedPort.html#ad88d8d901b29e3776281b943a4f47596 .

smwasif, 2019-12-09:
Hi traversaro,
                       Thank you very much for your kind reply. I tried port read with false parameter pass using this command `yarpImage=portLeft.read(0);` but it is resulting null image. The complete code which I wrote for this test is:
`yarp.matlab.LoadYarp;
import yarp.BufferedPortImageRgb
import yarp.BufferedPortBottle;
portLeft=BufferedPortImageRgb;
% disp('Registering port /matlab/LeftEye');
portLeft.close; % make sure the port is closed, calling open twice hangs 
portLeft.open('/matlab/LeftEye');
% disp('Please connect the port /matlab/LeftEye to an image source');
yarp.Network.connect('/icubSim/cam/left/fovea','/matlab/LeftEye');
yarpImage=0;
yarpImage=portLeft.read(0);`

The result of this code is `yarpImage=[]`. Please guide me what is wrong here.

lornat75, 2019-12-09:
I don't clearly see why this code fails. Unless there are better ideas, I would propose these suggestions:

- copy/translate the code to C++ and see it it works, there could be issues external to the code 
- remove the close before the open, not sure how much this has been tested and may be causing problems
- I am unfamiliar with the Matlab YARP API, but I would double check that the return type from a BufferedPort (portleft.read()), this in C++ returns a pointer and I have no idea how this is mapped in Matlab
- remove the explicit `connect` and do it from a terminal i.e. `yarp connect /icubSim/cam/left/fovea /matlab/LeftEye'

none if the above is a solution, but it may provide some hints for debugging

vtikha, 2019-12-09:
Dear @smwasif ,

Could you please try [this](https://github.com/robotology/yarp/blob/master/bindings/java/examples/matlab/FastImageConversion.m) example on your system and see if there are any differences? 

This is basically an example that replicates the behaviour of a ```YARP module``` in Matlab. To quit the process you need to connect a ```write port``` to the following ```/matlab/read``` and send a ```quit``` message. 

```bash
> yarp write ... /matlab/read
> quit
```

What it does is pretty simple: 
 - receives a YARP image in ```/matlab/img:i```
 - converts it to Matlab format
 - re-converts it back to YARP and sends it on ```/matlab/img:o```

Just as a test, please connect ```/icubSim/cam/left``` instead of ```fovea```.

smwasif, 2019-12-10:
Dear @vtikha 
                      Thank you very much for so kind direction. I tested your referred example with terminal connection of yarp ports, and it worked smoothly. I further tested same my written code in my last message with explicit connection from the Matlab with modification of left eye image instead of fovea and it worked. It shows that the problem is with foveal image which is returning a null image with same set of commands. The modified test code which is working is:
  `yarp.matlab.LoadYarp;
import yarp.BufferedPortImageRgb
import yarp.BufferedPortBottle;
portLeft=BufferedPortImageRgb;
% disp('Registering port /matlab/LeftEye');
portLeft.close; % make sure the port is closed, calling open twice hangs 
portLeft.open('/matlab/LeftEye:i');
% disp('Please connect the port /matlab/LeftEye to an image source');
yarp.Network.connect('/icubSim/cam/left','/matlab/LeftEye:i');
yarpImage=0;
yarpImage=portLeft.read(false);`

smwasif, 2019-12-10:
Dear @vtikha
Thank you very much for so kind direction. I tested your referred example with terminal connection of yarp ports, and it worked smoothly. I further tested same my written code in my last message with explicit connection from the Matlab with modification of left eye image instead of fovea and it worked. It shows that the problem is with foveal image which is returning a null image with same set of commands. The modified test code which is working is:
`yarp.matlab.LoadYarp; import yarp.BufferedPortImageRgb import yarp.BufferedPortBottle; portLeft=BufferedPortImageRgb; % disp('Registering port /matlab/LeftEye'); portLeft.close; % make sure the port is closed, calling open twice hangs portLeft.open('/matlab/LeftEye:i'); % disp('Please connect the port /matlab/LeftEye to an image source'); yarp.Network.connect('/icubSim/cam/left','/matlab/LeftEye:i'); yarpImage=0; yarpImage=portLeft.read(false);`

Please resolve issue with foveal image and share any such example if it works with foveal image.

smwasif, 2019-12-12:
Dear,
         This is to further inform that only Cartesian image can only be obtained from Matlab with above given code.  The above mentioned problem of null image acquisition exits for both Logpolar and Fovea images of iCub.
Please give some from your precious time to address this issue. Waiting for your kind reply.

pattacini, 2019-12-12:
Hi @smwasif 

What is the piece of code responsible for publishing logpolar and/or fovea images on the ports you're attempting to connect to?

Perhaps this one: https://github.com/robotology/logpolar ?

smwasif, 2019-12-12:
Deat @pattacini ,
                                Thank you very much. I just edited only one line in the test code, otherwise it is same as given above, and which  is:
`yarp.matlab.LoadYarp;
import yarp.BufferedPortImageRgb;
import yarp.BufferedPortBottle;
portLeft=BufferedPortImageRgb;
% disp('Registering port /matlab/LeftEye');
portLeft.close;
% make sure the port is closed, calling open twice hangs
portLeft.open('/matlab/LeftEye:i');
% disp('Please connect the port /matlab/LeftEye to an image source');
yarp.Network.connect('/icubSim/cam/left/**logpolar'**,'/matlab/LeftEye:i');
yarpImage=0;
yarpImage=portLeft.read(false);`

smwasif, 2019-12-12:
Dear @pattacini ,
Thank you very much. I just edited only one line in the test code, otherwise it is same as given above, and which is:
`yarp.matlab.LoadYarp; import yarp.BufferedPortImageRgb; import yarp.BufferedPortBottle; portLeft=BufferedPortImageRgb; % disp('Registering port /matlab/LeftEye'); portLeft.close; % make sure the port is closed, calling open twice hangs portLeft.open('/matlab/LeftEye:i'); % disp('Please connect the port /matlab/LeftEye to an image source'); yarp.Network.connect('/icubSim/cam/left/**logpolar'**,'/matlab/LeftEye:i'); yarpImage=0; yarpImage=portLeft.read(false);`

vtikha, 2019-12-12:
Hi @smwasif 

I believe that the ```logpolar``` and the ```fovea``` additions of the camera ports of the simulator were added for an EU project a very long time ago and are no longer supported. Consider that on the physical robot there is only one version of the images coming out of each cameras which is: ```/icub/cam/left``` and ```/icub/cam/right```. Therefore please consider only as **standard** these ports and if you require ```fovea``` or ```logpolar``` images you will have to generate them yourself (from the original stream of images) using off-the-shelf algorithms.

smwasif, 2019-12-12:
Dear @vtikha,
                          Thank you very much for this precious guide. I shall now use only standard images. As in past I used foveal images just for the sake of less computational cost otherwise the Cartesian images can also be used.

closed 2020-12-15 by pattacini