robotology/community #376: 🏫 VVV20 − Third International Winter School on Humanoid Robot Programming 🤖

pattacini, 2019-12-13:
Hi @robotology/everyone 👋 

The **Open Call** to attend the 3rd edition of our **VVV School** is now available!

🌐 Check it out at http://www.icub.org/phdschool.

### Important dates

- Application deadline: **January 8**.
- Notification of acceptance: **January 15**.

The school begins on the morning of Tuesday 10th and finishes on the evening of the 19th.
Students are expected to check in on Monday 9th and check out on Friday 20th. Sunday 15th will be free.

[image]

[image]

paolo-viceconte, 2020-01-08:
Hi,

I am trying to complete my application for the VVV20 (previously saved in
order to update my CV) but I am currently facing the error "No more choices
left" at the selection of the "Main Research Area" in page 2. How can I
complete the application?

Thank you for your help.

Best,
Paolo Maria Viceconte

ChiaraBartolozzi, 2020-01-08:
Dear @paolo-viceconte 

I will look into this issue and hope to solve it as soon as possible. Should we fail in solving the problem by today, we will extend the deadline.
I'll update you soon.

Best
Chiara

paolo-viceconte, 2020-01-08:
Dear Chiara,

thank you for your quick answer.

I will try again later or, if the issue is still there, tomorrow.

Best regards,
Paolo Maria Viceconte

paolo-viceconte, 2020-01-08:
Dear Chiara,

the issue seems to be solved now, thank you again.

Best regards,
Paolo Maria Viceconte

HosameldinMohamed, 2020-01-08:
Hi,

I have submitted an application when there was only available "Speech" for "Main Research Area", should I resubmit a new application to change it? My research area is in state estimation so I guess the closest in the new list is "Control".
Thanks.
Hosam

ChiaraBartolozzi, 2020-01-08:
Hi @HosameldinMohamed 
no worries, the application is sort of painful, I will change it in your application.
Best
Chiara

HosameldinMohamed, 2020-01-08:
> Hi @HosameldinMohamed
> no worries, the application is sort of painful, I will change it in your application.
> Best
> Chiara

Thank you very much @ChiaraBartolozzi !

closed 2020-12-12 by pattacini