robotology/community #377: 🚀 iCub Distro 2019.12.feat-01 Released

pattacini, 2019-12-16:
Hi @robotology/everyone :wave:

### :rocket: TL;DR
We're happy to announce that our latest iCub software distro **2019.12.feat-01** is now available.

To download the source code with the correct tags, check out our up-to-date [Software Version Table][1].

:warning: Jump to the section **Prerequisites** further down the page to see whether your robot is eligible for this upgrade.

### :book: SW Distros Overview
We have changed our flow to deliver iCub software bundles that comprise several repositories with matching tags (e.g. [`yarp`][2], [`icub-main`][3], [`icub-basic-demos`][4]...).

#### :calendar: Time-based Distros
Starting today, these bundles will follow a **time-based release cycle**. As a result, bundles will be termed as **YYYY.MM Distro** to indicate the year and the month of the release.

Each year, we aim to yield **4 official distros**, which will be synchronized with the outcome of our internal development stages. Thereby, we will have YYYY.02, YYYY.05, YYYY.08, YYYY.11 distros. Patches will be coded as **YYYY.MM.nn**.

These official distros will be shipped along with **binaries**.

#### :sparkles: Feature-based Distros
Occasionally, we may distribute software containing special features that we have to release out of the standard cycle. These events will most likely occur in conjunction with the release of a new iCub robot.

The **feature-based distros** will be termed **YYYY.MM.feat-nn** and, generally, will not be accompanied by binaries.

#### :point_right: YARP
Within this release flow, YARP is handled as a dependency for all the iCub repositories in the bundle since we clearly specify the YARP tag/branch each distro will rely on. Thus, YARP will continue to follow its own release cycle.

### :sparkles: 2019.12.feat-01 Distro
Upon releasing the new **iCub S/N 040 − iCubGenova07** robot, we have made available the 2019.12.feat-01 distro containing the handling of the new **Robot Face Expression** (RFE) unit, whose relevant activities have been tracked in the following issues/prs:
- https://github.com/robotology/icub-main/issues/570
- https://github.com/robotology/icub-main/pull/592
- https://github.com/robotology/icub-main/issues/595
- https://github.com/robotology/icub-main/pull/622

#### :warning: Prerequisites
Importantly, in order to build and run the distro correctly, pay attention to the notes below:
- Make sure that **CMake 3.12** is available on your system as it is required by the latest YARP releases. You can find the instructions to install CMake 3.12 on Ubuntu 18.04 at https://github.com/robotology/QA/issues/364.
- Given that [`robots-configuration`][5] contains changes in the XML files related to the motion control of **ETH robots**, we have updated the version of the code responsible for managing this as well as the firmware of the embedded boards. In particular, those robots with  `GENERAL::MotioncontrolVersion` equal to `5` have been now upgraded to version `6`. If you are a user of the following robots, please proceed with a complete **FW update**:

  | :robot: ETH robot | :label: Motion Control Version | Eligibility for Upgrade |
  | :--- | :---: | :---: |
  | `iCubErzelli01` | `6` | ✔ |
  | `iCubErzelli02` | `6` | ✔ | 
  | `iCubErzelli03` | `6` | ✔ |
  | `iCubGenova02` | `6` | ✔ |
  | `iCubGenova04` | `6` | ✔ |
  | `iCubGenova07` | `6` | ✔ |
  | `iCubMoscow01` | `6` | ✔ |
  | `iCubShenzhen01` | `6` | ✔ |
  | `iTeenGenova01` | `6` | ✔ |
  | `naoPrague01` | `-` | ✔ |
  | `iiwaBoston01` | `-` | ✔ |

- Instead, the **ETH robots** listed in the table below are still configured with a quite old Motion Control Version, hence **they cannot run with 2019.12.feat-01 distro**. It will be certainly our responsibility to align these robots to the latest versions in order to make them eligible for the next upgrade, which is expected to come soon. Stay tuned!

  | :robot: ETH robot | :label: Motion Control Version | Eligibility for Upgrade |
  | :--- | :---: | :---: |
  | `iCubDarmstadt01` | `4` | ❌ |
  | `iCubEdinburgh01` | `4` | ❌ |
  | `iCubHeidelberg01` | `4` | ❌ |
  | `iCubLausanne02` | `4` | ❌ |
  | `iCubNancy01` | `4` | ❌ |
  | `iCubNottingham01` | `4` | ❌ |
  | `iCubSheffield01` | `4` | ❌ |
  | `iCubSingapore01` | `4` | ❌ |
  | `iCubTwente01` | `4` | ❌ |
  | `iCubHamburg01` | `3` | ❌ |

- **CAN robots** are not involved by such FW change and thus their ecosystem can be upgraded smoothly :+1:

[1]: https://wiki.icub.org/wiki/Software_Versioning_Table
[2]: https://github.com/robotology/yarp
[3]: https://github.com/robotology/icub-main
[4]: https://github.com/robotology/icub-basic-demos
[5]: https://github.com/robotology/robots-configuration

GiulioRomualdi, 2019-12-16:
@robotology/iit-dic

pattacini, 2019-12-18:
Thanks to @dcam0050 (see https://github.com/robotology/icub-main/issues/631), we have fixed a regression bug related to the compilation of a deprecated device (very old, actually).

This led us to yield the new patch `2019.12.feat-02` pointing to `icub-main/v1.14.1` as per in our [Versioning Table][1].

[1]: https://wiki.icub.org/wiki/Software_Versioning_Table

serena-ivaldi, 2020-01-09:
Let us know when an update for iCubNancy01 is available!

closed 2020-12-12 by pattacini