robotology/community #378: Newly acquired iCub. Power Setting and etc.

iPsych, 2020-01-04:
## Brief description
Q1. Do I need any special cable or interface board between PSU and Robot?
Q2. Is there any 'upgrade' from pc104 to Jetson or some newer system?
Q3. Is it possible to build the main 'Server' in MacOS?

## Detailed context
Recently acquired iCub through government equipment replacement program.
I have iCub main Robot, and EA8080 manufactured by Elektro-automatik.
Do I need special board or connection between PSU and the Robot?

Due to the transport was placed by not previous user, but the government and managers, I didn't get the full setting. I received the Main Robot, and had to buy the EA8080 separately.
The iCub is sleeping sadly.

pattacini, 2020-01-04:
Hi @iPsych 👋 
Welcome to **Robotology**! (You should have received the email invitation to join us)

It's very unfortunate that you didn't get the whole equipment! Anyway, let's try to go through your points.

### ❔ Q1
@Fabrizio69 @Fabrizio69 could you please address this?
By the way, what is the "nickname" of your robot? The convention is `iCub<City><Num>`.

### ❔ Q2
Actually, the name `pc104` is outdated as the latest iCub's do have i7-based boards in the head, which are quite powerful.

We do not have any plan at the moment to "upgrade" or replace the onboard resources but this is not a hitch as the philosophy behind our approach is that the iCub represents a node of the cluster of machines (Fig 1) each user can build around the robot to fulfill his own specific requirements. This way, you can attach to the 1 Gb/s network a computer equipped with GPU to do AI processing for example.

The so-called "plumbing" among all the machines of the cluster is taken care of using our middleware [YARP][1], which is responsible for the inter-process communication as well as for facilitating the orchestration of complex architectures (Fig 2) made of many modules running on a distributed system.

| |
| :---: |
| ![cluster][2] |
| Fig 1. Cluster | 
| ![architecture][3] |
| Fig 2. Architecture | 

### ❔ Q3
I think that setting up a MacOS system to make it play the role of the iCub server should be doable, provided that you know how to generate a LAN where to connect the robot to. I'm not a MacOS user, but maybe @vtikha @traversaro @mbrunettini could chime in and give you some heads-up if needed.

Also, our server laptop usually exports a disk space as a shared resource. However, in the minimal configuration, you could certainly get around this and just plug your machine.

Hope this helps for now while waiting for information regarding **Q1**.
Don't hesitate to get back to us if you have further questions or need more clarification.

[1]: https://yarp.it
[2]: https://user-images.githubusercontent.com/3738070/71764495-c758d700-2ee8-11ea-845b-488652f49f6b.png
[3]: https://user-images.githubusercontent.com/3738070/71764468-895bb300-2ee8-11ea-821b-504c0ba943ee.png

iPsych, 2020-01-05:
Hi, @iPsych
Thank you for kind welcome : )
❔ Q1
The nickname should be re-given, since the iCub moved from Daejun to Seoul : )
Do you need the serial #?

❔ Q2
Thanks for this clear diagram! I fully understand, and figured out where I can put the 'intensive' computation load.

❔ Q3
Sadly, the HDD in the laptop came with the icub is broken, so I removed it now.
Is there any tutorial or disk-image to install the iCub server? I have checked below, and successfully installed Yarp, and could simulate icub in MacOS Catalina : )
> http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions
If the power issue is addressed, I can test the possibility of MacOS server (possibly Mac Mini below icub as server).

pattacini, 2020-01-06:
@iPsych 

Just created https://github.com/robotology/robots-configuration/pull/147 to address the city change 😉 Moreover, the S/N is not required as of now, unless anyone from our robot techies will ask for it.

http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions is actually the recipe to yield a new server laptop. Don't know whether it can be extended to cover the MacOS case, though.
@mbrunettini any thought?

mbrunettini, 2020-01-07:
Unfortunately, MacOS is not supported to cover the role of iCub Server. There afre some configurations (eg. gateway and NAT) that we do not support on MacOS

iPsych, 2020-01-07:
@mbrunettini  Thank you for reply. So, I can use YARP in MacOS installed computer as a kind of 'terminal', but the 1st computer connected directly to iCub should be a linux machine, right?
Which os is latest 'stable' to follow the instruction @pattacini commented?

iPsych, 2020-01-07:
@Fabrizio69, @mbrunettini
Would you let me know what should be 'installed' or 'conncected' between iCub and PSU(EA8080), please? Should I order some board? 
Also, is there any printable 3D model file of iCub Face? Since the white face goes yellow, I hope to re-print with new material and try to some experiment like [Prof. Metta did ](https://www.researchgate.net/publication/281185973_An_articulated_talking_face_for_the_iCub/references)
(

mbrunettini, 2020-01-07:
There must be a linux machine working as server, while the YARP software and your code must run on different  machines with the OS you like among supported ones (Ubuntu Linux, Debian Linux, MacOS and Windows). You can find instructions for a so called "generic iCub machine" on [iCub Wiki](http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions): instructions details are for Linux, but since the required steps are essentially IP address, NFS client and user environment setup, they can be easily adapted for MacOS.
The iCub Server can run Latest stable releases of Ubuntu or Debian linux.

Regarding other questions, like PSU and 3D face model, I am not the right person to refer to, sorry.

pattacini, 2020-01-07:
Thanks @mbrunettini 👍 

@iPsych I've added up Q4 to the list for the 3D model of the iCub face.
Oustanding points are concerned with Q1 and Q4.

Let me grab the reference people for Q1 and Q4.

Fabrizio69, 2020-01-07:
Hi @iPsych,
Regarding the **Q1** question it's necessary use a custom cable to connect the robot to the two power supplies EA-PS 8080-60 2U. Can you confirm that you have two power supplies?
The external power cable is produced by us, for a commercial quotation you can contact my colleague Andrea Derito (andrea.derito@iit.it)

iPsych, 2020-01-07:
@Fabrizio69 
Oops, do I need two EA-PS8080 units?

pattacini, 2020-01-07:
> Oops, do I need two EA-PS8080 units?

Yep, your iCub does need both the 12 V and 48 V power supply.

marcellosavoldi, 2020-01-07:
Hi @iPsych  ,and welcome,
I can try to resolve your " Q4" problem about the CAD for the robot face.

First, we use a PTC CAD to model the mechanical parts of iCub : Creo
Are you able to open this file? 
In this case you can find the face part file "rc_ist_009_p_077" at the following link:
https://github.com/icub-tech-iit/cad-mechanics-public/blob/master/projects/icub/icub1_icub2/

Second, is very important that use the right material for the face part, its must be semitrasparent to permise to allow at the light of the LED to create a good facial expression.
We identify the semitrasparent white as "semitrasparent-opalino white polished" 
I dont think you can replicate it using 3d print technology, we mould it using a silicone stamp.

[image]

Third, There are 3 supports to glued on the internal surface of the face to correctly assemble the boards for the facial expressions, It's no easy to remove them, I think you need to create the 3 parts too:
RC_IST_009_P_081_09
RC_IST_009_P_082_09
RC_IST_009_P_080_09
material: generic PA-white ( or a white PLA 3D printing part is good)

[image]

Also the eyelids support and the motor supports are glued , and they have to remove ( it's of ergal material, could be easy remove them, probably at the expense of the plastic part)

[image]

Please, contact me for any doubt or details.

iPsych, 2020-01-18:
@marcellosavoldi 
Thanks for detailed explanation.
I tried to open .asm file, but it cannot be opened in Creo Parametric.
Since the file size is really small, I am worry that the upload might be incomplete.
Also can I open them in AutoCad?

traversaro, 2020-01-18:
Hi @iPsych, the `.asm` file is an assembly file that cannot be used on its own, as it references all the other parts that it contains in the form of other `.asm` files and of `.prt` files. 
To open those files, you need to close all the repository `cad-mechanics-public`, in particular making sure that you use the Git LFS option (otherwise all files will be really small and unusable) as explained in `cad-mechanics-public` README: https://github.com/icub-tech-iit/cad-mechanics-public/blob/master/README.md .