robotology/community #379: Request tutorials about yarpmanager

YANGGANG1971, 2020-01-09:
On a physical robot, we used the yarpmanager to manage the files below. E.g. I want to find the original C++ files (uncompiled), what should I do?
 
[image]

In addition, I noted that on the right of the figure, there is an option called Parameters as follows. It is strange that on my computer, I cannot find the file called dragonfly2_config_left_bayer_640_480.ini and even cannot find the folder named camera. 
Where are these files? 
 
[image]

In addition, I try to use the terminal to configure settings. Type on the terminal.
$ yarpview --from  camera/ dragonfly2_config_left_bayer_640_480.ini
but failed as follow. How do I configure these settings if I want to use the terminal?

[image]

Nicogene, 2020-01-09:
Hi @YANGGANG1971 ,

 `dragonfly2_config_left_bayer_640_480.ini`  and `ServerGrabberDualDragon,ini` are configuration files for running the dragonfly cameras of `iCub`.

In order to have it available, you have to clone and install `robots-configuration`. 
But the problem that I see here is that you are trying to launch the device from the laptop, it should be launched from the `icub-head` or `pc104`, because the dragonfly cameras are attached to the head of the robot.
In fact in `yarpmanager` application you can see that the "host" entry is `pc104`.

YANGGANG1971, 2020-01-09:
My general question is how to find the original codes (uncompiled) through yarpmanager. 
To control the physical robot, some demos have been compiled and installed, shown above. 
I wanna know the detailed codes and change these demos for my own project.

YANGGANG1971, 2020-01-09:
Also, learn to configure the settings using yarpmanager.

Nicogene, 2020-01-10:
> My general question is how to find the original codes (uncompiled) through yarpmanager.

I don't know if I understood your question but you can't reach the source code through yarpmanager.

If you tell me which executable want to modify I can point you where it is the corresponding source code.
> Also, learn to configure the settings using yarpmanager.

You can find [here](http://yarp.it/yarpmanager.html) some information how to use `yarpmanager` and how it works. This documentation page is quite old, if it missing some information you need, let us know.

YANGGANG1971, 2020-01-15:
I want to know that the tutorials about demoRedBall. 
(https://github.com/robotology/icub-basic-demos/tree/master/demoRedBall)
It is better to be close to this kind of tutorial below and it is easy to be understood. (http://www.icub.org/software_documentation/icub_hello_icub.html)

pattacini, 2020-01-15:
Hi @YANGGANG1971
 
We don't have a tutorial on `demoRedBall` unfortunately and we don't have time right now to write it up.

YANGGANG1971, 2020-01-15:
Got it. Thank you anyway.
In this yarpmanager tutorial (https://www.yarp.it/yarpmanager.html)
$ cd <path to YARP local repository>/data/yarpmanager/tests 
$ yarpmanager-console 
>> help 
     # a list of yarpmanager-console keywords 
>> list app 
     # you should see EyesViewer-Localhost, FakeEyes, ... 
>> load app EyesViewer-Localhost 
>> run 
>> connect 
     # you should see moving red bar 
While connecting, errors occurred. 

[image]

Nicogene, 2020-01-15:
You should type `run` followed by the ID of the application.
In this example it should be:

```
>> run  EyesViewer-Localhost
```

Nicogene, 2020-01-21:
@YANGGANG1971 if you have not further need of help I will close this issue.

closed 2020-12-15 by pattacini