robotology/community #384: 6DoF Pose Visualisation

arrenglover, 2020-01-28:
## Brief description
Is there any pre-existing tools to quickly/easily visualise a single (or many) 6-DoF poses as defined by <x, y, z, roll, pitch, yaw> or other representation.

## Detailed context
We have some pose estimates coming from the Vicon, which we want to compare to some estimates from visual/IMU estimation. We want to visualise the two to compare how well our estimates are as our camera moves. If there is already a solution in the yarp/icub environment I would want to use that.

I've been told that I can use the YARP-ROS bridge and publish the orientation estimates through ROS PoseStamped messages, then to use this rqt plugin to visualize rotations http://wiki.ros.org/rqt_pose_view . However, I was hoping to avoid installing ROS etc. 

I don't know so well the:
Rviz plug-in? iCubGUI?

Thanks!

@simbamford 
## Accompanying material

pattacini, 2020-01-28:
Hi @arrenglover 

As far as I know, the easiest solution would be relying on Rviz, hence ROS.
I don't think we have other tools ready to be used in this respect. I might be mistaken, but `iCubGui` is not up to the task of displaying reference frames within the simulated world.

If you don't like introducing yet another middleware but think that using a library is still doable (it's a dependency anyway), then my recommendation goes in favor of [VTK][1], which is capable of reproducing professional results with only a little effort.

I have quite an experience with VTK and I'm very happy about it. If you plan to give it a shot I could provide heads-up.

Consider that learning how to use VTK is definitely a plus for future developments, thus it is certainly a good way of investing part of your time 😉 

[1]: https://vtk.org

arrenglover, 2020-01-29:
Thanks @pattacini as discussed, I'll give VTK a go

closed 2020-12-08 by pattacini