robotology/community #387: Difference of class between PoriodicThread and RFModule

YANGGANG1971, 2020-03-03:
I noticed that the classes of both PoriodicThread and RFModule are used for the initializing configuration, setting period, updating module and releasing/closing. I want to know the difference between them.

lornat75, 2020-03-03:
RFModule is not spawning any new thread, it is called by the main thread itself. PeriodicThread spawns a new thread, so if you create one in the main you end up with two threads. In general RFModule is a way to encapsulate the activities of the main thread, which often involve monitoring an input port that control the your executable lifecycle, change parameters etc. In addition you have the chance to perform activities from within the update function. 

If you need to create additional threads you can create and start one or more PeriodicThread(s) and monitor them from the RFModule update function (start/stop/susped them, change parameters etc).

Hope this helps.

drdanz, 2020-03-04:
Just for reference, `RFModule` can also be executed in a separated thread calling [`runModuleThreaded()`](https://yarp.eu/classyarp_1_1os_1_1RFModule.html#a9c3ff595d8113c6b3ed47f0b5a8616c8)

YANGGANG1971, 2020-03-05:
Thank you very much.

closed 2020-12-15 by pattacini