robotology/community #388: <gazecontrollerclient> was found but could not open and unble to open the gaze controller

YANGGANG1971, 2020-03-05:
In iCub_SIM, the snippet of codes can be run correctly. 

>   Property optGaze;
>         optGaze.put("device","gazecontrollerclient");
>         optGaze.put("remote","/iKinGazeCtrl");
>         optGaze.put("local","/tracker/gaze");
>         if (!clientGaze.open(optGaze))
>         {
>             yError()<<"Unable to open the Gaze Controller";
>             return false;
>         }

But, for the physical robot (not a simulator), we ran into the problem follow. 
In prerequisite, we have opened iKinGazeCtrl . 

[image]

What kind of settings should I check for the physical robot?

The entire codes can refer to https://github.com/vvv-school/tutorial_gaze-interface

pattacini, 2020-03-05:
Hi @YANGGANG1971 

Try the following:
- Launch only `iKinGazeCtrl` on the real robot by doing:
  ```sh
  $ iKinGazeCtrl 2>&1 | tee output.log
  ```
- Can you attach here a zip containing the file `output.log`?
- Open up a shell where to type in the commands below:
  ```sh
  $ yarp rpc /iKinGazeCtrl/rpc
  >> get info
  ```
  What's the output?
- Continue issueing:
  ```sh
  >> look 3D (-0.5 0.0 0.36)
  >> look 3D (-0.5 0.2 0.36)
  >> look 3D (-0.5 -0.2 0.36)
  >> look 3D (-0.5 0.0 0.36)
  ```
  Is the head moving?

YANGGANG1971, 2020-03-06:
1. output.log as follows

[output.zip](https://github.com/robotology/QA/files/4297150/output.zip)

2. The head can move and the shell is shown as follows. 

[image]

pattacini, 2020-03-06:
Thanks @YANGGANG1971 for the test!

The results show that `iKinGazeCtrl` operates correctly ✔️ 
Therefore, the problem must be either on the client-side or in the way the client tries to talk to `iKinGazeCtrl`.

I've seen that you launched the module `gaze_python-tracker` as the client. Would you be willing to upload it as a [`Gist`][1] for inspection?

### Side note
I've realized from the output of `get info` that the recognized HW version is `1.0`, hence quite old. Which robot is yours?

Asking because, from the log, it appears that `iKinGazeCtrl` relies on the following configuration files:
 - `/home/icub/.local/share/yarp/contexts/iKinGazeCtrl/config.ini`

In turn, this conf file is NOT located in the robot directories and that's quite weird if we consider that you're doing these experiments on the iCub server machine (i.e. `icubsrv`).

Do you have the following installed/configured on your `icubsrv`?
1. The env variable `YARP_ROBOT_NAME`: run `echo $YARP_ROBOT_NAME`. 
1. The [`robots-configuration`][2] repo installed correctly.

[1]: https://github.com/gist
[2]: https://github.com/robotology/robots-configuration

YANGGANG1971, 2020-03-06:
Thanks for your prompt response. 

echo $YARP_ROBOT_NAME
>>iCubOsaka01
Indeed, the version is too old. Many times, I wanted to update it. But I am a bit afraid of that while updating like robots-configuration, the existing demos may not be run correctly.  

Also, I uploaded the codes as a Gist for inspection below.
https://gist.github.com/YANGGANG1971/c3046e88072d7284b99db68f9529ab38

pattacini, 2020-03-06:
@YANGGANG1971 your gist doesn't seem to be Python. Could you clarify?

I meant the HW version of the robot, and you cannot do much in this sense unless you'll upgrade the iCub 😉 

So, `YARP_ROBOT_NAME` is available, although `iKinGazeCtrl` gets configured based on its standard options, which are not robot specific. This may lead to incongruences.

pattacini, 2020-03-06:
> So, `YARP_ROBOT_NAME` is available, although `iKinGazeCtrl` gets configured based on its standard options, which are not robot specific. This may lead to incongruences.

I got it. iCubOsake01 doesn't contain any robot-specific conf file for `iKinGazeCtrl`.
I consider this a bug. We will fix this soon.

cc @marcoaccame

pattacini, 2020-03-06:
Quick update on this. The above fix has been applied to `robots-configuration/devel`. Thus, if you update `robots-configuration` to the latest `devel` branch and then install it, you can then launch this from `icubsrv`:

```sh
$ iKinGazeCtrl --from iKinGazeCtrl.ini
```

pattacini, 2020-03-06:
@YANGGANG1971 if you're concerned about the effect of this overall update, you might consider using this [configuration file][1] straight away.

[1]: https://github.com/robotology/robots-configuration/commit/a3b68b85888717c9be9ad752ec2f9784a845e385

YANGGANG1971, 2020-03-09:
> @YANGGANG1971 your gist doesn't seem to be Python. Could you clarify?
>
> I meant the HW version of the robot, and you cannot do much in this sense unless you'll upgrade the iCub 😉
>
> So, [...]

I updated a Gist for inspection. 
In fact, there are three files-two .cpp files and one .py file.
The file of gaze_python-detector.cpp is to detect the position of the ball. 
The file of gaze_python-detector.cpp is to control the movement of iCub to gaze at the ball.
The file of python_connect.py is to build a communication interface between two .cpp files.

YANGGANG1971, 2020-03-09:
> @YANGGANG1971 if you're concerned about the effect of this overall update, you might consider using this [configuration file](https://github.com/robotology/robots-configuration/commit/a3b68b85888717c9be9ad752ec2f9784a845e385) straight away.

Indeed, I am concerned about the effect of the overall effect. So I want to update the necessary part.  
As your suggestion, is the straight way that I just copy this configuration file to icubsrv pc?
If no, how do I use this kind of straight way specifically?

Nicogene, 2020-03-09:
Hi @YANGGANG1971 ,

> As your suggestion, is the straight way that I just copy this configuration file to icubsrv pc?

yes sure, in order to try it just copy it somewhere and then in that directory run `iKinGazeCtr `

YANGGANG1971, 2020-03-09:
I greatly appreciate your help.

pattacini, 2020-03-09:
@YANGGANG1971 using the correct configuration file will simply let `iKinGazeCtrl` get configured with the set of options corresponding to your hardware, hence it won't sort out the problem you've been experiencing.

The C++ Gist you shared looks good to me. From your screenshot, it doesn't seem that there are address conflicts but who knows. Better off to shut down the robot and `icubsrv` and switch them on again (power cycle) to clean up the whole system. This is a quick test you can do.

pattacini, 2020-03-17:
Closing, feel free to open it up again if needed.

closed 2020-12-15 by pattacini