robotology/community #390: Cannot find the option of CREATE_PYTHON while compiling bindings?

YANGGANG1971, 2020-03-10:
While trying to compile [Yarp bindings](http://wiki.icub.org/wiki/Python_bindings#iCub_bindingsl) for python (ccmake ../), I cannot find the option of CREATE_PYTHON (see below). By hitting t to toggle advanced mode, I cannot still find the option of CAREAT_PYTHON. 

[image]

The related-icub software versions are below. 

> yarp version 3.1.0 
> Ubuntu 16.04
> cmake version 3.5.1 ($ CMake  --version)
> python 2.7 
> swig 3.0.8

traversaro, 2020-03-10:
As documented in https://www.yarp.it/yarp_swig.html, first you need to enable the `YARP_COMPILE_BINDINGS` option, and  then you will find the `CREATE_PYTHON` option.

YANGGANG1971, 2020-03-10:
It works. Thanks a lot.

closed 2020-12-15 by pattacini