robotology/community #391: How to change python 2.7 to python 3, while yarp bindings?

YANGGANG1971, 2020-03-10:
While finishing the settings of yarp bindings, the default python version is 2.7.
But, the current is that python 2.7 has stopped to update and the python 3 is to recommend. 
So how to change the default python version whiling setting up yarp bindings for python?

traversaro, 2020-03-10:
cc @diegoferigo any suggestion?

Nicogene, 2020-03-10:
If I remember correctly there is the cmake variable ` YARP_USE_PYTHON_VERSION`.

Once put it you have tor delete from cache all Python cmake variable set (press `t` to see them all) and then reconfigure.

Can we enforce to make the bindings find python 3 instead of the 2? 

cc @drdanz @traversaro

pattacini, 2020-03-10:
I'm not an expert on this, but it appears that for what concerns `icub-main`, the version can be already customized through a [CMake option][1].

[1]: https://github.com/robotology/icub-main/blob/master/bindings/CMakeLists.txt#L55

diegoferigo, 2020-03-10:
I just checked on a clean build folder and the following happens:

```
❯ cmake -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON ..
[...]
--  [x] Bindings (YARP_COMPILE_BINDINGS)
--  [ ]   Java bindings (CREATE_JAVA)
--  [x]   Python bindings (CREATE_PYTHON)
--  [ ]   Perl bindings (CREATE_PERL)
--  [ ]   CSharp bindings (CREATE_CSHARP)
--  [ ]   TCL bindings (CREATE_TCL)
--  [ ]   Ruby bindings (CREATE_RUBY)
--  [ ]   Lua bindings (CREATE_LUA)
--  [ ]   Octave bindings (CREATE_OCTAVE)
[...]
```

```
❯ grep PYTHON CMakeCache.txt
CMAKE_INSTALL_PYTHONDIR:PATH=lib/python2.7/dist-packages
CREATE_PYTHON:BOOL=ON
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
PYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython2.7.so
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND
YARP_USE_PYTHON_VERSION:STRING=
//ADVANCED property for variable: CMAKE_INSTALL_PYTHONDIR
CMAKE_INSTALL_PYTHONDIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_EXECUTABLE
PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_INCLUDE_DIR
PYTHON_INCLUDE_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY
PYTHON_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY_DEBUG
PYTHON_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: YARP_USE_PYTHON_VERSION
YARP_USE_PYTHON_VERSION-ADVANCED:INTERNAL=1
```

However, if you use either what @Nicogene suggested, or the following, Python 3.6 is correctly detected:

```
❯ cmake -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON -DPython_ADDITIONAL_VERSIONS=3.6 ..
[...]
```

```
❯ grep PYTHON CMakeCache.txt
CMAKE_INSTALL_PYTHONDIR:PATH=lib/python3/dist-packages
CREATE_PYTHON:BOOL=ON
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3.6
PYTHON_INCLUDE_DIR:PATH=/usr/include/python3.6m
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND
YARP_USE_PYTHON_VERSION:STRING=
//ADVANCED property for variable: CMAKE_INSTALL_PYTHONDIR
CMAKE_INSTALL_PYTHONDIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_EXECUTABLE
PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_INCLUDE_DIR
PYTHON_INCLUDE_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY
PYTHON_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY_DEBUG
PYTHON_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: YARP_USE_PYTHON_VERSION
YARP_USE_PYTHON_VERSION-ADVANCED:INTERNAL=1
```

Refer to [`FindPythonLibs`](https://cmake.org/cmake/help/v3.0/module/FindPythonLibs.html).

---

@Nicogene @drdanz CMake 3.12 has a new [`FindPython`](https://cmake.org/cmake/help/v3.12/module/FindPython.html) that can be useful. If we want to avoid finding Python 2, there's a variant that only finds Python 3. I use it on other projects and it works amazingly well.

Btw I just notices the YARP does not find SWIG installed from source (and installed in /usr/local). Instead, using the package from the repos works fine.

YANGGANG1971, 2020-03-10:
is your suggestion that I can use [Findpython](https://cmake.org/cmake/help/v3.12/module/FindPython.html) of CMake3.12?

diegoferigo, 2020-03-10:
@YANGGANG1971 No that was a suggestion about how to enhance YARP to make it detect automatically Python 3 (it requires a new PR to update the sources). In your case, you can use either the `YARP_USE_PYTHON_VERSION` cmake options as @Nicogene [suggested](https://github.com/robotology/QA/issues/391#issuecomment-596993867), or the `Python_ADDITIONAL_VERSIONS` as I wrote above.

pattacini, 2020-03-10:
@diegoferigo can you report on here below the signature of the `find_package` you've used along with `FindPython` to target Python-3 only?

We could improve `icub-main/bindings` with that.

diegoferigo, 2020-03-10:
From the [`FindPython` documentation](https://cmake.org/cmake/help/v3.12/module/FindPython.html):

```
This module looks preferably for version 3 of Python. If not found,
version 2 is searched. To manage concurrent versions 3 and 2 of
Python, use FindPython3 and FindPython2 modules rather than this one.
```

diegoferigo, 2020-03-10:
Related https://github.com/robotology/robotology-superbuild/pull/359

pattacini, 2020-03-10:
Kept still open in case we'd like to comment any further on what's next with YARP.

jgvictores, 2020-03-10:
Tiny note (disclaimer: I still use both Python 2 and 3, on different machines): Python 2.7 is the end of the Python 2 line of development, which reached End of Life on January 1st 2020.

YANGGANG1971, 2020-03-12:
> delete from cache all Python cmake variable set (press `t` to see them all)

 delete from cache all Python cmake variable set (press t to see them all) ?
Sorry, it is a bit confusing. What is cache all Python cmake variable set? How to delete it in detail?

Nicogene, 2020-03-12:
Sorry maybe I was unclear, there are 2 ways to delete che variable in `CMakeCache.txt`, through `ccmake` or manually delete them from `CMakeCache.txt`.

I prefer with `ccmake`

1. Run `ccmake`. in the build directory
2. Press `t` for show also advanced options and entries.
3. Press `/` for start searching
4. Type `python` and press enter
5. Pressing 'n' you browse all the `cmake` variables containing the word "python"
6. Edit `YARP_USE_PYTHON_VERSION` with the python version you need
7. Then delete all the other cmake variables containing the "python" word
In particular repeat steps 3. 4. 5. and press `d` to delete the variable.
8. Finally, when all the variables are cleaned up, press `c`, `g` and `e`.

Et voila' :sparkles: 
Run `make` and you should compile the bindings with python 3.

diegoferigo, 2020-03-12:
Or, if your CMake configuration is trivial to do again, remove the build folder and configure the project from scratch :)

pattacini, 2020-03-12:
> Or, if your CMake configuration is trivial to do again, remove the build folder and configure the project from scratch :)

@YANGGANG1971 is particularly concerned about keeping the "integrity" of his setup.
Better off not doing this then.

YANGGANG1971, 2020-03-13:
> Sorry maybe I was unclear, there are 2 ways to delete che variable in `CMakeCache.txt`, through `ccmake` or manually delete them from `CMakeCache.txt`.
> 
> I prefer with `ccmake`
> 
> 1. Run `ccmake .` in the build directory
> 2. Press `t` for show also advanced options and entries.
> 3. Press `/` for start searching
> 4. Type `python` and press enter
> 5. Pressing `n` you browse all the `cmake` variables containing the word "python"
> 6. Edit `YARP_USE_PYTHON_VERSION` with the python version you need
> 7. Then delete all the other cmake variables containing the "python" word
>    In particular repeat steps 3. 4. 5. and press `d` to delete the variable.
> 8. Finally, when all the variables are cleaned up, press `c`, `g` and `e`.
> 
> Et voila' 
> Run `make` and you should compile the bindings with python 3.

Following your suggestion, I make some changes below,

Settings 

>           YARP_USE_PYTHON_VERSION  python3  
> CREATE_PYTHON   ON 

Delete  options  

> CMAKE_INSTALL_PYTHONDIR     lib/python2.7/site-packages   
> PYTHON_EXECUTABLE  /usr/bin/python 
> PYTHON_INCLUDE_DIR  /usr/include/python2.7 
> PYTHON_LIBRARY  /usr/lib/x86_64-linux-gnu/libpython2.7.so 
> PYTHON_LIBRARY_DEBUG  PYTHON_LIBRARY_DEBUG-NOTFOUND 

But we got the errors below.  
/home/icub/Pictures/Screenshot from 2020-03-13 17-17-20.png

The output log of ccmake ../ is below.
[CMakeOutput.log](https://github.com/robotology/QA/files/4328546/CMakeOutput.log)

Nicogene, 2020-03-13:
Hi @YANGGANG1971 ,

you have to put the version in `YARP_USE_PYTHON_VERSION`, you can checking typing:
```
$ python3 --version
```

In my case is `3.6.9`

YANGGANG1971, 2020-03-13:
$ python3 --version
>>Python 3.7.6
I try to input Python 3.7.6, python3, python3.7.6 and python3.7.
All of these failed.

jgvictores, 2020-03-13:
Note that Python `3.7.6` could be from Anaconda, in which case you may have more than two versions of Python. Could you copy+paste the output of `which python` and `which python3`?

YANGGANG1971, 2020-03-13:
It is from Anaconda.
$ which python
/home/icub/anaconda3/bin/python
$ which python3
$ /home/icub/anaconda3/bin/python

Nicogene, 2020-03-13:
Try just putting `3.7.6`

YANGGANG1971, 2020-03-13:
Great, putting 3.7.6, there are no errors while compiling.
But after 
$ make 
$ make install
$ python3
>>import yarp
[image]

But while executing the codes below, it seems OK.
$python2
>> import yarp

YANGGANG1971, 2020-03-13:
Also, I downloaded the python 3.5.2. 
First, type 3.5.2
$ make
$ make install
  
The same problem will appear. 
[image]

Nicogene, 2020-03-13:
Could you please check `$PYTHONPATH`?
Typing 
```bash
$ echo $PYTHONPATH
```
This has to point where `yarp.py` and `_yarp.so` are located.

so in your case:

For `3.7.6`
```bash
$ export PYTHONPATH="/usr//local/lib/python3.7/dist-packages/"
```

For `3.5.2`
```bash
$ export PYTHONPATH="/usr//local/lib/python3/dist-packages/"
```

Nicogene, 2020-03-13:
I advise you to add this line in your `bashrc`.

YANGGANG1971, 2020-03-16:
A new error occurred.

[image]

YANGGANG1971, 2020-03-17:
It has been solved. It is due to the version of CMake...
Sorry about that. 
I appreciate your help.

closed 2020-12-08 by pattacini