robotology/community #392: when compiling icub bindings for python, I running into some problems after enabling ICUB_COMPILE_BINDINGS.

YANGGANG1971, 2020-03-10:
The detailed procedures of icub bindings for python
http://wiki.icub.org/wiki/Python_bindings#iCub_bindingsl
Step 1  install yarp bindings (successfully)
Step 2 icub bindings

> 	$  cd $ICUB_ROOT/bindings
>     $  mkdir build  && ccmake ../

After enabling ICUB_COMPILE_BINDINGS, I tried to configure the CMAKE GUI and the errors occurred. 
  
[image]

pattacini, 2020-03-10:
Hi @YANGGANG1971 

Regarding this, be careful of:
- https://github.com/robotology/icub-main/issues/645: Python binding in `icub-main` are currently broken unless you'll follow the workaround proposed by @diegoferigo in that issue.
- I've just opened up https://github.com/robotology/icub-main/pull/647 to better deal with Python-3.

That said, from your screenshot, I'm unable to see what is causing the error. Thus, could you please drop here the output log of `cmake .`?

Further, what version (git hash or tag) of `icub-main` are you using?

Our recommendation is always to update your system in order to incorporate any bug fix we have added in the past, which may be relevant. As you're a bit concerned about the integrity of your robot setup, bear in mind that you can build bindings on a different machine that you can profitably connect to the cluster where instead you can perform the update lightheartedly.

YANGGANG1971, 2020-03-12:
The output log of ccmake ../ is below. 
 
[CMakeError.log](https://github.com/robotology/QA/files/4323484/CMakeError.log)
[CMakeOutput.log](https://github.com/robotology/QA/files/4323485/CMakeOutput.log)

I am using the version of icub-main is v1.12.0-8gc3698cb

Does it mean that I cannot use the iCub bindings now?

In addition, may I know the difference between icub bindings and yarp bindings?

pattacini, 2020-03-12:
@YANGGANG1971 

### Warm recommendation
You're working with a very old version of the software that dates back to almost 2 years ago. Going further this way there's a high probability you'll stumble upon many other problems.

Therefore, my warm recommendation is to leave untouched the cluster of the robot along with its server machine and set up a computer where you can start from the very beginning with the latest software. This computer can then be connected to the cluster and used in the loop for your experiments.

### Error
The error occurred during the CMake configuration reports that
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /usr/local/src/robot/icub-main/bindings/CMakeFiles/CMakeTmp
```
which is quite weird as CMake seems unable to find the minimum set of components to configure the build framework. This might be due to a compromised CMake cache or the like and should lead you to clean up your system, which you don't want to do. So, have a look at my recommendation above 😃

### Bindings
- YARP bindings do expose the C++ API contained within the `yarp` repository to other languages like Python. For example: YARP Ports, Bottles, Joint Control Interface.
- iCub bindings do expose the C++ API contained within the `icub-main` repository to other languages like Python. For example: the kinematics library, the handling of the skin.

In particular, the handling of the skin is making the iCub bindings fail in Python. As a workaround, please follow what @diegoferigo suggested in https://github.com/robotology/icub-main/issues/645#issue-573334009

pattacini, 2020-03-17:
Closing, feel free to open it up again if needed.

closed 2020-12-15 by pattacini