robotology/community #393: In iCub simulator, how do I make a change, e.g. change the color to green from red?

YANGGANG1971, 2020-03-12:
In [mover.cpp](https://gist.github.com/YANGGANG1971/a125e6aa05bd23ef179898e5f692e978), I guess the snippet of codes below indicated the basic settings of the simulator. 
What does it mean in detail? how do I make a change, e.g. change the color to green from red?
            

>             cmd.addString("world");
>             cmd.addString("mk");
>             cmd.addString("ssph");
>             cmd.addDouble(0.03);
>             cmd.addDouble(x0[0]);
>             cmd.addDouble(x0[1]);
>             cmd.addDouble(x0[2]);
>             cmd.addDouble(1.0);
>             cmd.addDouble(0.0);
>             cmd.addDouble(0.0);
>             cnt=0;
>             init=false;

pattacini, 2020-03-12:
Hi @YANGGANG1971 

You'll find out the information on how to manipulate objects within the simulator in our Wiki at http://wiki.icub.org/wiki/Simulator_README.

In particular, the last 3 doubles account for the object color in the RGB domain. Therefore, you should change
```c++
cmd.addDouble(1.0);
cmd.addDouble(0.0);
cmd.addDouble(0.0);
```
into
```c++
cmd.addDouble(0.0);
cmd.addDouble(1.0);
cmd.addDouble(0.0);
```

YANGGANG1971, 2020-03-12:
@pattacini  Thanks a lot.
I will check it.

closed 2020-12-08 by pattacini