robotology/community #395: In the Cartesian reference, how to reach the desired position by constrainting some joints?

YANGGANG1971, 2020-04-28:
For controlling iCub, I want to let the iCub's finger reach the desired position, in fact, I can do it through the methods below,
>     Vector computeHandOrientation()
>     {
>         Matrix R(3,3);
>         R(0,0)=-1.0; R(0,1)= 0.0; R(0,2)= 0.0;
>         R(1,0)= 0.0; R(1,1)= 0.0; R(1,2)=-1.0;
>         R(2,0)= 0.0; R(2,1)=-1.0; R(2,2)= 0.0;
>         return dcm2axis(R);
>     }
>  o=computeHandOrientation()
>  iarm->goToPoseSync(approach_x,o);  
> //approach_x are center-of-mass of the pixels of left eye and right eye
>  iarm->waitMotionDone();

In fact, I just move the elbow joint and shoulder joint to reach the desired position. How do I constraint the other joint movement?

pattacini, 2020-04-29:
Hi @YANGGANG1971 

I can point you at our [main documentation][1] illustrating the use of the Cartesian Interface. In particular, I deem the following section quite pertinent:

>One useful feature is the possibility to enable/disable some degrees of freedom on-line. For example, the arm comes with 10 possible DOF's (the torso is included) that the user may or may not want to use at the same time while reaching.
In the following snippet of code we ask to enable the torso pitch and yaw joints (which are off by default).
>
>```c++
>Vector curDof;
>icart->getDOF(curDof);
>cout<<"["<<curDof.toString()<<"]"<<endl;  // [0 0 0 1 1 1 1 1 1 1] will be printed out
>Vector newDof(3);
>newDof[0]=1;    // torso pitch: 1 => enable
>newDof[1]=2;    // torso roll:  2 => skip
>newDof[2]=1;    // torso yaw:   1 => enable
>icart->setDOF(newDof,curDof);
>cout<<"["<<curDof.toString()<<"]"<<endl;  // [1 0 1 1 1 1 1 1 1 1] will be printed out
>```
>To sum up, a value of '1' makes the joint a real degree of freedom, a value of '0' switches it off, while the special value of '2' indicates that the joint status won't be modified (to skip it and proceed forward).

Essentially, you would achieve your goal by doing:
```c++
Vector newDof(icart->getDOF().length(), 0);
newDOF[0] = newDOF[1] = newDOF[2] = newDOF[3] = 1.0;
icart->setDOF(newDof, newDof);
```

[1]: http://www.icub.org/doc/icub-main/icub_cartesian_interface.html#sec_cart_useinterface

pattacini, 2020-04-29:
You could further refine the joints range by fiddling with the [`yarp::dev::ICartesianControl::setLimits()`][1] method.

[1]: http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a403a98ab58c38e1193ed9df021d7d4e8

pattacini, 2020-05-09:
Closing, feel free to open it up again if needed.

closed 2020-12-08 by pattacini