robotology/community #396: icub@pc104 install problem

hoesung, 2020-04-29:
## Brief description
Hi, I am trying to install iCub. but It is my first time to handle with yarp. :(
I completely follow directions in below:
http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions#Install_the_ssh_keys_for_password-less_login_on_PC104

But, I could not connect to icub@pc104.
What can I do more? please, give me some directions

## Detailed context
ssh icub@pc104
ssh: connect to host pc104 port 22: No route to host

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.1  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::eef4:bbff:fe2f:3ada  prefixlen 64  scopeid 0x20<link>
        ether ec:f4:bb:2f:3a:da  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 145  bytes 11317 (11.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xf7c00000-f7c20000  

## Accompanying material

pattacini, 2020-05-04:
Did you try doing a `ping pc104` first?
If you won't receive any positive response, then it's likely that `pc104` is not a known host. To get around this, you might want to add `pc104` to `~/.ssh/known_hosts`.

mbrunettini, 2020-05-04:
Try to ping the pc104 to see:
1) if the host IP is resolved
2) if the host is reachable

than we can correct the issue
BTW, @pattacini : the known_hosts file is not used to resolved the host IP, but it used by the SSH connection to store and check the host SSH Key fingerprint

pattacini, 2020-05-09:
Closing, feel free to open it up again if needed.

closed 2020-12-15 by pattacini