robotology/community #398: How to operate each individual finger

YANGGANG1971, 2020-05-05:
I would like to only let the index finger point at a specific object and other fingers are bent. How do I achieve this?

pattacini, 2020-05-09:
Fingers are standard joints you can control in the configuration space in different modalities (e.g. `position`, `velocity`, `position-direct`) in the same way you control the joints of the arms, the legs or the head.

You only ought to know that fingers belong to `left_arm` and `right_arm` iCub's parts according to the following mapping:

| part index | joint |
|:---:|:---|
| 0 | shoulder |
| 1 | shoulder |
| 2 | shoulder |
| 3 | elbow |
| 4 | wrist |
| 5 | wrist |
| 6 | wrist |
| 7 | **fingers adbuction** |
| 8 | **thumb  abduction** |
| 9 | **thumb proximal** |
| 10 | **thumb distal** |
| 11 | **index proximal** |
| 12 | **index distal** |
| 13 | **middle proximal** |
| 14 | **middle distal** |
| 15 | **ring + pinkie** |

To know how to control those joints, please refer to the resources listed below:
- http://www.icub.org/doc/icub-main/icub_motor_control_tutorial.html
- https://github.com/vvv-school/tutorial_joint-interface

Further, for your knowledge, we have high-level software components implementing coordinated pointing movements as per https://github.com/robotology/icub-main/pull/413.

YANGGANG1971, 2020-05-09:
Q1: When I used  

> iarm→getDOF(curDof); 

I will get 

> 1.000000	 1.000000	 1.000000	 1.000000	 1.000000	 1.000000	 1.000000	 1.000000	 1.000000	 1.000000 

It seems that there are only 10 joints rather than 15 joints

Q2: How to implement the function below? It seems that in https://www.yarp.it/, there is no explanation about far. 
$ yarp rpc /actionsRenderingEngine/cmd:io
>> pfar (x y z)
https://github.com/robotology/icub-main/pull/413

pattacini, 2020-05-09:
As mentioned in the tutorials I pointed you at as well as in this [post][1], don't use the `cartesiancontrollerclient` device driver to control a joint in the configuration space. Instead, resort to `remote_controlboard` device.

`ActionRenderingEngine` (aka `ARE`) is not a YARP thing, it is rather a component bundled in the `icub-main` repository. It's quite a big module and thus takes some time to be explored and learned. Anyway, here is its doc page: http://www.icub.org/doc/icub-main/group__actionsRenderingEngine.html.

[1]: https://github.com/robotology/QA/issues/399#issuecomment-626178216

closed 2020-12-08 by pattacini