robotology/community #399: While defining a different effector, errors occured.

YANGGANG1971, 2020-05-06:
While defining a different effector following the link below, the program can be compiled, but cannot be executed.
http://www.icub.org/software_documentation/icub_cartesian_interface.html
>    IEncoders *iencs;
>      
>         drvArm.view(iencs); //get interface 
>         drvArm.view(pos); 
> 
>         int nEncs;
>        iencs->getAxes(&nEncs);
>        Vector encs(nEncs);
>        iencs->getEncoders(encs.data());
> 
> 
>         Vector joints;
>         iCubFinger finger("right_index");                  // relevant code to get the position of the finger tip
>         finger.getChainJoints(encs,joints);                // wrt the end-effector frame
> 
>         Matrix tipFrame=finger.getH((M_PI/180.0)*joints);
>         Vector tip_x=tipFrame.getCol(3);
>         Vector tip_o=yarp::math::dcm2axis(tipFrame);
>         iarm->attachTipFrame(tip_x,tip_o);                // establish the new controlled frame 

[image]

The entire code can refer to the link below. 
https://www.dropbox.com/s/0qfebkjkd0ra5f6/tracker.cpp?dl=0

pattacini, 2020-05-09:
Hi @YANGGANG1971 

There must be a bug in your code. Therefore, you should carry out careful troubleshooting where you might employ high-level debuggers or just fill in the code with lots of print instructions with the goal of identifying the line that yields seg-fault. 

That said, the device driver `cartesiancontrollerclient` does not implement either `IPositionControl` or `IEncoders` motor interfaces. To that end, you shall open up a `remote_controlboard` driver.

YANGGANG1971, 2020-05-09:
If I uncommented the sinppet of codes below, it seems that the program works well. I confirmed that I opened up the remote_controlboard driver. 
> int nEncs;
> iencs->getAxes(&nEncs);
> Vector encs(nEncs);
> iencs->getEncoders(encs.data());

YANGGANG1971, 2020-05-09:
The PC will get stuck. So it is very weird.

pattacini, 2020-05-09:
The snippet you posted doesn't seem to contain any opening of the correct device driver.

You should call something like the following:
```c++
Property options;
options.put("device", "remote_controlboard");
options.put("remote", "/icub/left_arm");
options.put("local", "/local");

PolyDriver driver_arm;
driver_arm.open(options);
driver_arm.view(iencs);
// ...
```

YANGGANG1971, 2020-05-09:
Actually, the snippet has existed and see below.
>         Property optArm;//Enable arm device
>         optArm.put("device","cartesiancontrollerclient");
>         optArm.put("remote","/icubSim/cartesianController/right_arm");
>         optArm.put("local","/cartesian_client/right_arm");
>         PolyDriver drvArm;
>         drvArm.open(optArm);
>         drvArm.view(iarm);
>         drvArm.view(iencs); //get interface 
>         drvArm.view(pos);
The entire and detailed project can refer to this link below.
https://www.dropbox.com/s/0qfebkjkd0ra5f6/tracker.cpp?dl=0

pattacini, 2020-05-09:
Exactly, your code contains only one driver. You need the following two device drivers at the same time instead:
- `cartesiancontrollerclient`
- `remote_controlboard` 

Refer to my snippet above for how opening the second.

YANGGANG1971, 2020-05-09:
Okay, thank you. I will add it and test it.

YANGGANG1971, 2020-05-09:
It works. Thanks

YANGGANG1971, 2020-05-09:
But, I am wondering why I need to open two devices.
```
cartesiancontrollerclient
remote_controlboard
```

pattacini, 2020-05-09:
They are two different device drivers:
- `cartesiancontrollerclient` deals with the Cartesian control of a manipulator, where the goal is to control the end-effector in the Cartesian space.
- `remote_controlboard` deals with the control of the joint in the configuration space.

The controls underneath, the purposes, and the algorithms are different so are the responsibilities held by the software components.

closed 2020-12-15 by pattacini