robotology/community #401: Potential issue with apt installation of icub-common

kothiga, 2020-05-13:
## Brief description

I think I may have come across a small issue with installing `icub-common` from a fresh installation. In short I think this issue may be caused by a combination of `ycm-cmake-modules` being bundled together with `icub-common` and `apt` providing an old version of `cmake`. At the end of the I provide my solution to this issue, but I am wondering what you would recommend as the most stream line approach.

## Detailed context

From a fresh installation of `ubuntu 18.04`, after doing the standard 

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 57A5ACB6110576A6
sudo sh -c 'echo "deb http://www.icub.org/ubuntu bionic contrib/science" > /etc/apt/sources.list.d/icub.list'
sudo apt update
sudo apt install icub-common
```

I end up getting the following error:

```
The following packages have unmet dependencies:
 icub-common : Depends: ycm-cmake-modules but it is not going to be installed
               Depends: cmake (>= 3.12.0)
E: Unable to correct problems, you have held broken packages.
```

Which is obviously caused by `ycm-cmake-modules` requiring a newer version of `cmake` than what `apt` provides (`3.10.2`).

Previously if I wanted to install `ycm` from [source](https://github.com/robotology/ycm), it would be pretty easy to purge the old `cmake` and either install a newer version via `snap` (`3.17.2`), or to build `cmake` from [source](https://anglehit.com/how-to-install-the-latest-version-of-cmake-via-command-line/) (`3.16.4` or higher).

The issue is that if I have done either of these options, `apt` doesn't "know" about the newer version of `cmake` that is already installed and instead tries to install the old `3.10.2` version which immediately throws the above error of a version conflict.

## Solution

My solution was to follow this [guide](https://apt.kitware.com/) to point `apt` at a repository to supersede the old `cmake`. 

After setting that up when I install `icub-common`, `ycm-cmake-modules` is able to recursively find and install the newer `cmake` (`3.17.2`).

I just wanted to point out that this is a problem that some people may stumble upon. I am unsure if pointing `apt` at another repository is the most elegant solution, but I wanted to quickly share my quick solution to this.

What are your thoughts?

## Accompanying material

pattacini, 2020-05-13:
Thanks @kothiga for reporting the issue and your workaround 👍 

I reckon that @mbrunettini has already come across something related while gearing up in view of distro 2020.05. Thus, we should be able to get it fixed soon, hopefully.

cc @drdanz

mbrunettini, 2020-05-13:
Thanks @kothiga : an updated cmake is mandatory for the latest releases of YARP and that's why there is the dependency in ycm-cmake-modules (but not only).
We should:
1) check dependencies of yarp cmake packages for the upcoming release 
2) improve our instructions were it specifies that a recent cmake is required and provide the istructions to install from a repository as you suggested (there is [official kitware repo](https://apt.kitware.com/) )
let me clarify that this issue only affects Ubuntu 18.04, while latest LTS (20.04) has a more recent version of cmake: packages for 20.04 will be provided with the upcoming release of our software

mbrunettini, 2020-05-25:
I added the instruction to add KITWARE repository on [icub wiki](https://wiki.icub.org/wiki/Linux:Installation_from_binaries#Dependencies)

As specified in the wiki, only packages for Ubuntu 18.04 are affected.

I think we can close this now

pattacini, 2020-05-25:
Closing then 👍 
@kothiga feel free to add up further comments if required.

closed 2020-12-15 by pattacini