robotology/community #402: How to connect PC camera to iCub simulator?

YANGGANG1971, 2020-05-14:
I am trying to connect the camera from PC to iCub simulator as below. 
[image]

After finishing the connection, I want to move the ball using my hand and let the robot's finger point at the ball position from the virtual screen. 

Could you give me a suggestion or a tutorial? How should I do it? What learning materials can I get?  Appreciate it.

YANGGANG1971, 2020-05-19:
I also notice that the iTalk project might be helpful.
http://wiki.icub.org/italk/dox/html/index.html
After reading the materials above,  I am still confused and how to connect it to iCub simulator.

pattacini, 2020-05-19:
Hi @YANGGANG1971 

Relevant info can be browsed at http://wiki.icub.org/wiki/Simulator_README#PROJECT_A_VIDEO_STREAM_ON_A_SCREEN.

I'm not an expert of this though.

YANGGANG1971, 2020-05-19:
Sorry to disturb you again. 
http://wiki.icub.org/wiki/Simulator_README#PROJECT_A_VIDEO_STREAM_ON_A_SCREEN.
Following this instruction above, I get this below,
[image]
It did project a predefined video to the screen in the iCub simulator. The predefined video is from 

> yarpdev --device test_grabber --name /test/video --mode ball

But it is weird. 
(1) While projecting a video, there is no iCub robot in the simulator. 
I set screen ON in iCub_parts_activation.ini 
(2) How do I get my PC camera info through yarp? 

I know that you are not an expert on this, but could you give me a direction? Maybe I can explore this.

I also found that this method was mentioned in ITalk program below.
https://www.fose1.plymouth.ac.uk/socem/crns/italk/private/deliverable/ITALK-Y1-D1.1-icub-software.pdf

Actually, I am lost. Sorry about annoying you.

YANGGANG1971, 2020-05-19:
All command lines are as follows.
> yarpserver --write
> iCub_SIM
> yarpdev --device test_grabber --name /test/video --mode ball
> yarp connect /test/video /icubSim/texture/screen

YANGGANG1971, 2020-05-19:
While I type the commands below
$ yarp rpc /icubSim/head/rpc:i
$ set pos 0 -60
It shows below.
/home/icub/Pictures/Screenshot from 2020-05-19 22-11-20.png
I guess for the question
 (1) While projecting a video, there is no iCub robot in the simulator. Only a screen was shown in the simulator.
The iCub existed and just the position has an overlap with the screen. 
So how do I adjust the position of screen or iCub in the simulated world?

EliseiFrederic, 2020-05-19:
Hi,Your "set pos" command is not for your bash/console. It's directed towards the simulator. Should be typed/sent in the rpcHope it helps,Frederic

vtikha, 2020-05-19:
Dear @YANGGANG1971 
sorry for the late reply, I was sidetracked by other projects and forgot to add my two cents. 

To get the images from the webcam on the screen is practically how you did for the `test_grabber` device. 
There are actually a few ways of doing this: 

**1 - Use the `opencv_grabber`**

You would need to install `yarp` with `opencv` and make sure that  the option `ENABLE_yarpmod_opencv_grabber` is set to `ON`. If I am not mistaken this would allow you to use `opencv_grabber` as a yarp device.  

After that you can check if the `opencv_grabber` device is available looting at the results produced by the `yarpdev --list` command.

Now you can run the following command:
```bash
yarpdev --device opencv_grabber
```
and connect the output port `/grabber` to the screen port as you did previously. 

**2 - Use the usbcamera**

This uses `video4linux` and therefore can be used to stream from a webcam
You will probably need to use `ENABLE_yarpmod_grabber ON`
```bash
yarpdev --device grabber --subdevice usbCamera --d /dev/video0
```

**3 - Use the ffmpeg**

If you have ffmpeg installed on your pc you can also use that, set 
`ENABLE_yarpmod_ffmpeg_grabber ON` 

and launch

```bash
yarpdev --device ffmpeg_grabber --v4l2

```

**Question on position** 

As for the `iCub overlapping` with the screen I am not actually sure of what you are trying to do.

When it all start for the first time, with the default `viewpoint` of the simulator, the screen actually hides the iCub. The user needs to navigate in the environment/world with the keys and the mouse. Use keys `w,a,s,d ` and mouse for rotating. 

Hope this all clears up a bit everything

YANGGANG1971, 2020-05-19:
@EliseiFrederic 
Sorry
I explained it wrong. 
Firstly, I type yarp rpc /icubSim/head/rpc:I on the terminal.
At the same terminal, I input in the set pos 0 -60 in the rpc.
This command can be executed well. 
But the weird thing is if I use yarpview and connect it. 

> yarpview --name /viewerR --x 360 --y 10 --compact
> yarp connect /icubSim/cam/right /viewerR

In the /viewerR, I can see the iCub arm.
But, in the iCub Simulator, I cannot find the iCub robot and even I try to drag and hold the mouse from different directions.
[image]
[image]

YANGGANG1971, 2020-05-19:
@vtikha 
It works after your solution on Question on position.

I will continue to test how to get live image from webcam.

YANGGANG1971, 2020-05-19:
> **3 - Use the ffmpeg**
> 
> If you have ffmpeg installed on your pc you can also use that, set
> `ENABLE_yarpmod_ffmpeg_grabber ON`
> 
> and launch
> 
> ```shell
> yarpdev --device ffmpeg_grabber --v4l2
> ```

While I want to use FFmpeg, firstly I installed FFmpeg  by typing below

>         sudo apt-get update
> 	sudo apt-get install ffmpeg

But, when I enter the repository of yarp/build, compile yarp, 

> ccmake ../

I found the errors below
[image]

vtikha, 2020-05-19:
Hi @YANGGANG1971, I mentioned `ffmpeg` if you already installed on your pc. If you choose to go with `ffmpeg` you would need to get the source files, compile them with the correct options and recompile YARP to be able to use them. It is not a straightforward solution.  

I would suggest to go with point **1** or **2**.

YANGGANG1971, 2020-05-20:
Following your suggestion, when I type 
$ cmake -DCREATE_DEVICE_LIBRARY_MODULES=ON  -D=ENABLE_yarpmod_opencv_grabber=ON ../

I ran into some problems as below  

> -- YARP Version: 3.2.1 (3.2.1+dirty)
> -- Using:
> --   CMake: 3.11.4 (/usr/local/bin/cmake)
> --   C Compiler: GNU 5.4.0 (/usr/bin/cc)
> --   CXX Compiler: GNU 5.4.0 (/usr/bin/c++)
> -- Build type: Release
> -- CMake modules directory: /usr/local/src/robot/yarp/cmake;/usr/local/src/robot/yarp/extern/ycm/ycm-0.10.4/find-modules;/usr/local/src/robot/yarp/extern/ycm/ycm-0.10.4/modules;/usr/local/src/robot/yarp/extern/ycm/ycm-0.10.4/3rdparty;/usr/local/src/robot/yarp/extern/ycm/ycm-0.10.4/cmake-3.14/Modules
> -- Checking GNUCXX version 3/4 to determine  OpenCV /opt/net/ path
> -- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
> -- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
> -- OPENCV_LIBRARY not found turning off OPENCV_FOUND
> -- OPENCV_CVAUX_LIBRARY not found turning off OPENCV_FOUND
> -- OPENCV library or headers not found. Please search manually or set env. variable OPENCV_ROOT to guide search.
> CMake Error in /usr/local/src/robot/yarp/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
>   Imported target "MPI::MPI_C" includes non-existent path
> 
>     "/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent"
> 
>   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
> 
>   * The path was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and references files it does not
>   provide.
> 
> 
> 
> CMake Error in /usr/local/src/robot/yarp/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
>   Imported target "MPI::MPI_C" includes non-existent path
> 
>     "/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent"
> 
>   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
> 
>   * The path was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and references files it does not
>   provide.
> 
> 
> 
> CMake Error at /usr/local/share/cmake-3.11/Modules/FindMPI.cmake:1179 (try_compile):
>   Failed to generate test project build system.
> Call Stack (most recent call first):
>   /usr/local/share/cmake-3.11/Modules/FindMPI.cmake:1220 (_MPI_try_staged_settings)
>   /usr/local/share/cmake-3.11/Modules/FindMPI.cmake:1480 (_MPI_check_lang_works)
>   cmake/YarpFindDependencies.cmake:340 (find_package)
>   CMakeLists.txt:76 (include)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/usr/local/src/robot/yarp/build/CMakeFiles/CMakeOutput.log".
> See also "/usr/local/src/robot/yarp/build/CMakeFiles/CMakeError.log".
>

YANGGANG1971, 2020-05-20:
When I tried
$ cd yarp/build
$ cmake -DCREATE_DEVICE_LIBRARY_MODULES=ON  -D=ENABLE_yarpmod_opencv_grabber=ON ../
$ make 
Error occured 
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp: In member function ‘virtual void DumpThread::run()’:
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:30: error: ‘IplImage’ was not declared in this scope
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                               ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:39: error: expected primary-expression before ‘)’ token
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                                        ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:40: error: expected ‘)’ before ‘itemEnd’
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                                         ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:429:39: error: expected primary-expression before ‘)’ token
>                  int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
>                                        ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:429:40: error: expected ‘)’ before ‘itemEnd’
>                  int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
>                                         ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:438:77: error: ‘CV_FOURCC’ was not declared in this scope
>                  videoWriter.open(videoFile.c_str(),CV_FOURCC('H','F','Y','U'),
>                                                                              ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:439:58: error: ‘cvSize’ was not declared in this scope
>                                   fps,cvSize(frameW,frameH),true);
>                                                           ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:33: error: ‘cvarrToMat’ is not a member of ‘cv’
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                  ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:49: error: ‘IplImage’ was not declared in this scope
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                                  ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:58: error: expected primary-expression before ‘)’ token
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                                           ^
> src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/build.make:62: recipe for target 'src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o' failed
> make[2]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o] Error 1
> CMakeFiles/Makefile2:5110: recipe for target 'src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/all' failed
> make[1]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2
>

YANGGANG1971, 2020-05-20:
@vtikha
I check the version of opencv as below
$pkg-config --modversion opencv
>>2.4.9.1

YANGGANG1971, 2020-05-20:
I notice this solution, but I am still confused that how I should do. 
https://github.com/robotology/yarp/issues/1672

Nicogene, 2020-05-20:
Hi @YANGGANG1971 ,

which version of Ubuntu you have?
For supporting OpenCV 4 we dropped the support for OpenCV 2 since it is quite old, I suggest you to install a newer version of OpenCV, at least > 3.0.0
It may solve your compilation problem

YANGGANG1971, 2020-05-20:
Ubuntu 16.04

YANGGANG1971, 2020-05-20:
It is very weird. When I try to install OpenCV 3.0 or 4.0. It always failed. Do you have any suggestions or links that I can follow?

vtikha, 2020-05-20:
@YANGGANG1971  what do you mean it always fails? where? at which step?  anyway have a look at the following:

**Install from source**

The following steps will install **opencv** with **contrig options** for both **python and c++** (i m trying to cover all aspects)

```bash
sudo apt update
sudo apt install build-essential cmake git pkg-config libgtk-3-dev libavcodec-dev libavformat-dev libswscale-dev libv4l-dev libxvidcore-dev libx264-dev
sudo apt install libjpeg-dev libpng-dev libtiff-dev gfortran openexr libatlas-base-dev python3-dev python3-numpy libtbb2 libtbb-dev libdc1394-22-dev
```

**create directory to contain all opencv files**

```bash
mkdir ~/opencv_source
```

**clone repos**

```bash
cd ~/opencv_source
git clone https://github.com/opencv/opencv.git
git clone https://github.com/opencv/opencv_contrib.git
```

**run cmake**

```bash
cmake -D CMAKE_BUILD_TYPE=RELEASE \
    -D CMAKE_INSTALL_PREFIX=/usr/local \
    -D INSTALL_C_EXAMPLES=ON \
    -D INSTALL_PYTHON_EXAMPLES=ON \
    -D OPENCV_GENERATE_PKGCONFIG=ON \
    -D OPENCV_EXTRA_MODULES_PATH=~/opencv_source/opencv_contrib/modules \
    -D BUILD_EXAMPLES=ON ..
```

**compile and install opencv in your system**

```bash
make -j2
sudo make install
```

you should be good to go.

Nicogene, 2020-05-20:
It may be whorty to invest in an upgrade at least to Ubuntu 18.04, because it has opencv 3.3.2 but more important it is a supported distribution .

Checking [this table](https://wiki.icub.org/index.php?title=YARP_Supported_Distributions&diff=22419&oldid=22397#Old_releases) for using Ubuntu 16.04 you have to downgrade yarp to 3.0 and all the rest of the icub ecosystem.

You may fix this issue, but more can come out later.

See [here](https://wiki.icub.org/index.php?title=YARP_Supported_Distributions&diff=22419&oldid=22397#Supported_releases) the supported distributions

YANGGANG1971, 2020-05-20:
@EliseiFrederic 
If I update my Ubuntu 16.04 to 18, do I need to reinstall all software including yarp and iCub?

YANGGANG1971, 2020-05-20:
Following your suggestion, it can be installed. 
But when I type 

> pkg-config --modversion opencv

The terminal still prompts 2.4.9.1.

Also, I tried below

> $ cmake -DCREATE_DEVICE_LIBRARY_MODULES=ON -D=ENABLE_yarpmod_opencv_grabber=ON ../
> $ make

get 

> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
> /home/icub/anaconda3/lib/libtbb.so.2: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
> /home/icub/anaconda3/lib/libtbb.so.2: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFGetField@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
> //usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_generate@UUID_1.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFClose@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFOpen@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetField@LIBTIFF_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
> //usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_unparse_lower@UUID_1.0'
> /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
> collect2: error: ld returned 1 exit status
> src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/build.make:279: recipe for target 'bin/ENABLE_yarpmod_opencv_grabber=ONyarpdataplayer' failed
> make[2]: *** [bin/ENABLE_yarpmod_opencv_grabber=ONyarpdataplayer] Error 1
> CMakeFiles/Makefile2:5766: recipe for target 'src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all' failed
> make[1]: *** [src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2
>

YANGGANG1971, 2020-05-20:
> Hi @YANGGANG1971 ,
> 
> which version of Ubuntu you have?
> For supporting OpenCV 4 we dropped the support for OpenCV 2 since it is quite old, I suggest you to install a newer version of OpenCV, at least > 3.0.0
> It may solve your compilation problem

I upgrade to Ubuntu 18. and Opencv version is 3.2.0. But I cannot compile the code as below. 

> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp: In member function ‘virtual void DumpThread::run()’:
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:30: error: ‘IplImage’ was not declared in this scope
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                               ^~~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:30: note: suggested alternative: ‘DumpImage’
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                               ^~~~~~~~
>                               DumpImage
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:39: error: expected primary-expression before ‘)’ token
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                                        ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:428:40: error: expected ‘)’ before ‘itemEnd’
>                  int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
>                                         ^~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:429:39: error: expected primary-expression before ‘)’ token
>                  int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
>                                        ^
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:429:40: error: expected ‘)’ before ‘itemEnd’
>                  int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
>                                         ^~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:438:52: error: ‘CV_FOURCC’ was not declared in this scope
>                  videoWriter.open(videoFile.c_str(),CV_FOURCC('H','F','Y','U'),
>                                                     ^~~~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:438:52: note: suggested alternative: ‘CV_16UC’
>                  videoWriter.open(videoFile.c_str(),CV_FOURCC('H','F','Y','U'),
>                                                     ^~~~~~~~~
>                                                     CV_16UC
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:439:38: error: ‘cvSize’ was not declared in this scope
>                                   fps,cvSize(frameW,frameH),true);
>                                       ^~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:439:38: note: suggested alternative: ‘dysize’
>                                   fps,cvSize(frameW,frameH),true);
>                                       ^~~~~~
>                                       dysize
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:37: error: ‘cvarrToMat’ is not a member of ‘cv’
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                      ^~~~~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:37: note: suggested alternative: ‘SparseMat’
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                      ^~~~~~~~~~
>                                      SparseMat
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:49: error: ‘IplImage’ was not declared in this scope
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                                  ^~~~~~~~
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:49: note: suggested alternative: ‘DumpImage’
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                                  ^~~~~~~~
>                                                  DumpImage
> /usr/local/src/robot/yarp/src/yarpdatadumper/main.cpp:467:58: error: expected primary-expression before ‘)’ token
>                      cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
>                                                           ^
> src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/build.make:62: recipe for target 'src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o' failed
> make[2]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o] Error 1
> CMakeFiles/Makefile2:5169: recipe for target 'src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/all' failed
> make[1]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2
>

YANGGANG1971, 2020-05-21:
@vtikha
Currently, I can compile all files, just update every to the latest.
Ubuntu-->20
Yarp-->3.3.2
iCub-->v1.15.0
But, when I type
>  yarpdev --device opencv_grabber --name /image:o --camera 0 
get 
> [ERROR]/usr/local/src/robot/yarp/src/libYARP_os/src/yarp/os/YarpPluginSettings.h:240 bool yarp::os::YarpPluginSettings::readFromSelector(const string&): Cannot find "opencv_grabber" plugin (not built in, and no .ini file found for it)
> Check that YARP_DATA_DIRS leads to at least one directory with plugins/opencv_grabber.ini or share/yarp/plugins/opencv_grabber.ini in it
> yarpdev: ***ERROR*** could not find device <opencv_grabber>
> [ERROR]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:503 static int yarp::dev::Drivers::yarpdev(int, char**): yarpdev: ***ERROR*** device not available.
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:511 static int yarp::dev::Drivers::yarpdev(int, char**): Suggestions:
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:512 static int yarp::dev::Drivers::yarpdev(int, char**): + Do "yarpdev --list" to see list of supported devices.
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:515 static int yarp::dev::Drivers::yarpdev(int, char**): + Or append "--verbose" option to get more information.
>

pattacini, 2020-05-21:
Most likely, if you call for the devices list:
```sh
$ yarpdev --list | grep -i opencv
```
you won't find it out.

Then, either you forgot to tick in `opencv_grabber` from within CMake mask at YARP configuration time, or OpenCV is not correctly recognized so you won't have that option available. In the latter case, check whether `OpenCV_DIR` env variable is pointing to the correct place (if you didn't `make install` OpenCV).

YANGGANG1971, 2020-05-22:
I guess maybe before installing yarp, I need to install some dependencies.
I found some errors. 

sudo apt-get install  qtdeclarative5-qtquick2-plugin qtdeclarative5-window-plugin
>> Unable to locate package qtdeclarative5-qtquick2-plugin
>>  Unable to locate package qtdeclarative5-window-plugin

sudo apt-get install qtdeclarative5-qtmultimedia-plugin qtdeclarative5-controls-plugin
>>Unable to locate package qtdeclarative5-qtmultimedia-plugin
>>Unable to locate package qtdeclarative5-controls-plugin

sudo apt-get install qtdeclarative5-dialogs-plugin libqt5svg5
>>Unable to locate package qtdeclarative5-dialogs-plugin

YANGGANG1971, 2020-06-02:
Thank you a lot. Now it can project the image from pc to the virtual screen.
Sorry, I reopen this issue. 
But, I experienced one new problem.
When I created a green ball in the simulated world, the program can detect its position (the red point indicates that we find it). 
[image]

When I try to detect the ball in the virtual screen, we failed to detect the ball.
[image]

The complete code is as follows.
[detector.zip](https://github.com/robotology/QA/files/4718229/detector.zip)

pattacini, 2020-06-02:
Hi @YANGGANG1971,

Happy to see that you managed to make it work, eventually 👍 

Anyway, this issue has nothing to do with what you're asking. Please, next time, refrain from mixing up arguments in a single issue and open threads for each different topic 😉 

That said, Computer Vision is a complex research discipline and what works in a synthetic world won't work generally in the real world, unless you'll resort to photorealistic details (and still we know that this won't be enough).

Then, try to look it up on the net, as there are plenty of resources for this particular task, with code included.

For example, with a quick search on the internet, I've come across this [demo][1] 🎥 that has also the [code][2] available.

[1]: https://www.youtube.com/watch?v=hQ-bpfdWQh8
[2]; http://akaifi.github.io/MultiObjectTrackingBasedOnColor

closed 2020-12-08 by pattacini