robotology/community #406: Acquiring hand pose using simulated markers

gccunha1, 2020-06-02:
Hi!

## Brief description
I am looking for a way to simulate acquiring the hand pose using markers, both in the palm and in the back of the hand, in the iCub_SIM.

## Detailed context
The main objective is to make the reading of the pose closer to what would be in the real robot. I don't necessarily need to use markers. 

Some of my MSc thesis advisors mentioned that maybe I could attach objects to the palm and back of the hand. I see this is possible for the palm, using the GRAB FUNCTION (MAGNET), mentioned in http://wiki.icub.org/wiki/Simulator_README. Can something similar be done to the back of the hand?

After reading the contents in the link above, we also noticed it is possible to project video streams on surfaces. Can this be used to project some kind of marker on the hand?

pattacini, 2020-06-03:
I don't think you can use the MAGNET function to do what you aim to, as it should be working only with the palm and not with the back of the hand, as per the documentation.

Also, I doubt that you can reuse the projecting video stream feature along with the hand, as I've seen it working in the specific context of displaying videos on a fixed screen. However, I might be mistaken about this (@vtikha).

Given that the kinematics of the simulated iCub is known exactly, you could consider creating specific "weightless" markers whose motion could be controlled in real-time in order to have them tracking the hand with given offsets. If the hand won't be moving too fast, at a high level, the final behavior should replicate what you're looking for.

At any rate, there're been already a couple of active threads in researching how to visually acquire and then use the pose of the iCub hand, which you might find interesting and inspiring:
1. C. Fantacci, U. Pattacini, V. Tikhanoff, and L. Natale, “Visual end-effector tracking using a 3D model-aided particle filter for humanoid robot platforms,” 2017 IEEE RSJ Int Conf Intelligent Robots Syst Iros, pp. 1411–1418, 2017, doi: 10.1109/iros.2017.8205942.
1.  C. Fantacci, G. Vezzani, U. Pattacini, V. Tikhanoff, and L. Natale, “Markerless Visual Servoing on Unknown Objects for Humanoid Robot Platforms,” 2018 IEEE International Conference on Robotics and Automation (ICRA), pp. 3099–3106, 2018, doi: 10.1109/icra.2018.8462914.
1.  P. Vicente, L. Jamone, and A. Bernardino, “Online Body Schema Adaptation Based on Internal Mental Simulation and Multisensory Feedback,” Frontiers in Robotics and AI, vol. 3, 2016, doi: 10.3389/frobt.2016.00007.
1.  P. Vicente, L. Jamone, and A. Bernardino, “Markerless Eye-Hand Kinematic Calibration on the iCub Humanoid Robot,” Frontiers Robotics Ai, vol. 5, p. 46, 2018, doi: 10.3389/frobt.2018.00046.

cc @claudiofantacci @giuliavezzani @lornat75 @vicentepedro @lorejam @alexbernardino

gccunha1, 2020-06-03:
Thank you for the quick reply!
I will be trying your suggestion regarding the weightless markers, since it seems to be a good solution for my problem, and will keep in mind the works you mentioned.

closed 2020-12-09 by pattacini