robotology/community #408: High-level specifications for iCub & iCub stand

frank-foerster, 2020-06-26:
## Brief description
We will be "moving house" at some point and for that matter, I need to know some high-level technical specs
of both the iCub itself, as well as of its stand. I have been searching around a bit, but could not find all necessary information, so hence the question here. 

## Detailed context
I need to know the spatial dimensions (height, width, breadth), and the weight of both the iCub robot and its stand.
In the following I will state what I found, question marks indicate that I could not find this information.
And given that this is for deciding a place for robot, `width', and `breadth' should possibly refer to the maximum extension
of the robot arms (imagine a ballerina stretching the arms out, and twirling around its own axis)

_iCub_:
**_weight_**: 25kg 
(according to [> this](https://robots.ieee.org/robots/icub/) outside source. Please correct if this is wrong)

**_height_**: ?

**_width_**: ?
(assume maximally stretched out arms)

**_breadth_**: ?
(I would assume the same as the width, assuming the ballerina scenario mentioned above)

_iCub stand_:
**_weight_**: ?

**_height_**: ?
(for height I would assume the height including the length/height of the pole to which the iCub is attached)

**_width_**: 91cm 

**_breadth_**: 80cm
(I took or estimated width/breadth based on a design PDF on the IIT server, a link to which I can't find at the moment)

Many thanks for any help.

pattacini, 2020-06-29:
Hi @frank-foerster 

We'll give you this info soon. What is the HW version of your iCub? Does it have the battery backpack (probably not)?

frank-foerster, 2020-06-29:
Hi @pattacini ,

it doesn't have a backpack. It has been upgraded 5 years ago or so, so I'm not totally sure what verison it counts as now.
Thanks

pattacini, 2020-06-29:
Can you look it up in https://icub-tech-iit.github.io/documentation/icub_versions/?

pattacini, 2020-06-29:
@frank-foerster we're going to give you precise dimensions straight from the CAD.
Concerning the weights instead, it'd be better off referring to https://github.com/robotology/QA/issues/61 where we took measurements on a bunch of representative robots using the scale.

In particular, your robot should be considered of the category of `iCubGenova01` ➡ `30.3 kg`.

salvi-mattia, 2020-06-30:
I hope it can help [all measure in mm]

[image]
****
I don't know wich kind of trolley you are using, this is the most common one

[image]

frank-foerster, 2020-06-30:
Hi @pattacini and @salvi-mattia ,

thank you very much for the excellent information. Judging by the list that Ugo linked, I think we have version 1.4. We do have the 2nd version of the head, and  version 2.0 seems to include another head-related improvement. However, our head was upgraded either before we got the skin sensors or at the same time if I remember correctly. So I think our "head upgrade" is not the same as the one that is part of the v2 upgrade. 

But in terms of weight and dimensions the information you provided and linked here is more than sufficient. 

Many thanks!

closed 2020-12-15 by pattacini