robotology/community #409: Cannot find "usbCamera" plugin while projecting streaming to yarp port?

YANGGANG1971, 2020-06-29:
Following the [guidance ](https://github.com/robotology/QA/issues/402), the integrated webcam can be projected to iCub simulator. Thanks for your help. 

Recently, I am trying to use the desktop to do some experiments. There is no integrated webcam. So I moved on to using the usb camera by video4linux. 

1. ccmake ../    (recomplie yarp)
2. ENABLE_yarpmod_grabber ON
3. make 
4. yarpdev --device grabber --subdevice usbCamera --d /dev/video2
 (The external usbcam streaming is /dev/video2.)
But, each time, Cannot find "usbCamera" plugin will occur as follows. 
> [ERROR]/usr/local/src/robot/yarp/src/libYARP_os/src/yarp/os/YarpPluginSettings.h:240 bool yarp::os::YarpPluginSettings::readFromSelector(const string&): Cannot find "usbCamera" plugin (not built in, and no .ini file found for it)
> Check that YARP_DATA_DIRS leads to at least one directory with plugins/usbCamera.ini or share/yarp/plugins/usbCamera.ini in it
> yarpdev: ***ERROR*** could not find device <usbCamera>
> yarpdev: ***ERROR*** driver <grabber> was found but could not open
> [ERROR]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:503 static int yarp::dev::Drivers::yarpdev(int, char**): yarpdev: ***ERROR*** device not available.
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:511 static int yarp::dev::Drivers::yarpdev(int, char**): Suggestions:
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:512 static int yarp::dev::Drivers::yarpdev(int, char**): + Do "yarpdev --list" to see list of supported devices.
> [INFO]/usr/local/src/robot/yarp/src/libYARP_dev/src/yarp/dev/Drivers.cpp:515 static int yarp::dev::Drivers::yarpdev(int, char**): + Or append "--verbose" option to get more information.

Nicogene, 2020-06-29:
Hi @YANGGANG1971 ,

`grabber` is just the server that publish the images over the yarp network.
Depending on the version of `yarp`/`icub-main` you have you have to enable the flag [`ENABLE_usbCamera`](https://github.com/robotology/yarp/blob/master/src/devices/usbCamera/CMakeLists.txt#L21) in YARP **OR** `ENABLE_icubmod_usbCamera` in `icub-main` while running `cmake`/`ccmake`.

Originally was in the `icub-main` repository, at some point it has been moved to `yarp`.

YANGGANG1971, 2020-06-29:
My yarp/icub_main version is v1.15.0.
I did not find the options of ENABLE_usbCamera in YARP and ENABLE_icubmod_usbCamera in icub_main.
But, I found the ENABLE_yarpmod_usbCamera and ENABLE_yarpmod_usbCameraRaw in YARP as follows.
[image]
But I ran into some errors as follows.
[image]

Nicogene, 2020-06-29:
`ENABLE_yarpmod_usbCamera` is the correct option sorry, you have to install `video 4 linux` first:
```bash
sudo apt install libv4l-dev
```

YANGGANG1971, 2020-07-03:
Okay, I will test it and let you know. Thanks a lot.

pattacini, 2020-07-11:
Closing, feel free to open it up again if needed.