robotology/community #417: ClusterWidget: port /pc104 is not responding

tKsome, 2020-07-14:
**Describe the bug**
We have configured the laptop and icub server, but the yarpmanager cannot connect to the port /pc104

**Expected behavior**
We can ping to the pc104 and ssh to it, but when use yarpmanger connect the pc104, the yarprun is useless.
I just configure the ssh.

**Screenshots**
[WARNING]Cannot connect to /pc104 
[ERROR]ClusterWidget: port /pc104 is not responding 

**Configuration (please complete the following information):**
 - OS:Ubuntu18.04.4
 - yarp version:3.2.0
-  compiler:cmake3.17

pattacini, 2020-07-14:
Hi @tKsome 

Did you already check whether launching `yarprun` manually works out?
Do simply the following:
1. ssh into `pc104`
1. launch `yarprun --server /pc104`
1. try again if the yarpun broker is now alive (green light) from within the `yarpmanager`

pattacini, 2020-07-21:
Closing, feel free to open it up again if needed.

closed 2020-12-15 by pattacini