robotology/community #420: Instalation / Linking problem

jbaladron, 2020-07-14:
Hello, I am trying to install the robotology superbuild, and I am getting the following error:

[ 87%] Linking CXX executable ../../bin/yarpmanager
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetnum@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tputs@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgoto@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetent@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetstr@NCURSES6_TINFO_5.0.19991023'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libedit.so.2: undefined reference to `tgetflag@NCURSES6_TINFO_5.0.19991023'
collect2: error: ld returned 1 exit status
make[5]: *** [src/yarpmanager/CMakeFiles/yarpmanager.dir/build.make:625: bin/yarpmanager] Error 1
make[4]: *** [CMakeFiles/Makefile2:6568: src/yarpmanager/CMakeFiles/yarpmanager.dir/all] Error 2
make[3]: *** [Makefile:152: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:107: robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1265: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:95: all] Error 2

I have installed already libncurses5-dev and libncursesw5-dev

I am using Linux Mint 20

Any help would be appreciated. Thanks!

traversaro, 2020-07-16:
Hi @jbaladron , do you happen to have a conda workspace or some other non-official library active on your terminal? As any C++ project, all the libraries used by YARP need to be compatible with one another.

pattacini, 2020-07-22:
Closing, feel free to open it up again if needed.

closed 2020-12-15 by pattacini