robotology/community #427: How to let physical icub move as smooth as possible?

YANGGANG1971, 2020-09-07:
iCub simulator works well, but when moving the same code to physical iCub, the robot performs tremblingly. How to let physical iCub move as smooth as possible? 

I tried to utilize speed control to let physical iCub move as smooth as possible.
However, I am confused about the following code. If I set up as follows, does that mean moving from point A to point B, the iCub sim will move at the given constant speed (i.e. 0.01 m/s along each axis)? But when I tested the code on physical iCub, it seems that physical iCub does not move at a constant speed. How do I set the physical iCub move at the given constant speed? 

>          ICartesianControl *iarm;
>         Vector xdot(3); // move the end-effector along y-axis at specified velocity
>         xdot[0]=0.01;    // [m/s]
>         xdot[1]=0.01;
>         xdot[2]=0.01;
>         Vector odot(4); // no rotation is required
>         odot=0.0;  
>         iarm->setTaskVelocities(xdot,odot);
>        iarm->goToPose(x,o); 

In addition, does 5  in the parentheses mean that it will take max. 5 s from one point to the other in 3-D space? Changing this parameter seems to improve the smoothness of movement.
`  iarm->setTrajTime(5);

pattacini, 2020-09-07:
Hi @YANGGANG1971 

For the considerations I reported in #30 and #208, my warm recommendation is that you don't start with the method `setTaskVelocities()` but rather make use of **`goToPose()`** and/or **`goToPoseSync()`**.

YANGGANG1971, 2020-09-10:
Thanks for your inspirational suggestion.

closed 2020-12-08 by pattacini