robotology/community #428: iCub's Range of motion in 3-D space or Cartesian coordinate?

YANGGANG1971, 2020-09-07:
iCub's Range of motion in 3-D space or Cartesian coordinate? (x, y, z)
I want to control the physical iCub, could you give me the technical details?
In the icub wiki, http://wiki.icub.org/wiki/ICub_joints_limit, I only found the joint limit.

pattacini, 2020-09-07:
Hi @YANGGANG1971 

Unfortunately, we don't have precise details yet regarding the 3D workspace of the iCub, although we deem that this kind of information turns out to be particularly relevant, thus we will certainly publish something in this respect in the near future.

However, I can point you at `robotology/workspace-estimation`(https://github.com/robotology/workspace-estimation), which is a repository containing some code developed in the past by @alecive right for estimating those quantities you're interested in.

It's not documented but it looks a solid baseline. You could derive pictorial representations of the reaching workspace like the animation reported below (see videos in https://github.com/robotology/workspace-estimation/tree/master/examples):

[image]

Hope this helps for the time being.

YANGGANG1971, 2020-09-09:
Thanks a lot. It is useful.

YANGGANG1971, 2020-09-09:
This workspace did not include finger reaching workspace, right? Only for arm?

pattacini, 2020-09-09:
Usually, the reaching and/or dexterous workspace gets estimated for the end-effector only as it is in this case too.

However, nothing prevents you from applying the same approach to each finger of the hand. In this new context, the workspace would be computed with respect to the frame attached to that hand instead.

YANGGANG1971, 2020-09-09:
Thanks,  in https://github.com/robotology/icub-workspace-estimation/tree/master/examples): I want to ensure whether end-effector refers to the center of the palm.

closed 2020-12-15 by pattacini