robotology/community #429: Tutorial on iCub blink or face emotion?

YANGGANG1971, 2020-09-10:
Now, we want to control the physical robot as naturally as possible. Do you have any tutorials on iCub blink or face emotion? Or any website I  can refer to?

pattacini, 2020-09-10:
Hi @YANGGANG1971 

is your robot [iCubOsaka01][1]? Just asking beforehand as there exist different hardware versions underneath the use of facial expressions.

[1]: https://github.com/robotology/icub-tech-support/issues?q=is%3Aissue+author%3AYANGGANG1971

YANGGANG1971, 2020-09-10:
yes, iCubOsaka01

pattacini, 2020-09-10:
Here we go:
- Application scripts for running the face expressions are contained in the [main bundle][1] and described in our [official documentation][2]. While reading the doc, bear in mind that your robot `iCubOsaka01` doesn't mount the latest RFE board.
- We have also developed in the past a specific module controlling the blinking behavior. It's called [`iCubBlinker`][3] and is hosted at https://github.com/robotology/funny-things.

[1]: https://github.com/robotology/icub-main/tree/master/app/faceExpressions
[2]: https://robotology.github.io/robotology-documentation/doc/html/group__icub__faceExpressions.html
[3]: http://robotology.github.io/funny-things/module/iCubBlinker.html

YANGGANG1971, 2020-09-10:
Thanks, I also tried that.

pattacini, 2020-09-10:
Well, the bash scripts (e.g. [`cycle.sh`][1]) do simply rely on standard port-based communication.

Thereby, the following command:
```sh
echo "set all hap"  | yarp rpc /icub/face/emotions/in
```
has clearly its C++ counterpart:
```c++
yarp::os::BufferedPort<yarp::os::Bottle> port;
port.open("/emotions/out");
yarp::os::Network::connect(port.getName(), "/icub/face/emotions/in", "tcp");

yarp:os::Bottle& cmd = port.prepare();
cmd.clear();
cmd.addString("set all hap");
port.writeStrict();
```

[1]: https://github.com/robotology/icub-main/blob/master/app/faceExpressions/scripts/cycle.sh#L8

YANGGANG1971, 2020-09-10:
Thanks, I will test it on our physical robot.

YANGGANG1971, 2020-09-11:
Following your suggestion, it seems something wrong while opening icubDemoScripts to call modules followed by the tutorial
 https://github.com/robotology/funny-things#step-by-step-guide. Before that,  I have execute yarpserver --write and iCub_SIM.

YANGGANG1971, 2020-09-11:
Also, for the physical robot, the error even occurred at compiling and see the figure blow.
[image]

I guess it is due to yarp version or iCub_main version. In fact, I do not update the version, because I have many program relying on the current software version. 
My desktop configuration is as follows,
yarp 3.1.0
Ubuntu 16.04
cmake 3.5.1
swig 3.0.8
icub_main v1.12.0

Do you have any idea for addressing this issue, or other way to realize iCub blink?

pattacini, 2020-09-11:
Hi @YANGGANG1971 

From the error reported by `yarpmanager`

[image]

it seems a problem related to the availability of `yarprun`. So, you'd better check this out.

Unfortunately, `funny-things` is NOT versioned according to our [Software Versioning Table][1], so you ought to go back in the `funny-things` Git history up to the point in time when your `yarp` and `icub-main` versions have been delivered. This is a bit tricky but it's the fastest workaround, I believe.

[1]: http://wiki.icub.org/wiki/Software_Versioning_Table

kt10aan, 2020-09-11:
@YANGGANG1971 , the iCubOsaka01 was last updated around September 2018 if I remember correctly. Hopefully that information will narrow down the git history search.

YANGGANG1971, 2020-09-14:
> Hi @YANGGANG1971
> 
> From the error reported by `yarpmanager`
> 
> [image]
> 
> it seems a problem related to the availability of `yarprun`. So, you'd better check this out.
> 
> Unfortunately, `funny-things` is NOT versioned according to our [Software Versioning Table](http://wiki.icub.org/wiki/Software_Versioning_Table), so you ought to go back in the `funny-things` Git history up to the point in time when your `yarp` and `icub-main` versions have been delivered. This is a bit tricky but it's the fastest workaround, I believe.

I am a little bit confused. yarprun aims at killing, monitoring other programs on diff. set of machines. If I only use the iCub sim., do I need to yarprun --server /SERVERPORT? If indeed, how do I set the server name? p104?

pattacini, 2020-09-14:
The `yarprun` is a broker that is required to be running on machine `B` when you aim to launch an application from the machine `A` (that is where you rely on `yarpmanager`, for example) on machine `B`. Therefore, `yarprun` doesn't serve only for killing/monitoring remote applications but primarily for starting/stopping them.

When you want to launch an application on the same machine of `yarpmanager` you may replace `yarprun` broker with `localhost`.

From the snapshot you posted above, only `simFaceExpressions` seems correctly configured to run locally using the broker `localhost`, whereas the remainder of the modules are needing the `terminal` port of the `yarprun` server.

So, either you launch `yarprun --server /terminal` on your shell prior to clicking on <kbd>Run All</kbd> button, or simpler edit the application file to replace `terminal` with `localhost`.

YANGGANG1971, 2020-09-14:
Thanks, it works. But the new problem occurred as follows.
[image]
Error: cannot run gaze.lua on locahost: Can't execute command because no such file or directory. 
I am wondering by default, where gaze.lua should be put or where the .xml will first look for? Or how do I address this issue above?

pattacini, 2020-09-14:
The correct name of the module is indeed `simFaceExpressions`: fixed in https://github.com/robotology/funny-things/commit/927ffa0efbb1f81485e176a16e87a5a67bdd573a. You ought to edit the application file then.

The LUA script `gaze.lua` should get installed correctly in a location that shall be in the path. So, check it out by doing:
```sh
$ which gaze.lua
```

However, LUA does also need to be installed correctly. Follow these instructions: http://www.yarp.it/yarp_swig.html#yarp_swig_lua.

For testing blinking, `gaze.lua` is not necessary though.

YANGGANG1971, 2020-09-15:
In iCub Sim, now blinking is working. Thanks, but the gaze.lua still cannot work, although I have installed lua. It seems no affecting the blinking function. 
[image]

YANGGANG1971, 2020-09-15:
The physical iCub is also working. Thanks.

pattacini, 2020-09-15:
>The physical iCub is also working. Thanks.

Good to know 👍🏻 

The script `gaze.lua` has nothing to do with blinking as it merely asks the module `iKinGazeCtrl` to move the gaze toward some given locations, even randomly.

Did you try out one of the LUA examples available in YARP?
This one is a good start: https://github.com/robotology/yarp/blob/master/bindings/lua/examples/example.lua.

YANGGANG1971, 2020-09-15:
Thx,  I will try.

closed 2020-12-08 by pattacini