robotology/community #430: get hand joints software or hardware limits

kouroshD, 2020-09-11:
## Brief description
Similar to this wiki page http://wiki.icub.org/wiki/ICub_joints_limits where we can find the joint limits of iCub, I would like to know what are the software/hardware joint limits of the left and right hand.

## Detailed context

Is it possible to get the limits of the joints of the hand from a yarp device sth like https://www.yarp.it/classyarp_1_1dev_1_1IControlLimits.html#a67727bdb91bddeb1c070d44904d501cc that is implemented for the axis limits.

To be more precise, I need the limits for transforming the raw values of the [iAnalogSensor](https://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html) to the Degree. Therefore, the exact limits necessary for calibration process is: when the related axis goes to its boundaries, we should know what are the joints values  in order to calibrate tranform the raw values to degrees.

I saw there is the [calibrateSensor](https://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html#a277013600edac59c2059cb83f0d367f2) , [calibrateChannel](https://www.yarp.it/classyarp_1_1dev_1_1IAnalogSensor.html#ab69a5e6e94476ec8efdc2e656b64afa1) methods in iAnalogSensor device,  what are their use?!

@pattacini @xEnVrE

xEnVrE, 2020-09-11:
Hi @kouroshD,

if you want to extract plausible angular values for the joints of the entire chain of the finger starting from the motor encoders **and/or** the analog sensors, I suggest you reading [this](https://github.com/robotology/QA/issues/152) brief tutorial.

Regarding your specific question about limits, let me briefly add some context. From my experience, the maximum and minimum values that the analog sensors assume might depend from robot to robot. For this reason, the best solution would be to find these values by means of experiments and write down in a matrix. Once you have this calibration matrix, there is a specific method of the iCub forward kinematics API, namely [iKinFwd::iCubFinger::getChainJoints](https://robotology.github.io/robotology-documentation/doc/html/classiCub_1_1iKin_1_1iCubFinger.html#a0b9b05a6d0ea137eb00cff0f47481b2b), that allows you get the angles of the entire finger chain starting from the motor encoders, the analog sensors **and**, optionally, the calibration matrix.

Let me know if my explanation answers your question.

kouroshD, 2020-09-11:
Thank you @xEnVrE  for the response. I've checked already that issue, which is very useful.
However, in practice my question is how `chain[i].getMax()` and `chain[i].getMin()` are found in the issue you have mentioned. 

Also, I am curios to know:
 in `(1.0-std::min(1.0,std::max(0.0,analogs[2+i]/255.0)))` why it is written in this way, and why not simply `(1.0- analogs[2+i]/255.0)`? Because of possible spikes in the measures?!

kouroshD, 2020-09-11:
> From my experience, the maximum and minimum values that the analog sensors assume might change from robot to robot. 

I agree with you, I think it should depend on  how the hall effect sensors are placed in the fingers joints. Then, another question will arise is that, every time we turn on the robot, these values should be the same? Am I right? If  some maintenance goes on with the fingers, these values should be checked again. Is that correct?!

xEnVrE, 2020-09-11:
> However, in practice my question is how `chain[i].getMax()` and `chain[i].getMin()` are found in the issue you have mentioned.

These limits are set inside the [iCubFinger::allocate()](https://github.com/robotology/icub-main/blob/9b2a78301f5fc64a3223eb9515cc0c0f87c4dbcf/src/libraries/iKin/src/iKinFwd.cpp#L1841) method. E.g. at line [1932](https://github.com/robotology/icub-main/blob/9b2a78301f5fc64a3223eb9515cc0c0f87c4dbcf/src/libraries/iKin/src/iKinFwd.cpp#L1932), you find something like

```
pushLink(new iKinLink(0.0259, 0.0,       0.0, 0.0, 0.0, 90.0*CTRL_DEG2RAD));
```

where the last two arguments are exactly those returned by `getMin()` and `getMax()` respectively. However, nothing prevents you from aligning these limits with those coming from the robot using the method [iCubFinger::alignJointBounds()](https://robotology.github.io/robotology-documentation/doc/html/classiCub_1_1iKin_1_1iCubFinger.html#a6a3e0a8bf84bef2a74bcfd46fed52eee). This method accepts a list of `yarp::dev::IControlLimits` interfaces.

> Also, I am curios to know:
> in `(1.0-std::min(1.0,std::max(0.0,analogs[2+i]/255.0)))` why it is written in this way, and why not simply `(1.0- analogs[2+i]/255.0)`? Because of possible spikes in the measures?!

I think so, basically to handle negative (slightly below 0) or > 255 readings.

> I agree with you, I think it should depend on how the hall effect sensors are placed in the fingers joints. Then, another question will arise is that, every time we turn on the robot, these values [...]

Not an easy question. Maybe the best solution is to repeat the calibration every so often or, at least, check if matches the one you have stored before doing an important experiment. Also, it depends on the kind of experiment you have in mind. If you can accept "some" errors, the ranged spanned by the sensors should be more or less the same.

pattacini, 2020-09-11:
`(1.0-std::min(1.0,std::max(0.0,analogs[2+i]/255.0)))` is just a safety guard to make 100% sure that nothing will escape those bounds, even if the source will be yielding data withing the allowed range.

kouroshD, 2020-09-11:
Thank you for explanation @xEnVrE . 
To complement you explanation,  we have 19 joints, each finger 4 joints except the middle 3 joints.
All the link numbers in [ iCubFinger::allocate()](https://github.com/robotology/icub-main/blob/9b2a78301f5fc64a3223eb9515cc0c0f87c4dbcf/src/libraries/iKin/src/iKinFwd.cpp#L1897-L1998)  are consistent with that and with the order abduction/adduction, proximal, and the two distal;. However,  thumb has 5 links, where the second one has no dof and is a fixed link: https://github.com/robotology/icub-main/blob/9b2a78301f5fc64a3223eb9515cc0c0f87c4dbcf/src/libraries/iKin/src/iKinFwd.cpp#L1898

closed 2020-12-08 by pattacini