robotology/community #431: Understand the Mechanism of the hand thumb oppose and proximal

kouroshD, 2020-09-16:
We have 2 joints for the thumb oppose and thumb proximal. Also, we have two motors/axes for controlling them.
But we have one hall position sensor to retrieve the joint information. 
Also, when we control the  thumb oppose, we see that the thumb proximal will move.
So, the question is how they are constrained to each other? What we are measuring with the hall sensor?

cc @pattacini  @xEnVrE @traversaro

pattacini, 2020-09-16:
Hi @kouroshD 

>We have 2 joints for the thumb oppose and thumb proximal

Reading the [code][1], it comes out that the feedback of the thumb opposition is retrieved straight from the motor, lacking the readout from the joint, so we have no hall sensor here.

Differently, the first hall readout goes feeding the thumb proximal joint. 

>Also, when we control the thumb oppose, we see that the thumb proximal will move.

Yep, the design foresees a mechanical coupling between those two joints of the thumb. When you manually move one of the two, the other will start moving too. However, the FW at the low-level acts in such a way that the two joints appear to be decoupled, instead, from the SW standpoint.

[1]: https://github.com/robotology/icub-main/blob/master/src/libraries/iKin/src/iKinFwd.cpp#L2166

kouroshD, 2020-09-17:
> Yep, the design foresees a mechanical coupling between those two joints of the thumb. When you manually move one of the two, the other will start moving too. However, the FW at the low-level operates in such a way that the two joints appear to be decoupled, instead, from the high-level SW standpoint.

I should say that in tests I have performed when the robot joints reaches its boundaries, the other joint moves as well, for example moving the thumb opposition close to its boundaries, causes the motion of thumb proximal. 
@ale-git 

> the hall sensor gives us the position of the proximal joint.

I will test this on iCub04 and will confirm it. Because, when I was moving the joints using `yarpmotorgui` this is not what I have observed.

pattacini, 2020-09-17:
>when I was moving the joints using yarpmotorgui this is not what I have observed.

Uhm, the `yarpmotorgui` doesn't allow you to see the low-level joint feedback but only the motors' readouts if I'm not mistaken.

pattacini, 2020-09-17:
>the FW at the low-level operates in such a way that the two joints appear to be decoupled, instead, from the high-level SW standpoint.

Perhaps, @randaz81 can also give heads-up on this.

kouroshD, 2020-09-17:
> Uhm, the yarpmotorgui doesn't allow you to see the low-level joint feedback but only the motors' encoders readouts if I'm not mistaken.

What I meant was the fact that I was giving the motor ref commands through the `yarpmotorgui` and checking the values of the analog sensors bay reading the related port.

xEnVrE, 2020-11-12:
I am re-opening this issue to ask clarifications about some doubts that I have on the coupling between thumb opposition and thumb proximal.

Considering the firmware (at least the one that I think is used in iCubGenova01, should be [this one](https://github.com/robotology/icub-firmware/blob/60c6b0485d178a08f124e000212239fc32ddd51b/motorControllerDsp56f807/common_source_code/pwm_decoupling.c#L126)) it is clear that these two are coupled as also @pattacini clarified in a previous comment.

In the past, we had issues while trying to take advantage of the iCub forward kinematics library resulting in situations like this

[image]

where basically the predicted configuration of the thumb (check the contour in red) is different from the actual one.

As clarified in previous comments, the `iKinFwd` library takes the feedback for the thumb opposition straight from the motor encoder. Thus, I would like to ask whether this "motor reading" is already taking into account the coupling or if this is just the position of one of the two motors that control the thumb opposition.

cc @kouroshD

pattacini, 2020-11-12:
Ok @xEnVrE 👍🏻 
We can work on this in our next sprint `PI05-sprint03-fix` (i.e. >= 23 Nov).

xEnVrE, 2020-11-12:
> Reading the code, it comes out that the feedback of the thumb opposition is retrieved straight from the motor, lacking the readout from the joint, so we have no hall sensor here.

After a very interesting journey in the iCub schematics it seems that there is an additional hall sensor for the thumb opposition actually.

[image]

[image]

pattacini, 2020-11-13:
Thanks @xEnVrE for the journey 😄
We'll have a look at it to understand from the ground up why we're not employing it, that is if it's a mistake or there's a reason.

pattacini, 2020-11-25:
Let me report here below what @xEnVrE and @randaz81 have just found out during their analysis:
1. The hall sensor of the thumb opposition is actually exposed to the high-level software, although not through the usual analog port but rather through the port `/...[right|left]_arm/state:o`. The forward kinematics lib makes use of it consistently.
1. The joint encoders are those used to close the control loop, actually. 

Thus, the problem reported in https://github.com/robotology/QA/issues/431#issuecomment-726087161 might be due to either a malfunctioning encoder or it's needed to perform calibration again.

xEnVrE, 2020-11-25:
Thank you @pattacini for reporting this. I just wanted to add that our analysis is limited to CAN robots like `iCubGenova01`.

pattacini, 2020-11-26:
From a quick discussion, it turned out that ETH robots behave similarly.

Closing then, feel free to comment further if needed.

closed 2020-12-08 by pattacini