robotology/community #434: icubosaka01: how to reset up on icubsrv ?

YANGGANG1971, 2020-09-21:
I ran into [a login loop issue](https://github.com/robotology/icub-tech-support/issues/983). Now I am considering whether I need to reinstall all software on icubsrv. Is it complicated? Could you give me some tutorials?

pattacini, 2020-09-21:
Hi @YANGGANG1971 

It's unfortunate that you got stuck in this problem.
In the next days, @mbrunettini will give you a walkthrough guide to attempt to get around this problem.

YANGGANG1971, 2020-09-23:
Thanks, @pattacini @mbrunettini

YANGGANG1971, 2020-09-25:
We have fixed the login issue. But if possible, we still intend to learn how to reset up icubsrv in case, also for backup.

pattacini, 2020-09-25:
@YANGGANG1971 you asked the same thing twice in two different places.
Closing here.

closed 2020-12-08 by pattacini