robotology/community #439: CAN bus terminating resistor on strain2 force/torque sensors ?

mherrb, 2020-10-01:
## Brief description

Does the strain2 sensor include a 120 Ohm terminating resistor on the CAN interface ?

## Detailed context

We have bought individual strain2 sensors to be used in our applications. I'm currently using a 2m long cable on which I have a 120 Ohms terminator on the CAN controller end, but nothing on the sensor end.

It works well most of the time, but from time to time the sensors stays in boot mode at power up. From what I know this can be caused by the sensor getting CAN data in the 1st 5 seconds after power up. I'm sure our application is not sending anything (nor is the Linux kernel while booting and setting up the CAN bus drivers.

So I'm wondering it it could be some line noise caused by the lack of termination resistor on the sensor end of the cable.

simeonedussoni, 2020-10-01:
There is a solderable jumper (JP1) near the TP28 on the TOP side. This jumper soldered, the CAN bus is terminated at 120 ohms. You can check on your device.

maggia80, 2020-10-01:
Hi @mherrb, the **strain2** does not have the 120 Ohm terminating resistor. It's better to add it externally on your CAN line

closed 2020-12-15 by pattacini