robotology/community #440: how can we do a horizontal rotation (left-right) of camera images?

pfdominey, 2020-10-08:
## Brief description
Greetings, we are trying to do an experiment where we want to create an illusion by having the images from the two cameras rotated around the horizontal axis so the left and right are reversed.  

## Detailed context
We have a yarp port containing the image coming from the yarpdev camera output that is connected to another input port for a head mounted display.  For subjects wearing the HMD, we want to create the illusion that left is right and right is left.  I wonder if there is a parameter somewhere that I can change to generate the rotated image.
thanks for your help.
cheers
Peter

## Accompanying material

PeterBowman, 2020-10-08:
Hello. Are these RGB cameras (webcams)? The `opencv_grabber` device accepts the `flip_x` and `flip_y` parameters, which might be what you want: [ref](https://github.com/robotology/yarp/blob/6478740a2055e00256fedfa47d96717346e35e99/src/devices/opencv/OpenCVGrabber.cpp#L117-L118).

pfdominey, 2020-10-08:
Thanks Peter!  In fact, they are the cameras in the iCub eyes.  Do you think what you suggest can be done with these cameras?
thanks
cheers
Peter

PeterBowman, 2020-10-08:
I'm afraid I can't help, I have no experience nor have I actually worked with a real iCub. Anyway, it is my guess that iCub eyes host a pair of dragonfly cameras ([ref](https://github.com/robotology/robots-configuration/tree/40b8e8893fd11e42fff53a1454854319bd922ad7/iCubEdinburgh01/camera)). In the device description, there is no appropriate parameter for frame flipping/mirroring ([ref](https://robotology.github.io/robotology-documentation/doc/html/group__icub__dragonfly2.html)).

pattacini, 2020-10-08:
Hi @pfdominey 

Glad to hear from you! Hope you're doing fine.

Assuming that you're dealing with only one stream of right and left images glued together side by side (otherwise you could simply swap the 2 different flows), I don't think there's already an option to achieve what you're looking for.

Anyway, I'm not an expert of HMD; perhaps @drdanz and/or @randaz81 plus @S-Dafarra can chime in providing heads-up on this as they're developers/users of the Oculus device.

At any rate, flipping the images should be rather easy, boiling down to just one [OpenCV call][1]. One could code that in a simple relay module running in between the images source and sink.

The same `cv::flip()` function is also used within the [`opencv_grabber`][2] YARP device.

[1]: http://opencv-tutorials-hub.blogspot.com/2015/11/flip-image-in-opencv-by-using-flip-function-opencv-hub-cplusplus-code.html
[2]: https://github.com/robotology/yarp/blob/master/src/devices/opencv/OpenCVGrabber.cpp#L255-L266

pfdominey, 2020-10-09:
Hi Ugo, Good to hear from you!  Do you know if there is an .ini file somewhere in the icub yarp world where we can change a parameter to invoke this flip?
Thanks,
cheers
Peter

pattacini, 2020-10-11:
Hi Peter,

I don't know how you're generating the aggregate image conveying both the left and the right cameras' output.

Probably, you're relying on the following lines somewhere in your startup applications:
```xml
<module>
  <name>yarpdev</name>
      <parameters> --from camera/ServerGrabberDualDragon.ini --split false</parameters>
  <node>pc104</node>
</module>
```
as per https://github.com/robotology/icub-main/blob/master/app/default/scripts/cameras_calib.xml.template#L5-L9.

If this is the case, you may try out the following:
```xml
<module>
  <name>yarpdev</name>
      <parameters> --from camera/ServerGrabberDualDragon.ini --split false --left_config camera/dragonfly2_config_right.ini --right_config camera/dragonfly2_config_left.ini</parameters>
  <node>pc104</node>
</module>
```
This will tell the `ServerGrabberDual` device to swap the left with the right.

pfdominey, 2020-10-11:
Hi Ugo, Thanks.  But one thing - We dont wnt to swap left and right eye, but to perform a flip around the vertical axis so that within a given image, left is right and right is left.
Any suggestions?
is there a "flip" parameter somewhere?
Otherwise
We will probably try to do this in the program that reads the output from the yarpdev in calibratedcamers for the two eyes.
Let me know if you have any ideas
thanks!

pattacini, 2020-10-11:
Then, I'm afraid we haven't a parameter for that flip, at least to my knowledge.

S-Dafarra, 2020-11-23:
For a completely unrelated investigation, I was reading at the OVR headset device code, and I found the optional parameter ``flipInput` https://github.com/robotology/yarp-device-ovrheadset/blob/master/src/devices/ovrheadset/OVRHeadset.cpp#L572. I don't know if it is related, but it reminded me of this issue. It should be fairly easy to test. You need to run the Oculus device and then press the "F" button on your keyboard (I guess) on the window that appears (the one where you can see what the operator is seeing). Relevant code https://github.com/robotology/yarp-device-ovrheadset/blob/a2f3b2b45a1b06f5fa03b2bae3166f4de9fc8652/src/devices/ovrheadset/OVRHeadset.cpp#L1278-L1282

drdanz, 2020-11-23:
Another possible related thing is [`yarp::sig::Image::setTopIsLowIndex`](https://yarp.it/git-master/classyarp_1_1sig_1_1Image.html#a6f8200f072a307ca44a166dcd4afb723) which should flip the image vertically.
There's no option for horizontal flipping, thought.

closed 2020-12-08 by pattacini