robotology/community #441: Documentation on how to write the configuration files for connecting multiple ETH boards

plinioMoreno, 2020-10-22:
## Brief description
We want to write the configuration files for running yarprobotinterface on a custom setup of ETH control boards, to be able to start from a small set of files to a larger set of files (which will be the files to control our robot Vizzy). To understand better how to write the files, we would appreciate a pointer to the documentation (or tutorials) on how to write the files. 

## Detailed context
Our setup so far is one ETH board, which is controlling three motors. We can communicate to the ETH board, which is an mc4. 
[image]

Our idea is to add other boards in daisy chain after we manage to have this first setup running.

## Accompanying material

plinioMoreno, 2020-10-28:
Following the picture, and looking into the configuration file examples at [this single ETH motor control](https://github.com/robotology/robots-configuration/tree/master/experimentalSetups/single-ETH-2FOC-motor) and at [this single ETH motor control](https://github.com/robotology/robots-configuration/tree/master/experimentalSetups/singleETHboard), we were able to run `yarprobotinterface`. The files we are using for the setup are at [this repo](https://github.com/vislab-tecnico-lisboa/eth-boards-setups)

We noticed that in the IP address 10.0.1.XX, XX has to be less than 100, otherwise we get the error "theNVmanager::Impl::transceiverZ() cannot obtain from TheEthManager a EthResource * for IP = 10.0.1.100"

After changing the IP address to 10.0.1.25, when running `yarprobotinterface`, we get the messages as follows

```
[DEBUG]eth::parser::print(pc104Data) for PC104:
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345
[DEBUG]PC104/PC104TXrate =  1
[DEBUG]PC104/PC104RXrate =  5
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms
[WARNING]in EthReceiver::config() the config socket has queue size =  1048576 ; you request ETHRECEIVER_BUFFER_SIZE=
[DEBUG]eth::parser::print(boardData) for BOARD knee-eb10-j0
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES:
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.25
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS:
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  knee-eb10-j0
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.25 (knee-eb10-j0) with timeout = 0.02 sec and period of missing report = 60 sec
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.25
[WARNING]embObjMC BOARD  knee-eb10-j0 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
[DEBUG]embObjMC BOARD  knee-eb10-j0 :fromConfig() detected that IMPEDANCE parameters section is found
[INFO]embObjMC BOARD  knee-eb10-j0 IMPEDANCE section: parameters successfully loaded
[INFO] from BOARD 10.0.1.25 (áNÂØ>☺), src LOCAL, adr 0, time 1396s 827m 353u: (code 0x00000034, par16 0x0001 par64 0x0000000000000005) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::verifyBoardPresence() found BOARD knee-eb10-j0 with IP 10.0.1.25 after 0.002659 seconds
[ERROR]EthResource::verifyBoardTransceiver() detected different mn protocol minor versions: local = 15 , remote = 14 : cannot proceed any further.
[ERROR]ACTION REQUIRED: BOARD knee-eb10-j0 with IP 10.0.1.25 needs a FW update.
[ERROR]EthResource::verifyEPprotocol() cannot verify BOARD knee-eb10-j0 with IP 10.0.1.25 : cannot proceed any further
[ERROR]embObjMotionControl: failed verifyEPprotocol. Cannot continue!
```

@marcoaccame and @davidetome, the board where we are running the setup is an MC4. Which firmware file should we use for updating the firmware? Help on this will be very much appreciated.

I'm also sending the printscreen of the FirmwareUpdater of the MC4 board

[image]

davidetome, 2020-10-29:
Ciao @plinioMoreno , you can find the application [here](https://github.com/robotology/icub-firmware-build/tree/master/ETH/MC4PLUS/bin/application)

To upload the application: 
1. Select the board and put it in Maintenance mode by clicking `Force ETH maintenance` button
2. Flash the firmware clicking on `Upload Application`
3. Click on `Restart ETH  Board(s)`

plinioMoreno, 2020-10-29:
Thanks for the pointer to the file @davidetome. I realized that the firmware version has to match between the pc and the board, so I had to use an older version of `mc4plus.hex`, and now that message does not show up anymore. To further advance on the configuration files, I have a couple of questions about the options for the motor configuration:

On the [connection ports where the motors are connected](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/main/single-ETH-1-motor-test-vislab/hardware/motorControl/knee-eb10-j0-mc_service-V25.xml#L23), is there a document where can I find the numbers associated to the physical ports?

On the [motor type definition](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/main/single-ETH-1-motor-test-vislab/hardware/mechanicals/knee-eb10-j0-mec-V25.xml#L19), which are the options for the type of motors?

If we want to add another board, connected in daisy chain fashion, what are the additions that we need to do?

Thanks in advance!

davidetome, 2020-10-30:
Ciao @plinioMoreno ,

1. The connectors available on the `mc4plus` board to drive dc motors are `P2 P3 P4 P5` : 

[image]

So in the file, you have to write the connector you used.

2. The motor type, as far as I know at the moment can be : 
- `BLL_MOOG` for brushless motors on the robot (Moog is the brand of the motor); on the robot on shoulders/torso/legs for example
- `DC` for small DC motors; on the robot in the hands and on the head for example

3. To add a board you surely have to add relatives files and include them [here](https://github.com/robotology/robots-configuration/blob/master/experimentalSetups/single-ETH-2FOC-motor/test_knee_joint_V25.xml)

plinioMoreno, 2020-11-02:
Thanks again @davidetome,

In our setup (one `mcplus` with one motor connected), Now we are getting the message that the fault button is pressed, so that means that we have to connect the FAULT pins of the `mc4plus`. However, @rnun3s did not find any FAULT connection wire.

In the documentation of the electronics
[image]
we could not find the fault pins.

On the other hand @rnun3s found the FAULT pins on the EMS|2FOC boards (marked in red)
[image]

In our robot Vizzy, we have only `DC` motors connected to the control boards, and in our present setup, we had the firmware of the `BLL` boards to work with DC motors. In the case of the FOC boards, can we control DC motors? Or we need an special version of the firmware?

davidetome, 2020-11-03:
Ciao @plinioMoreno ,

The `FAULT` signals of the `mc4plus` are on `P6` connector:

- `P6_2_FLT12` 
- `P6_3_FLT34`

[image]

You have to connect them to `GND` to enable motors driving, otherwise, in a  normal state, they are pulled up on the board and the motors can't be driven, We usually manage that with a `FAULT` emergency button that interrupts the connection to ground once pressed. 

I don't know the wiring of your setup but the `mc4plus` can drive 4 `DC motors`; anyway if you need more detailed info about motor specs, driving etc. you could ask to @MrAndrea that is the responsible of the electronic designers.

cc @pattacini  @maggia80 @marcoaccame

maggia80, 2020-11-03:
@plinioMoreno the FW for the 2FOC used for controlling DC motors is the following: https://github.com/robotology/icub-firmware-build/blob/devel/CAN/2foc/2foc.r1.dc.hex

hereafter a schema about how to connect a DC motor to the 2FOC:

[image]

plinioMoreno, 2020-11-04:
Thanks @davidetome for the indication of the fault of the `mc4plus` board.

I hope you can help us with the following: In our current simplified setup (one DC motor connected to a MC4plus board, configuration files [here](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/main/single-ETH-1-motor-test-vislab)) we are able to initialize `yarprobotinteface` and `yarpmotorgui`. In the `yarpmotorgui` we see that our motor is in state "NOT_CONFIGURED". What do we need to add in the configuration files to see the encoder values? I guess is the Calibrator. Below you can see the yarpmotorgui screen

[image]

and the output of the `yarprobotinterface` after we pressed the RUN button at `yarpmotorgui`

```
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 4 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 5 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 6 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 7 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 8 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 9 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 10 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 11 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 12 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 13 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 14 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 15 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 16 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 17 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 18 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j0 IP 10.0.1.25, joint 0) -> current mode = cfgn, requested = pos
[ERROR]A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j0 IP 10.0.1.25, joint 0, current mode: cfgn, requested: pos
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=pos) for BOARD knee-eb10-j0 IP 10.0.1.25 after 18 attempts and 0.265068 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= pos ) for  BOARD knee-eb10-j0 (IP 10.0.1.25)   has failed checkRemoteControlModeStatus()
```

davidetome, 2020-11-04:
Ciao @plinioMoreno , 

 has stated in this [conf](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/041f22fef777e2b654fe9e28c5ecb9c2d4012b83/single-ETH-1-motor-test-vislab/hardware/motorControl/knee-eb10-j0-mc_service-V25.xml#L27) file the `mc4plus` is expecting 2 `AEA` sensors (absolute encoders)  usually placed at the joint level on `P10` and `P11` connectors and 2 `qenc` sensors (quadrature encoders) inside the motors on `P3` and `P4` connectors.

So, first of all, you should remove a joint from the file I linked and then set the values accordingly your setup...

cc @maggia80 @marcoaccame @pattacini

plinioMoreno, 2020-11-04:
Hi  @davidetome, I removed the extra joint in [the file](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/main/single-ETH-1-motor-test-vislab/hardware/motorControl/knee-eb10-j0-mc_service-V25.xml). After removing the extra motor, the behavior is the same as in my [previous comment](https://github.com/robotology/QA/issues/441#issuecomment-721775683)

Regarding the encoders, in this setup we do not have `AEA` sensors. But if I remove that type of encoders from the file, I get the following error:

```
[ERROR]ServiceParser::check_motion() cannot find PROPERTIES.JOINTMAPPING.ENCODER1
[ERROR]ServiceParser::parseService() gets same errors parsing SERVICE MC group
[ERROR]embObjMC  BOARD knee-eb10-j0 (IP 10.0.1.25)  cannot parse service
[ERROR]BOARD knee-eb10-j0 (IP 10.0.1.25)  Missing motion control parameters in config file
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open
```
Instead of removing the encoder `AEA`, what should we define on that type of encoder for running successfully?

plinioMoreno, 2020-11-04:
Hi @davidetome and @marcoaccame, we managed to discover by looking at some other configuration files and guessing, that:

- To read the encoder values, we added the `<param name="JointEncoderType">     "OPTICAL_QUAD" </param>` in the `mec` [file.](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/4bea5c03e40d997db3466e7df32ee7ee1d4c6ed7/single-ETH-1-motor-test-vislab/hardware/mechanicals/knee-eb10-j0-mec-V25.xml#L16) Now we can read the encoder values

- To ignore the `AEA` encoders we set the `ENCODER1` type to none as `<param name="type">             none      </param>` in [this file](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/4bea5c03e40d997db3466e7df32ee7ee1d4c6ed7/single-ETH-1-motor-test-vislab/hardware/motorControl/knee-eb10-j0-mc_service-V25.xml#L27)

- We believe we have to add the Calibration section on the xml files, for the motor to be started properly.

I would like to schedule a skype session to talk to you about our setup and the our ideas for the setup of our robot Vizzy. This because we want to have our simple setup operational during next week.

plinioMoreno, 2020-11-05:
Hi @davidetome and @marcoaccame,

We have been modifying the `Calibrator` configuration files and changed the control modes in [this commit](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/ba08837cd90e8f4c0095d695b4c00b510ee281a2), but we have not been able to control the motor. When we run `yarprobotinterface` we get:

```
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\yarprobotinterface.ini] (pwd)
||| found D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\yarprobotinterface.ini
||| finding file [config]
||| checking [D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\./test_knee_joint_V25.xml] (pwd)
||| found D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\./test_knee_joint_V25.xml
[DEBUG]Reading file D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\.\test_knee_joint_V25.xml
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x
[DEBUG]Reading file D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\.\test_knee_joint_V25.xml
[DEBUG]Preprocessor complete in:  0.0013773 s
||| finding paths [plugins]
||| checking [D:\Users\RNunes\Documents\Software\eth-boards-setups\single-ETH-1-motor-test-vislab\plugins] (pwd)
||| checking [C:\Users\RNunes\AppData\Roaming\yarp\config\plugins] (YARP_CONFIG_HOME)
||| checking [C:\Users\RNunes\AppData\Roaming\yarp\plugins] (YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\plugins] (YARP_CONFIG_DIRS)
||| checking [D:\Users\RNunes\Documents\Software\yarp\install/share/yarp\plugins] (YARP_DATA_DIRS)
||| found D:\Users\RNunes\Documents\Software\yarp\install/share/yarp\plugins
||| checking [D:\Users\RNunes\Documents\Software\icub-main\build/share/iCub\plugins] (YARP_DATA_DIRS)
||| found D:\Users\RNunes\Documents\Software\icub-main\build/share/iCub\plugins
yarp: Port /icub/yarprobotinterface active at tcp://10.0.1.104:10002/
[INFO]startup phase starting...
||| finding paths [plugins]
[DEBUG]eth::parser::print(pc104Data) for PC104:
[DEBUG]PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345
[DEBUG]PC104/PC104TXrate =  1
[DEBUG]PC104/PC104RXrate =  5
[DEBUG]EthSender is a PeriodicThread with txrate = 1 ms
[DEBUG]EthReceiver is a PeriodicThread with rxrate = 5 ms
[WARNING]in EthReceiver::config() the config socket has queue size =  1048576 ; you request ETHRECEIVER_BUFFER_SIZE=
[DEBUG]eth::parser::print(boardData) for BOARD knee-eb10-j0
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES:
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.25
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768
[DEBUG]ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS:
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/Name =  knee-eb10-j0
[DEBUG]ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE
[DEBUG]ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60
[DEBUG]eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.25 (knee-eb10-j0) with timeout = 0.02 sec and period of missing report = 60 sec
[DEBUG]TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.25
[WARNING]embObjMC BOARD  knee-eb10-j0 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
[DEBUG]embObjMC BOARD  knee-eb10-j0 :fromConfig() detected that IMPEDANCE parameters section is found
[INFO]embObjMC BOARD  knee-eb10-j0 IMPEDANCE section: parameters successfully loaded
[INFO] from BOARD 10.0.1.25 (ðS), src LOCAL, adr 0, time 4027s 514m 713u: (code 0x00000034, par16 0x0001 par64 0x000000000000076c) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG]EthResource::verifyBoardPresence() found BOARD knee-eb10-j0 with IP 10.0.1.25 after 0.0036037 seconds
[DEBUG]EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD knee-eb10-j0 with IP 10.0.1.25
[DEBUG]EthResource::cleanBoardBehaviour() has cleaned the application in BOARD knee-eb10-j0 with IP 10.0.1.25 : config mode + cleared all its regulars
[DEBUG]EthResource::setTimingOfRunningCycle() for BOARD knee-eb10-j0 with IP 10.0.1.25 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO]EthResource::askBoardVersion() found BOARD knee-eb10-j0 @ IP 10.0.1.25 of type mc4plus with FW = ver 3.22 built on 2019 Dec 10 16:14
[DEBUG] from BOARD 10.0.1.25 (ðS), src LOCAL, adr 0, time 4027s 558m 178u: (code 0x05000015, par16 0x1000 par64 0x000000000000000f) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[... cut ...]
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=pos) for BOARD knee-eb10-j0 IP 10.0.1.25 after 18 attempts and 0.260796 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= pos ) for  BOARD knee-eb10-j0 (IP 10.0.1.25)   has failed checkRemoteControlModeStatus()
```
Any pointers on what do we need to change to change the mode from `cfgn` to `pos` are very appreciated. We were trying to check how to define the calibrator, but we just found [this page](http://wiki.icub.org/wiki/Calibration_types_available_on_iCub_and_R1_robots)

davidetome, 2020-11-06:
Ciao @marcoaccame ,
As requested by @plinioMoreno, are you available for a skype meeting on Tuesday 10 AM for example?
It should be usefull the presence of @MrAndrea or @maggia80 for any needs on electronics, are you available?

Feel free to propose different date/time accordingly your needs.

cc @pattacini

plinioMoreno, 2020-11-06:
Hi @davidetome, on our side we are available the coming Tuesday at 10a.m. at your time zone (9a.m in Lisbon).

We will have the ETH boards setup and remote access to our desktop connected to the setup through anydesk.

davidetome, 2020-11-09:
Ciao @plinioMoreno ,

 Unfortunately tomorrow my peers are busy, anyway I spoke with @marcoaccame and in the meantime, it should be useful if you send us some detailed info about your setup.
In particular, to start analyzing it and gain some time we need to know:

1. Block diagram of your system 
2. Sensor type you use (only incremental enc?)
3. If you do not use `absolute` enc, we may need a hard stop and know which angle it correspond
4. Type of `actuators` and datasheet
5. Eventual reduction ratio between the `motor` and `joint`
6. Mechanical limits of the joint (position `min` and `max`=`hard stop`)

We can start to take a look at this material and give you a feedback asap.

Thank you in advance!

cc @marcoaccame @pattacini @maggia80

plinioMoreno, 2020-11-09:
Hi  @davidetome,
> 
> 
> Ciao @plinioMoreno ,
> 
> Unfortunately tomorrow my peers are busy, anyway I spoke with @marcoaccame and in the meantime, it should be useful if you send us some detailed info about your setup.

The configuration files are in [this repository](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/ba08837cd90e8f4c0095d695b4c00b510ee281a2/single-ETH-1-motor-test-vislab)
> In particular, to start analyzing it and gain some time we need to know:
> 
>     1. Block diagram of your system
> 
[image]
In this very simple setup, just one DC motor is connected to the mc4plus board

>     2. Sensor type you use (only incremental enc?)
> 
Just incremental encoders

>     3. If you do not use `absolute` enc, we may need a hard stop and know which angle it correspond
>
In the setup, the motor is free, so it is not linked to anything.
Regarding this point, Is it actually needed the hard stop? We are asking this because in our robot Vizzy, the fingers are underactuated, meaning that the finger limbs are controlled by a motor that pulls a string (like the ones for fishing), so we cannot apply the hard stop because the motor encoder angle cannot be mapped always to the same angular position. Due to this design constraint, we do not use any calibration and assume that the fingers are at home position when the robot is started.
 
>     4. Type of `actuators` and datasheet; also the rotation direction (`clockwise` or `counter clock`)
> 
The reference of the only connected motor is  https://www.faulhaber.com/fileadmin/Import/Media/EN_2232_SR_DFF.pdf

>     5. Eventual `reduction ratio` between the `motor` and `joint`

No reduction ratio
> 
>     6. Mechanical limits of the joint (position `min` and `max`=`hard stop`)

No min nor max because the motor is free
> 
> 
> We can start to take a look at this material and give you a feedback asap.
> 
> Thank you in advance!
> 
> cc @marcoaccame @pattacini @maggia80

davidetome, 2020-11-09:
> Hi @davidetome,
> 
> > Ciao @plinioMoreno ,
> > Unfortunately tomorrow my peers are busy, anyway I spoke with @marcoaccame and in the meantime, it should be useful if you send us some detailed info about your setup.
> 
> The configuration files are in [this repository](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/ba08837cd90e8f4c0095d695b4c00b510ee281a2/single-ETH-1-motor-test-vislab)
> 
> > In particular, to start analyzing it and gain some time we need to know:
> > ```
> > 1. Block diagram of your system
> > ```
> 
> [image]
> 
> In this very simple setup, just one DC motor is connected to the mc4plus board
> > ```
> > 2. Sensor type you use (only incremental enc?)
> > ```
> 
> Just incremental encoders
> 
> > ```
> > 3. If you do not use `absolute` enc, we may need a hard stop and know which angle it correspond
> > ```
> 
> In the setup, the motor is free, so it is not linked to anything.
> Regarding this point, Is it actually needed the hard stop? We are asking this because in our robot Vizzy, the fingers are underactuated, meaning that the finger limbs are controlled by a motor that pulls a string (like the ones for fishing), so we cannot apply the hard stop because the motor encoder angle cannot be mapped always to the same angular position. Due to this design constraint, we do not use any calibration and assume that the fingers are at home position when the robot is started.
> 
> > ```
> > 4. Type of `actuators` and datasheet; also the rotation direction (`clockwise` or `counter clock`)
> > ```
> 
> The reference of the only connected motor is https://www.faulhaber.com/fileadmin/Import/Media/EN_2232_SR_DFF.pdf
> 
> > ```
> > 5. Eventual `reduction ratio` between the `motor` and `joint`
> > ```
> 
> No reduction ratio
> 
> > ```
> > 6. Mechanical limits of the joint (position `min` and `max`=`hard stop`)
> > ```
> 
> No min nor max because the motor is free
> 
> > We can start to take a look at this material and give you a feedback asap.
> 
> Thank you!
> 
> > Thank you in advance!
> > cc @marcoaccame @pattacini @maggia80

Thank you @plinioMoreno , we'll let you know asap

cc @marcoaccame @pattacini @maggia80

plinioMoreno, 2020-11-11:
> Thank you @plinioMoreno , we'll let you know asap
> 
> cc @marcoaccame @pattacini @maggia80

Hi @davidetome @marcoaccame @maggia80 @pattacini did you have a chance to look at [our answer](https://github.com/robotology/QA/issues/441#issuecomment-724010452)?

davidetome, 2020-11-12:
Sorry @plinioMoreno but we had some big issues this week and we didn't look yet; Today I'm in IIT lab and I'll pick up a motor and a board to do some tests tomorrow morning. I hope to give you a starting/working point tomorrow!

cc @marcoaccame @pattacini @maggia80

plinioMoreno, 2020-11-12:
Hi @davidetome and @marcoaccame, we were able to control the motor in position mode in our very simple setup. The missing part was that the calibrator was [commented in the main xml file](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/ba08837cd90e8f4c0095d695b4c00b510ee281a2/single-ETH-1-motor-test-vislab/test_knee_joint_V25.xml#L15).

Now we are moving to a setup with two control boards, one `mc4plus` and one `ems4`. The drawing of the setup is the following:
[image]

Our main question here is how to specify the "address" of the motor that we want to control, which is port connected to the `2foc`. This port is defined in [this file](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/ef9304616889145e9be11a53c5e3f9876df594be/two-ETH-2-motor-test-vislab/hardware/motorControl/knee-eb10-j1-mc_service-V25.xml#L36) and has the structure `CAN1:X:X`. According to the drawing above, which is the correct port?

And also, the type of the board in [this line](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/ef9304616889145e9be11a53c5e3f9876df594be/two-ETH-2-motor-test-vislab/hardware/motorControl/knee-eb10-j1-mc_service-V25.xml#L18) is the correct one? Right now is `foc`.

Thanks in advance.

davidetome, 2020-11-13:
Ciao @plinioMoreno , I'm happy to hear about that; about your question I honestly can't answer... I've already asked my peers and we'll let you now 

cc @marcoaccame @maggia80 @pattacini

plinioMoreno, 2020-11-16:
@maggia80, thank you for the answer.

We were able to read the encoder value (in `yarpmotorgui`) of the motor connected to the `2foc` board, using the firmware that you pointed [above](https://github.com/robotology/QA/issues/441#issuecomment-721104672). One detail about this, to be able to see the encoder value correctly, we had to click on the `Run` button of `yarpmotorgui`. If we not click, the value is not displayed correctly. This is strange, because for other motors is not the case, e.g. to see the encoder of the DC motor connected to the `mc4plus`.

However, we are not able to control the motor yet. The calibrator file [here](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/ebb2adaf3f729493081aff84a91fa337b7d9c96f/two-ETH-2-motor-test-vislab/calibrators/knee-one-calib.xml), is failing with the following info:
```
[ERROR]Knee_one_Calibrator : Timeout while calibrating  0
[ERROR]Knee_one_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit

[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 4 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 5 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 6 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 7 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 8 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 9 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 10 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 11 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 12 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 13 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 14 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 15 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 16 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 17 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 18 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[ERROR]A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j1 IP 10.0.1.26, joint 0, current mode: cal, requested: idl
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD knee-eb10-j1 IP 10.0.1.26 after 18 attempts and 0.257374 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD knee-eb10-j1 (IP 10.0.1.26)   has failed checkRemoteControlModeStatus()
[DEBUG]knee-one-calibrator finished calibration of device knee-one-mc_wrapper
```

Thus, we wonder how to setup the calibrator on a free motor (not attached to anything) attached to the `2foc` board. We were able to setup a calibrator for the free motor connected to the `mc4plus`, [here](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/ebb2adaf3f729493081aff84a91fa337b7d9c96f/two-ETH-2-motor-test-vislab/calibrators/knee-calib.xml)

maggia80, 2020-11-17:
@plinioMoreno if you use a calibration type 5 (https://wiki.icub.org/wiki/Calibration_types_available_on_iCub_and_R1_robots#Calibration_5)  you should have the calibration 1 value different from 0. It's the PWM needed to move the joint to the HW stop.

plinioMoreno, 2020-11-17:
@maggia80, we want to figure out how to run the calibration for the fingers of the robot. The fingers have three joints, but only one motor that pulls them (see point 3 [here](https://github.com/robotology/QA/issues/441#issuecomment-724010452)). Since we do not have encoders on the finger limbs, we want to setup a calibration that does not move the motor, and sets the value 0. Any help on how to do this is appreciated.

maggia80, 2020-11-17:
ok I see. here:

`[DEBUG]Knee_one_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  -217.822`

it seems that the board is reading a value different from 0. Since the PWM is 0 the joint doesn't move and therefore it goes in timeout the calibration. You should switch off and on the board and try again. Then, send me the log

plinioMoreno, 2020-11-17:
@maggia80, we reset the boards and run the calibration. The log from the joint we want to calibrate:

```
[DEBUG]Knee_one_Calibrator : Calling calibrateJoint on joint  0 : type  5  with params:  0 0 0 0 0
[DEBUG]Knee_one_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  0

[ERROR]Knee_one_Calibrator : Timeout while calibrating  0
[ERROR]Knee_one_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit

[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 4 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 5 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 6 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 7 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 8 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 9 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 10 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 11 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 12 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 13 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 14 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 15 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 16 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 17 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[ERROR]A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j1 IP 10.0.1.26, joint 0, current mode: cal, requested: idl
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD knee-eb10-j1 IP 10.0.1.26 after 17 attempts and 0.251571 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD knee-eb10-j1 (IP 10.0.1.26)   has failed checkRemoteControlModeStatus()
[DEBUG]knee-one-calibrator finished calibration of device knee-one-mc_wrapper
```
We also did a test where we changed the value of `calibrattion1` to 1, guessing that if the PWM is very low, then it reaches the calibration value. However, it failed as well. The output as follows:
```
[DEBUG]Knee_one_Calibrator : Calling calibrateJoint on joint  0 : type  5  with params:  1 0 0 0 0
[DEBUG]Knee_one_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  0

[ERROR]Knee_one_Calibrator : Timeout while calibrating  0
[ERROR]Knee_one_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit

[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[... cut ...]
[ERROR]A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j1 IP 10.0.1.26, joint 0, current mode: cal, requested: idl
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD knee-eb10-j1 IP 10.0.1.26 after 17 attempts and 0.252155 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD knee-eb10-j1 (IP 10.0.1.26)   has failed checkRemoteControlModeStatus()
[DEBUG]knee-one-calibrator finished calibration of device knee-one-mc_wrapper
```

maggia80, 2020-11-17:
@plinioMoreno 

try with:

calibrationType: **0**   
calibration1 0 (PWM)
calibration 2 10 ( speed )
calibration 3 0  (setting after calibration ends)

plinioMoreno, 2020-11-17:
@maggia80, I was checking with @rnun3s, and the motors of the fingers will be connected to one `mc4plus`, so the test of the free motor is not needed for the `2foc`. In our simple setup one motor setup [here](https://github.com/robotology/QA/issues/441#issuecomment-724010452) connected to an `mc4plus`, we were able to start the free motor at position 0.

Regarding the `2foc`, we want to test the calibration type 5 with a DC motor connected to the `2foc`, having a hard stop. On t[his repository](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/4df1649a3474d88e84a800b05da3a555cb07102b/two-ETH-2-motor-test-vislab) you can find the calibration configuration file [that we tried](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/4df1649a3474d88e84a800b05da3a555cb07102b/two-ETH-2-motor-test-vislab/calibrators/knee-one-calib.xml). When the calibration starts, the motor starts to move in one direction, then we stop it manually, but then the motor does not move in the opposite direction and we get the timeout of the calibration as follows:

```
[DEBUG]Knee_one_Calibrator : Calling calibrateJoint on joint  0 : type  5  with params:  4000 500 0 0 0
 [DEBUG]Knee_one_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  19.802
[DEBUG] from BOARD 10.0.1.25 (á{Ù[×☻), src LOCAL, adr 0, time 21s 407m 640u: (code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01 + SET CONTROLMODE
[DEBUG]Knee_Calibrator : calib joint  0 ended
[DEBUG]Knee_Calibrator : set 1 : Calibration ended, going to zero!

[DEBUG]Knee_Calibrator : Sending positionMove to joint 0  (desired pos:  0 desired speed:  2  )
[DEBUG]Knee_Calibrator: checkGoneToZeroThreshold: joint: 0 curr: 0.000 des: 0.000 -> delta: 0.000 threshold: 2.000 output: 0.000 mode: pos
[DEBUG]Knee_Calibrator: checkGoneToZeroThreshold: joint: 0 completed with delta: 0.000 over: 2.000
[DEBUG]Knee_Calibrator : set 1 : Reached zero position!

[DEBUG]knee-calibrator finished calibration of device knee-mc_wrapper
[ERROR]Knee_one_Calibrator : Timeout while calibrating  0
[ERROR]Knee_one_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit

[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 3 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 4 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 5 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 6 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 7 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 8 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 9 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 10 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 11 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 12 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 13 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 14 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 15 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[WARNING]embObjMotionControl::checkRemoteControlModeStatus() has done 16 attempts and will retry again after a 0.010000 sec delay. (BOARD knee-eb10-j1 IP 10.0.1.26, joint 0) -> current mode = cal, requested = idl
[... cut ...]
[ERROR]A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j1 IP 10.0.1.26, joint 0, current mode: cal, requested: idl
[ERROR]failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD knee-eb10-j1 IP 10.0.1.26 after 18 attempts and 0.258886 seconds
[ERROR]In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD knee-eb10-j1 (IP 10.0.1.26)   has failed checkRemoteControlModeStatus()

```

@maggia80, I wonder if we can do a short skype (with remote access to the files) to check this.

maggia80, 2020-11-17:
@plinioMoreno @rnun3s I think that the calibration was successful. The positionMove failed. Have you tried to increase the desired speed from 2 to a higher value? i.e. 20

plinioMoreno, 2020-11-17:
@maggia80, the joint that was successful (`Knee_Calibrator`) is the motor connected to the `mc4plus`. But the one connected to the `2foc` is the `Knee_one_Calibrator`, which failed.

maggia80, 2020-11-18:
@plinioMoreno you are right, but have you tried to set a higher speed? maybe 2 is too slow. I don't see any other error in the files. 
However even if the motor failed the calibration you should be able to run it anyway by running yarpmotorgui --admin

plinioMoreno, 2020-11-18:
@maggia80, I changed the value to 20 and the motor does not move in the other direction after stopping. 

In this video, [higher_velocity.zip](https://github.com/robotology/QA/files/5562142/higher_velocity.zip), you can see the calibration file in the editor and the `yarpmotorgui` window (Thanks for reminding the `-- admin` option for running the calibration) . We tried with another DC motor and the behavior is the same, the motor starts to move for the hardstop, resets the value of the encoder and then nothing else happens and the calibration fails.

Can we schedule an skype, please? You will be able to access our pc remotely and it will not take much time for you to give us pointers on what to change.

Looking forward to hearing from you.

maggia80, 2020-11-19:
@plinioMoreno I am trying to get the help someone with the skills for solving your issue. I'm not the right person and I don't think it's easy to find the issue. i.e. we never used the 2FOC with DC motors and using Calibration Type 5, therefore we never tested it. There could be either a bug in the FW or a configuration issue.

maggia80, 2020-11-19:
@plinioMoreno looking at the video, the value of the encoder goes beyond 100. You should change this to 1000 or so. 

```
<param name="jntPosMax">                100    </param>
        <param name="jntPosMin">                  -100    </param>  
```

plinioMoreno, 2020-11-19:
@maggia80 , thank you for giving us more context on this problem. Just for us to have more information, do you want to send the link to the source code of the firmware that we are using? The version is in [this link](https://github.com/robotology/QA/issues/441#issuecomment-721104672)
> 
> 
> @plinioMoreno looking at the video, the value of the encoder goes beyond 100. You should change this to 1000 or so.
> 
> ```
> <param name="jntPosMax">                100    </param>
>         <param name="jntPosMin">                  -100    </param>  
> ```
> 
Ok. We will try this change in the motor contol file [here](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/4df1649a3474d88e84a800b05da3a555cb07102b/two-ETH-2-motor-test-vislab/hardware/motorControl/knee-eb10-j1-mc-V25.xml)

> it's strange also the fact that at a certain point the encoder reads 9.9 and not 0. And it seems it resets after 360. Do you know why? The calibrationDelta is 0.1 and just for testing purposing put it higher than 9.9

What I realized is that right after I stop the motor with my hand, the value of the encoder is changed to the value of `calibration2` divided by a factor (which 10 in this case). This pattern always happened.

plinioMoreno, 2020-11-19:
> 
> 
> @plinioMoreno I am trying to get the help from someone with the skills for solving your issue. I'm not the right person and I don't think it's easy to find the issue. i.e. we never used the 2FOC with DC motors and using Calibration Type 5, therefore we never tested it. There could be either a bug in the FW or a configuration issue.

Thank you for giving us more information. Now we know better what to expect. Do you want to cc the right person this messages? Should I start a new issue just about the `2foc` controlling DC motors?

plinioMoreno, 2020-11-19:
@maggia80 , we changed the joint limits [here](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/349b3567d78c0ee881185a0313a523ce366d3915/two-ETH-2-motor-test-vislab/hardware/motorControl/knee-eb10-j1-mc-V25.xml#L17), and the the result is still the same: The motor start to move, we stop it and the encoder value is changed, and then the calibration fails after the timeout is reached.

Let us know if we should do other test.

plinioMoreno, 2020-11-30:
@maggia80, have you been able to do tests with the DC motor and the `2foc` board? Otherwise, let us know if we should do tests here with a new version of the firmware.

maggia80, 2020-11-30:
hi @plinioMoreno sorry for not replying to you, but we are in a very busy period. end of the year (budget review, project review) + covid, it makes it challenging to be responsive as we would like to be.
The FW you have should already address the `calibration type 5`. Unfortunately, we cannot do a quick setup for testing it because of the period. Let me add to the issue @ale-git who did the FW, in case he may give you any suggestion.
Meantime, could you try, using `yarpmotorgui --admin`  to move the joint? Even if the calibration failed you could be able to run it anyway. It would be helpful to have a video of the motor and `yarpmotorgui` during the calibration.

closed 2020-12-15 by pattacini