robotology/community #442: Joint 3 of both arms in the iCub_SIM will not move if 15.385 degrees is commanded

gccunha1, 2020-11-05:
##

## Brief description
Hello!
I noticed that joint number 3 in the iCub_SIM right arm will not move if the reference is the minimum allowed value, which is 15.385 degrees. This happens both using code, as well as using yarpmotorgui.

## Detailed context
I previously posted issue #406 here about adding fiducial marker textures to the iCub_SIM and place them on the robot's hand.
I noticed that sometimes the hand and the markers would be in completely different positions, but thought I was the one miscalculating the world coordinates, but then realized this happened when a value close to 15.285 degrees was commanded to joint 3 of the right arm, and it ended up not moving. This happens with joint 3 of the left arm aswell.

I attached a .zip file containing a video of what happens in yarpmotorgui, which is identical when commands are sent using code.

## Accompanying material

gccunha1, 2020-11-05:
[Qt Robot Motor GUI V2.0 2020-11-05 16-06-23.zip](https://github.com/robotology/QA/files/5495612/Qt.Robot.Motor.GUI.V2.0.2020-11-05.16-06-23.zip)

For some reason the attachment didn't make it to the mais post.

pattacini, 2020-11-06:
Hi @gccunha1 

I've made a test with a command line swiss knife called `simpleClient`.

```sh
>> simpleClient --robot icubSim --part right_arm
-> get llim 3
llim: limits: (15.50 106.00)
-> set pos 3 15.50
pos: movint 3 to 15.50 
-> get encs
Bottle: get encs 3
encs: (-25.00 20.00 0.00 15.50 0.00 -0.00 0.00 59.00 20.00 20.00 20.00 10.00 10.00 10.00 10.00 10.00 )
-> set pos 3 106.0
Bottle: set pos 3 106.0
pos: moving 3 to 106.00
-> get encs
Bottle: get encs
encs: (-25.00 20.00 0.00 106.00 0.00 0.00 0.00 59.00 20.00 20.00 20.00 10.00 10.00 10.00 10.00 10.00 )
```

As you can see from the log above, the actual joint limits are `(15.5 106)` degrees. That can explain why you've been experiencing problems from the `yarpmotorgui`, which instead is wrongly allowing you to play with different bounds, apparently.

When it comes down to the code, you must have asked to move to `15.385` deg without checking whether this target was feasible, i.e. within the allowed range. To this end, rely on [`IControlLimits::getLimits()`][1].

Regarding the weird beahavior of `yarpmotorgui`, my suggestion is to open up an issue in YARP at https://github.com/robotology/yarp/issues. You may want to link to this issue as a reference.

[1]: http://yarp.it/git-master/classyarp_1_1dev_1_1IControlLimits.html#a67727bdb91bddeb1c070d44904d501cc

gccunha1, 2020-11-09:
> When it comes down to the code, you must have asked to move to 15.385 deg without checking whether this target was feasible, i.e. within the allowed range. To this end, rely on IControlLimits::getLimits()

Yes, I was not doing that and I actually didn't know that function, I was relying on the values from yarpmotorgui and I hard-coded the bounds. I will open the issue at  https://github.com/robotology/yarp/issues.

Thanks for the reply!

closed 2020-12-15 by pattacini