robotology/community #444: Ethernet and CAN cables - reference/link

plinioMoreno, 2020-11-23:
## Brief description
We want to know the type of cables (reference or link) that you use for the ethernet connections of the ETH boards. Also what kind of cables (reference or link) you use for the CAN connections between ems4 and 2foc boards.

## Detailed context
We want to upgrade our robot Vizzy, from the CAN-based control and power boards, to the ETH-based control and power boards.

cc @rnun3s

## Accompanying material

pattacini, 2020-11-23:
@Fabrizio69 and/or @emilianob80 could you help @plinioMoreno with this request?

emilianob80, 2020-11-23:
Hi @plinioMoreno , below are the codes of the electrical cables we use for the connections you requested :  

- ETH connection  

IIT cod. 6367 | CONNEX ITALIANA SRL | P556320 | codice prototipo A19693B1| [SKM_C25819032612450.pdf](https://github.com/robotology/QA/files/5583677/SKM_C25819032612450.pdf)
-- | -- | -- | -- | --

- CAN connection

**Long connections**

IIT cod. 85 | INDUSTRIFIL | TPU-3219.04A | 4xAWG32 TPU | [Industrifil_FlexCableAWG32.pdf](https://github.com/robotology/QA/files/5583630/Industrifil_FlexCableAWG32.pdf)
-- | -- | -- | -- | --

**Very short connections (less than 10 cm)** only green white black wires are used

IIT cod. 1590 | RS COMPONENTS S.P.A. | 660-3989 | https://github.com/robotology/QA/files/5583706/RS-.Cavo.6.conduttori.non.schermato.7x0.1.pdf) | 
-- | --| -- | --

**ATTENTION: cables 85 and 6367 are custom made**

Hope this is the information you needed. In case I remain available

cc: @Fabrizio69 @pattacini @maggia80

plinioMoreno, 2020-11-24:
@emilianob80 , thanks for the details on the cables. In the case of the custom made cables, we wonder if we can buy them from you. If we could purchase them from you, who can we contact? Thanks in advance.

emilianob80, 2020-11-24:
> @emilianob80 , thanks for the details on the cables. In the case of the custom made cables, we wonder if we can buy them from you. If we could purchase them from you, who can we contact? Thanks in advance.

Hi @plinioMoreno , as far as purchases are concerned, you can refer to @Fabrizio69 

cc: @pattacini

plinioMoreno, 2020-11-24:
Thanks @emilianob80. @Fabrizio69, how do we proceed to to do a purchase of ETH and CAN cables?

plinioMoreno, 2020-11-27:
@emilianob80, I did not an answer from @Fabrizio69, do you want to contact him for the purchase request?

plinioMoreno, 2020-11-30:
@Fabrizio69, a reminder about the purchase request of Ethernet and CAN cables for our `mc4plus`, `ems4` and `2foc` boards.

emilianob80, 2020-11-30:
@pattacini could you assign it to @Fabrizio69  that I can't? thanks

pattacini, 2020-11-30:
>@pattacini could you assign it to @Fabrizio69 that I can't? thanks

Done 👍🏻 
You did part of the job so it's correct that you remain an assignee.

emilianob80, 2020-11-30:
> > @pattacini could you assign it to @Fabrizio69 that I can't? thanks
> 
> Done 👍🏻
> You did part of the job so it's correct that you remain an assignee.

@pattacini ok! thanks

maggia80, 2020-11-30:
Hi @plinioMoreno ,

I think that it's easier for you and for us if you buy directly from Industrifill the cable for the CANbus. It's a standard cable. [see comment here](https://github.com/robotology/QA/issues/444#issuecomment-732208927)

For the ETH connections, we can't resell you the cable but it's quite expensive and I don't think it's a good idea for you to buy from us such a cable. I would suggest you this one:  [Alphawire Ecocable mini](https://it.rs-online.com/web/p/cavi-twistati-multicoppia/1368490/). It's quite small and it's a good commercial alternative. You could use this one for both CAN and ETH.